OctoPrint/OctoPrint #1176: [BUG] Canceled print upon connection

Myseka, 2016-01-04:
Was 3 - 4 hours into a print, went to check status of it with web viewer and it canceled my print.......

Need option to not write on connect.

Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1373
Changing monitoring state from 'Sending file to SD' to 'Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1373'
Connection closed, closing down monitor
Changing monitoring state from 'Offline' to 'Detecting serial port'
Serial port list: ['/dev/ttyACM0']
Connecting to: /dev/ttyACM0
Error while connecting to /dev/ttyACM0: 'Failed to enter programming mode'
Changing monitoring state from 'Detecting serial port' to 'Error: Failed to autodetect serial port, please set it manually.'
Failed to autodetect serial port, please set it manually.
Changing monitoring state from 'Offline' to 'Detecting serial port'
Serial port list: ['/dev/ttyACM0']
Connecting to: /dev/ttyACM0
Connecting to: /dev/ttyACM0
Changing monitoring state from 'Detecting serial port' to 'Opening serial port'
Connected to: Serial<id=0x2ed3170, open=True>(port='/dev/ttyACM0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: M110
Recv: start
Send: M110
Recv: echo: External Reset
Recv: Marlin1.0.2
Recv: echo: Last Updated: Jan  2 2016 22:21:40 | Author: (MakerFarm,Pegasus 10)
Recv: Compiled: Jan  2 2016
Recv: echo: Free Memory: 3126  PlannerBufferBytes: 1232
Recv: echo:V18 stored settings retrieved (366 bytes)
Recv: echo:SD card ok
Changing monitoring state from 'Connecting' to 'Operational'
Recv: ok
Send: M105
Recv: echo:Unknown command: ""
Recv: ok
Send: M20
Recv: Begin file list
Recv: BB8DOM~1.GC~
Recv: BB8_DO~1.GCO
Recv: HOLLOW~1.GCO
Recv: TESTGC~1.GCO
Recv: BB8DOM~2.GC~
Recv: End file list
Recv: ok
Send: M105
Recv: ok T:186.1 /0.0 B:103.3 /0.0 T0:186.1 /0.0 @:0 B@:0
Send: M105
Recv: ok T:181.8 /0.0 B:102.7 /0.0 T0:181.8 /0.0 @:0 B@:0

foosel, 2016-01-11:
That had nothing to do with you connecting to the web interface (and that's something that works perfectly fine):

    Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' 

Your printer severed the connection. There's nothing OctoPrint could have done about that.

    Recv: echo: External Reset

This implies that your firmware detected an external reset - power loss maybe?

Check your cabling, your power supply etc. Make sure to use a properly shielded USB cable to connect your printer to the machine running OctoPrint and to ideally not run that cable next to stepper motors (lots of noise).

closed 2016-01-11 by foosel

Myseka, 2016-01-11:
I figured out what causes it - Someone said Arduino resets when a
connection is attempted. Could possibly add a warning to connect?

ntoff, 2016-01-21:
That'd mean every time octoprint wants to connect you'll need to confirm it. Just turn off auto connect.

If you were thinking it'd detect when a print is in progress and warn you about connecting, it can't know anything is printing without first connecting so.....

Just cut the trace on your mega that leads to the reset pin https://www.arduino.cc/en/Main/ArduinoBoardMega2560 scroll down to Automatic (software) reset