OctoPrint/OctoPrint #1180: Printer aborted print after clicking resume

dguisinger, 2016-01-08:
I honestly don't know how to classify this issue, but I can describe what I did.
I was 11 hours into a 15 hour print.  Realizing that the oiler was going to stop the printer from reaching the last 100 layers I paused the print and removed the oiler (which requires me to change the filament).

I did this by clicking Pause, going into Control and moving the Y and Z axis's to get the head out of the way of the model.

During a filament change, my filament always gets stuck and wrestling with it always causes the print head to move.  Once complete, I hit Resume and of course the X axis was off and resume wasn't very successful, so I paused it again.

I raised the Z axis, and hit Home on the X/Y controls.
Everything homed correctly.

I lowered the Z axis, and hit Resume
The printer slowly started to move the print head back to the correct location, then all of a sudden I received a extruder heater error message saying the temperature had dropped.  After that happened, OctoPrint aborted the print, and dropped my serial connection.

So now I have no way to recover my print, and I feel like there may be one or two assumptions OctoPrint is making that aren't correct....

I'm assuming (though I don't know for sure) that the printer on a Home command turned off the heaters without OctoPrint knowing, since I didn't have the issue when I paused the first time.

I can understand wanting to abort the print if its a hardware failure, but in my case I was unaware the printer shut the heater off and would have gladly waited to warm it back up.

Is it by design that the printer does a hard abort after the temp drops?  Can this be changed so the current print state is not lost and the serial connection is not dropped?  Instead of a hard abort, maybe pause the printer?

closed 2016-01-08 by dguisinger

Salandora, 2016-01-09:
Just some note on this if you move the head away, OctoPrint is not able to remember where the real position was.
If your printer sends something like: "Error: Min/Max temp triggered" than it's not OctoPrint's fault, your printer on its own will stop doing anything to protect you and itself, until you manually reset it.

dguisinger, 2016-01-09:
It wasn't supposed to move, originally it felt locked in place, then half way through changing the filament the x axis unlocked and started moving.  Once I did my x homing, it was headed back to the correct place when the temp error threw.

Still seems to me there should be away to save where things left off, regardless of what the problem was....

foosel, 2016-01-09:
The problem is though that OctoPrint doesn't know what the last command was that the firmware executed before an error. It does know what it already sent to the printer and it indeed could save that, but that wouldn't necessarily help here. Thing is, firmwares have a planning buffer of a size unknown to OctoPrint. Movement commands aren't necessarily immediately executed when they are acknowledged (And a lot of existing versions of marlin don't even acknowledge command in the received order on top of that) but might still reside in the buffer for as much a several seconds (think long linear moves). If an error now occurs, anything in the buffer will be lost. OctoPrint has no way at all to figure out what of the commands it sent to the printer actually were executed already and what still resided in the buffer, so if it now just resumed to print from the last known position in the file, chances are high that depending on the model the print would go wrong anyhow. And hence there are things on the to do list that offer a better return of investment of resources than trying to implement a print recovery function that wouldn't achieve the desired results most of the time anyhow (and instead less to a likewise amount of tickets).

What we can do however is adding the position in the failed file to the fail event, that would allow plugin developers to implement something like that.

nophead, 2016-01-09:
If you are using Marlin it can be configured to turn the motors off after a
period of inactivity. That is probably why they lost position and then
needed to be homed. If your machine homes Z towards the bed you will then
have a big problem!

Best to disable the timeout as leaving the motors on is not really a
problem. Possibly Marlin also turns the nozzle off after a timeout, but it
would be a bug if it then issued a too cold error. I configure the firmware
so it does what it is told as much as possible and leave the decisions to
me.