OctoPrint/OctoPrint #1187: [request] small graph for fan speed

Lord2Vader, 2016-01-14:
Would be nice to help calibration/setting the fanspeeds right for a large or small prints and overhangs.

foosel, 2016-01-14:
There is no way to get the current fan speed from the firmware, so that might prove a bit tricky. It's pretty much a write-only thing.

The only thing that might be possible is scanning sent commands for M106 and M107 (from a plugin) and using that to plot, but that will only work for fan commands sent through OctoPrint (so not for SD prints).

Lord2Vader, 2016-01-14:
Yeah that what i meant. Repetierhost for example does it by checking where it is in the gcode (or just by sending the data, which octoprint does too) and setting the graph that way. Just a percentage checking. For example slic3r gives you the ability to set a range but you need to read the whole code to know whats going on. Just had a problem with the fan running to fast and blowing the nozzle cold. Changed the speed via octoprint terminal but wont get a response of whats going on if iam not at the printer ;)

ntoff, 2016-01-15:
Some slicers allow you to just set your maximum run speed. I had to limit mine to 80% because of the same reason as you, it cools the nozzle off.

VBen, 2016-12-29:
Perhaps it can be included in the temperature graph via plugin after the request #321 is completed

larp-welt, 2017-07-18:
I did my first plugin! :-)

It scans the code send to the printer for M106 and M107-commands and shows the fan speed last set inside the navbar.

I'm no programmer at all, I'm new to Ocotoprint, it is a long time, since I did something in python, it's mostly copy'n'past from other plugins, I give no support, it's just for me! ;-)

Translation only works in 50%. 

But it works in my testsystem and it is good enought for my own use!

Maybe someone likes it: https://github.com/larp-welt/OctoPrint-FanSpeed

Do with the code what ever you want!

[image]

KC703, 2017-07-21:
Brilliant!  Showing the fan speed is a great addition.

ntoff, 2017-07-21:
Awesome!

LazeMSS, 2021-03-31:
There are plugins for this:
https://plugins.octoprint.org/plugins/toptemp/

closed 2021-03-31 by LazeMSS