OctoPrint/OctoPrint #1188: Printrbot connecting

toboooy, 2016-01-15:
Hi,
I wanted to connect my Printrbot Metal to OctoPi running on my Raspberry Pi 2, but there are no serial ports to choose and when i want to set an own serial port (dev/tty/AMC0) it doesn't work. Only the serial port dev/tty/AMA0 works and it tries to connect with my printer, but after a while there stands "offline" again.

Any ideas?

nophead, 2016-01-15:
Note it is /dev/ttyACM0 not dev/tty/AMC0. If it doesn't become available to
select when you plug in the printer then you have a problem with the USB
connection.

/dev/AMA0 is the GPIO serial port so it is always present whether anything
is connected to it or not.

markwal, 2016-01-15:
This is also a good candidate for the Google Group and/or Plus Community as mentioned in the guidelines, but nophead is right, if ttyACM0 isn't showing up, it's because the Pi itself isn't noticing that there is a bot plugged in.  If you turned on your printer after visiting the webpage, you have to refresh your browser to get a new list of ports.  Give it a moment to come all the way on first.

closed 2016-01-15 by markwal

toboooy, 2016-01-18:
Hi,
First of all thanks for the answer. When I want to autodetect the serial
port there is a warning that OctoPi can't detect the serial port itself, so
i have to choose it manually. But I can't choose any serial port instead of
AUTO. When I want to set up a new serial port only /dev/ttyAMA0 is
detected, but /dev/ttyAMC0 not. So I can choose between AUTO and
/dev/ttyAMA0, but there is nothing with /dev/ttyAMC0. And the USB
connection is not the Problem I tested it with Cura on my PC.

Sorry for the incomplete question, but English is not my best subjekt in
school ^^.

Thanks Tobi

nophead, 2016-01-18:
What does dmesg show after you plug it in?
What does lsusb show?

foosel, 2016-01-18:
If you are not seeing the ttyACM0 port in OctoPrint, it's because the
operating system that OctoPrint is running on is not seeing your printer.
There's nothing that OctoPrint can do about that. You have to figure out
why the printer doesn't register with the OS properly. Maybe it does but
doesn't (for whatever reason) register under dev/ttyACM0 but under some
other device node.

You'll need to look into the output of dmesg in order to see what's
happening (that's basically one of Linux' logs, and of specific interest
when hardware is not being detected properly). Also look into the output of
lsusb - if that doesn't even list your printer, there's no wonder it
doesn't get detected.