OctoPrint/OctoPrint #1191: Timelapse Progress

splttingatms, 2016-01-17:
Is there a way to see the progress or time left to finish encoding a timelapse? There is no way for me to see if a timelapse is being rendered or an error has occurred after a print finishes. Is it safe to start another print while a video is being rendered?

ntoff, 2016-01-21:
nope. As far as I'm aware once you start a new print it nukes the old timelapse files. There was a conversation about it here I believe. Someone correct me if I'm wrong?

Thus far I've resorted to logging in via putty and using the "top" command to watch the ffmpeg process and when it's no longer at the top I know the timelapse has finished rendering.

splttingatms, 2016-02-02:
Is the rendering progress something that can be exposed via an octoprint extension? Do extensions have access to external processes beyond octoprint?

foosel, 2016-02-02:
Duplicate of #485, update soon.

closed 2016-02-02 by foosel