OctoPrint/OctoPrint #1192: USB exteral storage.

Loup911, 2016-01-18:
Allow the use of a USB hard drive or USB key to having more space on the server.

It would be helpful to use a configuration page with drop down menu to select the location where the data is recorded. And display the space/free space for each device.

Like:

Gcode   SDCard 
STL       SDA2 8gb usb stick
MPG     SDA1 1TB hard disk
...

KeiroD, 2016-02-11:
Technically this is already possible. If your Raspberry Pi already recognizes the USB stick and 1TB hard drive, you **could** use the items you listed as symlinks inside the SD card location.

Example: create a symlink from in the USB stick and hard drive to a folder inside the SD card. Say... `usb` and `hdd` for example. Anything you place in here will be reachable for the Pi simply because of the symlinks.

markwal, 2016-02-12:
You can mount devices like that anywhere you want into your hierarchy.  Plus, you can link folders, but also: http://docs.octoprint.org/en/master/configuration/config_yaml.html#folder

I imagine it'd be possible to make a plugin that makes a UI like you suggest.  However, there are some security concerns for exposing a remote API that allows paths like that to be modified remotely.  Of course, remote access to the heaters on your printer is also a concern :-).

jeffeb3, 2016-02-22:
I did make the symlinks to a USB drive, and it's been working perfectly for a long time. I'm ad avid Linux user though, so it might not be easy for everyone.

I started by shutting down the octopi service, just to make sure it wasn't changing anything I was moving. Then I copied the folders onto the USB drive that I wanted to reside there. After I was sure they were there and correct, I deleted the ones on the sd card, and replaced them with symlinks (soft) to the USB drive, which I mounted as /data, via fstab.

The folders I moved were: 
.octoprint/logs
.octoprint/timelapse
.octoprint/uploads
.octoprint/watched

Then I restarted the raspberry pi. So far, it's been working great. I have a reasonably fast USB drive (3.0) and it's got 64GB of space.

FlameSoulis, 2019-10-13:
I know this is an old issue and I have no idea when it was added, but OctoPrint's Settings now has a section just for changing the folder locations. The only way I can see a plugin being made for this is if it addresses fstab management... but in all honesty, one could also just install usbmount and call it a day.

I guess it's up for debate if this issue was technically resolved or is not resolved until some kind of system can manage the auto mounting of the drives and keep the paths needed.

LazeMSS, 2021-03-31:
@FlameSoulis  agreed. This can be handled by mount points or symlinks a @jeffeb3  wrote.

closed 2021-03-31 by LazeMSS