OctoPrint/OctoPrint #1193: Change default listening port Haproxy

JosvdP, 2016-01-19:
I'd like to suggest a feature to integrate more with Haproxy to be able to change the port (straight from the Octoprint Setup) at which Octoprint and webcam are listening, so for example set it to port 6222 instead of 80. Perhaps this could be combined with feature request #640?

The reason is mostly to make port forwarding in my router easier (I don't want to forward port 80). However, it is also needed in the scenario where one has several OctoPi's and you want to choose which one to connect to over internet, simply by typing a port number.

**Need some help for the workaround**
At the moment, I think I could get a workaround going, but I need some help. The documentation of HAproxy is, well, detailed but complex. What I'm trying to do is seemingly simple but I can't figure it out, inarguably due to my lack of linux knowledge.

I've tried the following in the `/etc/haproxy/haproxy.cfg`:
```frontend public
        bind *:80
        bind 0.0.0.0:443 ssl crt /etc/ssl/snakeoil.pem
        option forwardfor except 127.0.0.1
        acl chosenport dst_port 6222
        use_backend webcam if { path_beg /webcam/ }
        use_backend octoprint_socket if { path_beg /sockjs/ }
        use_backend octoprint if chosenport```

I've also tried removing the `bind *:80` but to no avail.

Could someone help me get on my feet and point me in the right direction? I suppose my last option is to throw this on stackoverflow.

foosel, 2016-01-19:
> The reason is mostly to make port forwarding in my router easier (I don't want to forward port 80). However, it is also needed in the scenario where one has several OctoPi's and you want to choose which one to connect to over internet, simply by typing a port number.

I'm sorry, but that doesn't make sense. Port forwarding doesn't mean you can forward port x on host a to only port x on port b. A proper router should allow you to forward port x on host a to a completely different port y on host b. Same goes for a scenario where you do something like hostA:8080->hostB:80, hostA:8081->hostC:80, etc.

These are standard network administration tasks your router should allow you to do. If it doesn't, it doesn't justify being called a router and you should complain to the manufacturer. OctoPrint is not a replacement for a proper network infrastructure.

That being said, OctoPrint is supposed to stay platform agnostic. It usually it's run on a raspberry pi using haproxy as reverse proxy in front if it, but that being the most common configuration doesn't mean it's the only supported one. That means the no core code will ever be added that adds a dependency to something platform specific as haproxy. That is one of the reasons why there is a plugin interface. So if anyone wants to add a plugin for haproxy configuration to OctoPrint, be my guest, but it won't get added to the core application.

JosvdP, 2016-01-20:
I'm sorry, I'm stupid. You are of course right. However, yesterday my router behaved very strange since it didn't want to forward anything to port 80 from a different external port. When I tried again today it worked. I must have overlooked something.

My thought process was that if I wanted to test whether it was related to port 80 (since it's special...) I should change the port that Octoprint serves from. At that stage I struggled with changing the port and that's when I thought it would be good if you could change that from Octoprint.

Now, it's not completely strange to suggest this since most browser related GUI's have an option to change the port.

Never mind my request though, the only reason I can think of this being handy is when there are other programs running on the Pi that want to serve from port 80...

ntoff, 2016-01-22:
>Now, it's not completely strange to suggest this since most browser related GUI's have an option to change the port.

>Never mind my request though, the only reason I can think of this being handy is when there are other programs running on the Pi that want to serve from port 80...

That should really only be for changing octoprint's native port. Maybe that *should* be a core feature but in general changing a port post-install is silly because if anything else is running then they'll conflict when the server first starts up anyway and you'd probably end up with an error log like "cannot bind to port XXXX" and a server that's still not up and running so there's no way to reach the GUI anyway. 

In order to change it you should really change it during setup before the first run which would then require octoprint to have a fully fledged installer available for every kind of OS it runs on to change the command line automatically.

foosel, 2016-01-25:
In any case, it's nothing that will be implemented. If someone desperately needs a way to change the port of haproxy, I suggest looking into a plugin, but I'd seriously recommend against stuff like that. Changing the port something listens on shouldn't be something that needs to be done that often that you need a UI for that. If it is necessary to change that often, one should consider if that isn't due to the process being at fault.

closed 2016-01-25 by foosel

JosvdP, 2016-01-26:
You're right. I'm sorry for using your time! I'll be more considerate putting in a request next time.

foosel, 2016-01-26:
Please, no need for excuses! I guess I got a bit too invested as well considering that it isn't the first time someone made a feature request triggered by some issues with getting their router to do a proper port forward (as in, machineA:portX => machineB:portY) and it can be a bit frustrating "being made responsible" for network administration issues ;)

noisemaker00, 2016-11-22:
Hi, I have the same problem:
I use a Netgear DG834G which does not have the feature to change the port (PAT). I can only declare a service and the port it listening to, then forward traffic on a specific IP without specify on which port.

So please, can you tell me how can I use Octoprint from outside my LAN?

foosel, 2016-11-22:
The FAQ is your friend: https://github.com/foosel/OctoPrint/wiki/FAQ#i-need-to-change-the-port-octoprint-is-listening-on-because-i-want-to-do-a-port-forward-and-that-port-is-already-used-in-my-router

noisemaker00, 2016-11-22:
Thanks a lot for your feedback. I'll try asap. Another short question: is it possible to force HTTPS throught haproxy in that configuration file?

foosel, 2016-11-22:
haproxy on OctoPi is configured as an SSL endpoint. But it is using a self signed certificate (read: your browser will complain about it) and hence also not forcing SSL. You might be able to enforce HTTPS there, either by switching off port 80 altogether or by configuring a redirect from 80 to 443. I strongly suggest consulting the haproxy documentation for anything related to that config file.

For futher questions though, please consult the mailing list or the G+ commnunity (both linked from the homepage). This is a bug tracker, not a support forum :)

ctgreybeard, 2016-11-22:
FWIW I have haproxy set up with a Let's Encrypt (https://letsencrypt.org) certificate and it redirects to https automatically. If I recall, the instructions are on the Let's Encrypt site or were easy to find.

bzed, 2017-12-15:
If somebody is still interested in working on a general fix for this bug, using https://traefik.io/ might be an option. It is easier to configure it automatically and it comes with integrated full blown letsencrypt support. Especialyl interesting for those who want to deploy OctoPrint with docker (no, not me, I'm not insane ;))