OctoPrint/OctoPrint #1194: octoprint how to translate another language?

klwpcs, 2016-01-21:
when i  do this "python setup.py babel_new --locale=<locale>", it display "-bash: syntax error near unexpected token `newline' "
i read this document:https://github.com/foosel/OctoPrint/wiki/Translating-OctoPrint,it say;"From there you can edit your language's translation. If it's not yet available, you can create it via",how to do?

foosel, 2016-01-21:
Your command is incomplete. '--locale' expects a locale identifier after the '='. For example '--locale=de' for translating to German, or '--locale=es' for translating to Spanish. 

See [ISO639-1](https://en.m.wikipedia.org/wiki/List_of_ISO_639-1_codes) for a list of language codes.

closed 2016-01-21 by foosel

klwpcs, 2016-01-21:
thanks!i see...