OctoPrint/OctoPrint #1195: if I want to  Change "OctoPrint" on navbar?

maoshenglinTw, 2016-01-21:
I want to Cange "OctoPrint" be some cool name on navbar. Where is start point in source code.
1.src/octoprint/plugins/softwareupdate/__init__.py
2.src/octoprint/server/views.py
3.src/octoprint/static/js/app/viewmodels/appearance.js
I changed three files mentioned earlier,but it's doesn't work.

Can someone get me tips or help?

foosel, 2016-01-21:
Not a bug, not a feature request. Doesn't belong into the issue tracker.

`grep` for "OctoPrint", change the lines that need changing, remember to re-run `python setup.py install` (or `~/oprint/bin/python setup.py install` if you are running OctoPi) after that and clear your browser cache.

closed 2016-01-21 by foosel

ntoff, 2016-01-22:
@foosel 

>Not a bug, not a feature request. Doesn't belong into the issue tracker.

Perhaps it could be? I believe you can add the name of your printer up there, why not have that name simply replace the word Octoprint? That way you could put whatever you like up there. 

Just a thought.

foosel, 2016-01-22:
That bit of branding is intentional ;)

ntoff, 2016-01-22:
I figured it was