OctoPrint/OctoPrint #1196: (quirk) You can "upload" files while the server is offline

ntoff, 2016-01-22:
#### What were you doing?

Messing about, clicking random stuff to see what breaks.

#### What did you expect to happen?

Nothing (literally nothing, I expected it would do absolutely nothing)

#### What happened instead?

The "Upload locally / Upload to SD" option appears.

#### Branch & Commit or Version of OctoPrint

1.2.9 dev 35+g660de61

#### Printer model & used firmware incl. version

Not connected to one

#### Browser and Version of Browser, Operating System running Browser

Chrome Version 47.0.2526.111 m

#### Link to octoprint.log

None, the server is offline so this must be a javascript thing running in the local browser?

#### Link to contents of terminal tab or serial.log

N/A (not connected)

#### Link to contents of Javascript console in the browser

http://i.imgur.com/vpMoZJ0.png

#### Screenshot(s) showing the problem:

(see above screenshot for issue/java console combo shot)

I have read the FAQ.

foosel, 2016-01-22:
Good point, that should be caught on the client side. Currently it will show that thing as soon as you drag a file on the tab, that should not happen if the server is offline.

Thanks for the heads-up!

ntoff, 2016-01-22:
It also shows up if you simply highlight the text on that "server is offline" message window, and drag that around on the screen, it doesn't even have to be a file.

foosel, 2016-01-25:
Should be fixed on both ``devel`` and ``maintenance`` branch, fix will be released with version 1.2.9.

closed 2016-01-25 by foosel

foosel, 2016-02-03:
Had to revert the commit that fixed the text being draggable issue too, wasn't possible to get that to work cross browser, the API I used is only available in Chrome and apparently Opera so far, and in Firefox the drag'n'drop stopped working completely.

ntoff, 2016-02-04:
Probably not a huge issue anyway. One small request that I don't think is worth its own ticket, is it possible to filter the files that are viewable in the upload dialog box? Like with most programs you can filter out files they can't open, is it possible to have Octoprint's upload dialog only show STL and gcode files? Perhaps a configurable list of supported files?

foosel, 2016-02-09:
Will now only show files allowed for upload (that can be more than just GCODE and STL, depending on the installed plugins).

commit 814355e2 (2016-01-25): Do not show drag-n-drop overlay if server is offline

Fixes #1196