OctoPrint/OctoPrint #1197: How to trigger plugin when a print completes, as in an event?

chasecromwell, 2016-01-24:
I am working on a plugin that runs some python when a print completes. Normally, you would just add the path to the python in a event trigger as [the config documentation] says to. However, I want the plugin to be configurable in the normal settings menu, ex. Turn on and off different event triggers for the plugin. How would this work? (http://docs.octoprint.org/en/master/configuration/config_yaml.html#events)

markwal, 2016-01-24:
Since this is really a developer support question, you might want to try posting to the Google Plus community (in the contribution guidelines) or see if anybody is hanging around on the irc.

If you've got your plugin running, you can listen to events directly in your python: http://docs.octoprint.org/en/master/plugins/mixins.html#eventhandlerplugin

chasecromwell, 2016-01-24:
Thanks @markwal

closed 2016-01-24 by chasecromwell