OctoPrint/OctoPrint #1199: unable to perform last piece of code to run octoprint

murfmv, 2016-01-27:
FredMurphysMac:OctoPrint-master fredm$ /usr/bin/python   run.py
/usr/bin/python: can't open file 'run.py': [Errno 2] No such file or directory
Installing octoprint on MacOS 10.10.1 Great instructions so far, get this error.
Please help not very good with Unix

Fred

foosel, 2016-01-27:
Looks like the wiki page is outdated/contains a typo.

Try ``python run`` instead of ``python run.py`` if you followed that guide. It needs a desperate update however. Marking it as such.

closed 2016-01-27 by foosel

murfmv, 2016-01-28:
Good Day Gina,
I went on Github this morning and found your updated instructions.  Thank you so much for them.  I’m sure it took some research adjust Xcode and switch this virtual envelope.
I’d seen some of this trying to get bCNC to install. I’ll have to followup and read about the virtual envelope, so i can be more up to date.  I’d like to pass your instructions to the fellow at bCNC to help Mac users get through this python mess.
Question about Octoprint.  I’m have an Overlord Pro printer. Do you have to set this profile up or can I just copy the profile from Cura?  I can’t wait to read the documentation for Octoprint.

Thanks again!

Fred

Fred Murphy
Lamberts Cove LLC
murfmv@gmail.com
617-335-8610

murfmv, 2016-01-28:
Hello once again,
Get an error message when trying to install a new plugin that says need to run pip manually.  Anyway to get Octoprint and pip together.

Sorry for being a pain.

Fred

Fred Murphy
Lamberts Cove LLC
murfmv@gmail.com
617-335-8610

foosel, 2016-01-28:
First of all - please be aware that replying to this thread has your messages showing up on the Github issue page where you first opened your issue too. I took the liberty to remove your address information that was included in your email body, since you probably don't want that up for everyone to see.

Secondly, I'm sorry, but I cannot help there. You could try the mailinglist or the g+ community (linked by the bot above), which is where support questions such as this should go anyways, and where there might be people who have setup OctoPrint on a Mac in the past and feel more at home in that OS than me. Plus, there are more people than me capable of helping out as well in any case ;)