OctoPrint/OctoPrint #1201: [request] User print tracking and who's printing in status.

daned33, 2016-01-29:
Hi, 

Our hacklab uses octoprint for our 3d printers! And we'd like to have individual print usage (how long they've printed for and how much material, and what files they've uploaded)

and also on the status tab who's printing at the moment!

Would it be possible to implement a feature like this?

Thanks,

Dan

gddeen, 2016-02-08:
Dan, The ClientOpened and ClientClosed server events could be "hooked" and presented in a list on the screen showing the last N users and time. Someone would have to write a plugin to do it.

daned33, 2016-02-09:
Hmm, thanks gddeen.

Ill take a look at plugin development in that case if I get a chance.

foosel, 2017-01-27:
I think this is actually a duplicate or an extension of #652

closed 2017-01-27 by foosel