OctoPrint/OctoPrint #1202: Virtual Env Setup on Windows Page not working

thomasdunn, 2016-01-29:
I am on Windows 7.

Sorry I don't have better details, I lost my output from when it was not working last night.  But following the install directions for virtualenv 'pip install virtualenv' did not work (https://github.com/foosel/OctoPrint/wiki/Setup-on-Windows).  There seems to be a problem with the latest version.  I saw the output of your run of pip install virtualenv and it showed version 13.1.2 so I installed that explicitly with this:

pip install -Iv https://github.com/pypa/virtualenv/archive/13.1.2.tar.gz

After that it was working.

Octoprint looks awesome.  Thanks.

foosel, 2016-01-29:
If anything that's a problem with `virtualenv` and nothing I can fix.

However, I just tried installing the current `virtualenv` version under Win7 and it worked fine for me, so apparently whatever issue that was resolved itself.

closed 2016-01-29 by foosel

thomasdunn, 2016-01-29:
Ok, thanks.  Something wrong with my environment.

Thank you for Octoprint!