OctoPrint/OctoPrint #1203: [request] more options for print time estimation

ntoff, 2016-01-30:
It would be nice if we could add a fixed time to the print estimation, as well as a percentage. Since I know my printer takes let's say 20 minutes to warm up, it'd be nice if I could set that in an option somewhere to add to the print time estimation, as well as correcting it by a percentage if I notice it's slightly out. I know it will never be perfect.

sanchosk, 2016-12-12:
I would suggest a bit different approach.
When doing the calculation, I would consider "a printer multiplier". For first print, the multiplier would be 1.00. The time originally estimated would be stored and compared with the actual print time of first accomplished print. A division of those will set the multiplier.
Each and every next print would adjust the multiplier in some way (up or perhaps down), but from statistical perspective this would be one of the most relevant methods of time estimation.

foosel, 2018-05-17:
OctoPrint 1.3.9 will ship with two new hooks [`octoprint.filemanager.analysis.factory`](http://docs.octoprint.org/en/maintenance/plugins/hooks.html#octoprint-filemanager-analysis-factory) and [`octoprint.printer.estimation.factory`](http://docs.octoprint.org/en/maintenance/plugins/hooks.html#octoprint-printer-estimation-factory) that will allow plugins to override both the initial gcode analysis implementation as well as the live print time estimation. 

That should allow people to experiment with this stuff and find better but possibly more hardware/workflow/setup specific approaches that provide more accurate results than a stock option that has to *always* work can.

Knock yourselves out with this ;)