OctoPrint/OctoPrint #1206: Printer-Errors/unrecognized behavior seems to interrupt printer-conntection

Qixuga, 2016-02-01:
hi!

thank your for work and that great peace of software for 3d-printing community.
I like to get my multec multirap m420 with dual extruder, using a megatronic 3.0 (like Spiderbot). Actualy i don't know the system running at that controller but i will ask multec.

At the beginning i thought the printer works finde - connection establised, single commandes can be placed, printer reacts as he should.

but with the start of the first job, printing hangs during the initialisation sequence (homing, auto-bed-leveling). I thought some commands could be the problem but with a bit of testing it seems that the well know unregonized or wrong answers of the controller-sw seems to be the problem. connection hangs, print only could be aborded. a connection and job restart is needed.

may i be possible to integrate a funktion to activate a kind like a error ignoration for that case? octoprint need to go on with the connection.

I also used android app gCodePrint / same as Multec Printer App to print which has a log with the same entries. the app ignores it and print on if it happen. i connected the creator some days ago because i thought error-messages are not a goal of the printer but they told me to ignore it.
can octoprint it too?

Thank you!
Best regards from germany, "upper-hessian" ;-)

foosel, 2016-02-01:
Please provide example output from your printer's firmware (`serial.log`, is usually empty and needs to be enabled explicitly, details on how can be found in the contribution guidelines linked above). I'm not sure I understand correctly but responses from the printer are the problem, if it's something along the lines of "Error: Unknown command" (already fixed in the current OctoPrint version 1.2.8) or if it's something else.

Also please provide your exact version number of OctoPrint (lower left corner of the web site).

Essentially - please provide more information, the bot linked what you need to provide so this can be looked into. Currently we are flying blind here ;)

Greetings from near Frankfurt!

Qixuga, 2016-02-01:
I used OctoPi an updated OctoPrint to 1.28 Version (master branch).
i already found the log and activate it at first use - so i have all the data at serial.log ;)

But there are no error-entries - i think this could be the reason but i don't know this excactly.
how ca i send it to you? email from octoprint.org site?

i also send you a screenshot of the printer-app for android witch output error-messages at the same places. may be i need to increase the log level at octoprint?

thank you,
greetings back "to the south" ;)

foosel, 2016-02-01:
`serial.log` should contain the full communication with your printer. Whatever makes it hiccup will be in there. But if the messages you are suspecting to be the problem aren't in there, then that's very odd and I'm not fully understanding the problem at all.

You can attach screenshots and .txt files to the issues here (see the line "Attach files by dragging & dropping..." just below the input field). You'll need to rename `serial.log` to e.g. `serial.log.txt` to be allowed to upload it. Same with `octoprint.log` (please also provide that).

Qixuga, 2016-02-01:
Here i comes - the serial.log. I checkt it and think there is no private content, isn't it?
[serial.log.txt](https://github.com/foosel/OctoPrint/files/112656/serial.log.txt)

Hope it helps to get the problem and another printer to work.
Thank you!

foosel, 2016-02-01:
Nah, it's really just the serial communication.

The only thing I see in there are a couple of line number mismatches, since the initial `M110` (reset line numbers) gets overtaken by other commands. I'll need to look into preventing that, but it's nothing critical so far. You could prevent that from happening by enabling the "Send a checksum with **every** command" option in the settings:

[image]

Qixuga, 2016-02-01:
OK and thank you!
I activate it but thats - if i understand you correctly - only one part of the solution.
may be after that if i send G28 (home all axis) after connection to the printer manualy he will homing his axis but also communication will be cut off. only reconnection helps. no usefull error-message found :-(
I don't know how the app solve the problem but i ask the developer cause he work on API-integration of your great printer host tool octoprint (its experimental activateable in app settings selectable as printer-target).

foosel, 2016-02-03:
I think I just spotted it, your firmware is sending garbage:

```
2016-02-01 11:30:39,063 - SERIAL - DEBUG - Send: G28
2016-02-01 11:30:52,146 - SERIAL - DEBUG - Recv: 
2016-02-01 11:30:52,154 - SERIAL - DEBUG - Recv: bedtemp: 28.53 : 0.014ok
```

That tiny `ok` there should be on its own line, not stuck to the end of the `bedtemp` line. Instead of disconnecting/reconnecting, you can use the "Fake Acknowledgement" button in the advanced options on the terminal tab:

[image]

You could also try if removing `G28` from the long running commands would get things going again:

[image]

For long running commands OctoPrint doesn't trigger an automatic "fake acknowledgement" if the communication looks dead, which in this case it does due to the swallowed ok, so removing `G28` from that list might be a workaround until your vendor releases a fixed firmware (contact them about that! Send them the above log excerpt and tell them that their printer is missing a newline there and hence doesn't confirm to the protocol).

In any case, if any of the above solves your issue, what you are seeing is definitely an issue caused by your printer not behaving properly, and I'm not going to fix that in OctoPrint - sorry, but I can't adapt to every firmware bug on the planet, for that there's too many firmware derivatives out there.

foosel, 2016-02-15:
@Qixuga any update on this?

Qixuga, 2016-02-15:
Hi, thank you for your reminder ;) ... I was busy last few days so i check it today and it work fine with the workaround! 
Thank you very much!
I gave your hint with this link to the developer.
I hope they will fix it with next release.
I'm also active in it-business and they gave me some well known answers, you know ;) But they did'nt deny it.

But with removing the "move to origin - g28" out of the long running cmd the printer works fine.

i will hope and wait for a firmware-fix and shure that not a fix in octoprint is needed - that was'n my intention. they work on some new features which will be released soon (beta-tests currently running)
But without your support i did'nt find any lead why that does'nt seems to work!

You found is!
I have to put you an nice coffee at me good friend from your local "gelato dealer" dolomiti! :-D

I will update the task if the printer developer fix the error for all other multec-users.
I Ask about the software: the M420 uses a fork of marlin-firmware on megatronics 3.0 control board.
I post also this hint to some bbs's - hope all other will help it too!

foosel, 2016-02-15:
Ok... In that case I hope you don't mind if I close this now since it certainly it is not verified that the behaviour is caused by your printer's firmware and not a bug in OctoPrint.

You can still comment on this issue, so don't worry about, it's only marked as closed so that I don't drown in open tickets ;)

closed 2016-02-15 by foosel

gretel, 2017-01-10:
my *Multec M420* with *MultiSense* option is now levelling (on `G29`) and printing fine using OctoPrint since `G28` got removed from the list as advised here.
additionally, enabling checksums for all commands did not hurt. thanks a lot you made my day. ❤️ 
going to get in touch with Multec support and see if they care about the firmware.

foosel, 2017-01-10:
@gretel I wish you luck with that one. It's a bit sad how many vendors don't bother fixing their firmware bugs.