OctoPrint/OctoPrint #1207: [request] Controls hiding per access level (admin/user)

thantik, 2016-02-01:
This one was suggested by some of the schools that DeltaMaker has their printers at so I'm posting this feature request on their behalf because I don't believe we're able to do such a thing as it currently stands.

Right now, the DeltaMaker ships with OctoPrint 1.2.8 -- We add two sections to the 'controls' section of the config.yaml -- both simply contain buttons with shortcuts to commonly used gcode commands and some customized marlin gcodes for setting up the printer long-term.

Occasionally we need the "advanced" commands to be accessed - right now we're doing some config.yaml swapping, and rebooting the machine to grant/revoke access.  It's clutzy.

I'm requesting a controls 'access level', so that additional buttons can be tied to an admin/user login, maybe it could be just a flag that we could set that's optional?  We'd like to be able to have the schools provide a login for the students, but also an alternative login for the teachers should we need to give them instructions for some advanced calibration procedures - things which we've found when the students have access to, they have a tendency to press and mess up the machine for others.

foosel, 2016-02-01:
Sounds a lot like it should already be covered by #1110 , or something that should be added to that TODO list instead of having a different ticket for it. Yay or nay?

thantik, 2016-02-02:
Yes, it seems so.  Sorry for not catching #1110; we don't really see a need for a granular permission system - just some things to use the current permissions system on. :)

foosel, 2017-02-13:
I just noticed that I forgot to close this as a duplicate a year ago 😱

closed 2017-02-13 by foosel