OctoPrint/OctoPrint #1213: [Evol?] Bloc video if user not connected.

seishiro63, 2016-02-08:
Hello,

I try to find a plugin / mode to block the video on the website if no one is connected.
I didn't find this option in the config pannel.

Did I miss something (allready exist somewhere)?
Is it possible to implement something like that?

Thank you.

foosel, 2016-02-08:
@seishiro63 could you please clarify what you mean with "block the video on the website if no one is connected"? If no one is connected, nothing is displaying the video, so... huh? :)

seishiro63, 2016-02-09:
I added a screenshot.
You can see that no one is connected. and we can see the video brodcasted from my usb webcam connected to the Raspberry.

[image]

foosel, 2016-02-09:
Yeah... well.. that's a duplicate of #1068 then and as stated in the FAQ a **proper** solution is [technically impossible](https://github.com/foosel/OctoPrint/wiki/FAQ#why-is-there-no-access-control-for-the-webcam-in-octoprint).

foosel, 2016-02-09:
And for clarification: someone is connected in the above screenshot. You. Doesn't matter if you are logged in or not, there is a connection active between your browser and the server. Hence the initial confusion about what you were even asking.

closed 2016-02-09 by foosel

seishiro63, 2016-02-09:
Ok
Thank you for clarification. 
I'll move my printer to my living room to my garage