OctoPrint/OctoPrint #1214: Sometimes when starting a print, the bed heater doesn't go on.

nixomose, 2016-02-10:
I haven't found a pattern yet, although it never happens on a first print or just after a reboot of the pi.
But sometime when I go to start a print, it will not attempt to raise the bed temperature of the printer, only the extruder, and then start printing.
Not a crisis, I now know to check for this when I start a print, so I restart octopi and then it works. I was just wondering if anybody else had seen such a thing. Perhaps this isn't the best forum for this but I wasn't sure where to post it.

nixomose, 2016-02-10:
#### What were you doing?

I tried to start a print by hitting the big green print button.
Sometimes it happens when I click "load and print" sometimes I need only hit the print button.

I've got a monoprice IIIP printer. I left the defaults for everything and it works well enough.

#### What did you expect to happen?

I expected the bed temp to go up, then the extruder temp to go up. 

#### What happened instead?

Only the extruder temp went up. Now that I think about it, the bed temp is probably one of the first commands to be sent, so perhaps there's some communication problem at the very beginning of the conversation with the printer. I'll check the terminal next time it happens.

#### Branch & Commit or Version of OctoPrint
Version: 1.2.8 (master branch)

#### Printer model & used firmware incl. version
monoprice iiip

#### Browser and Version of Browser, Operating System running Browser

chrome and firefox, not a browser problem.

#### Link to octoprint.log

I don't have a log of a failure, I'll get one next time it happens.

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s) showing the problem:

Salandora, 2016-02-10:
Sounds more like a problem with your start code in your slicer, what slicer do you use, can you please post your start code?

nixomose, 2016-02-10:
I just looked, it's the very first thing in the file, so that must be it.
I use cura.

M190 S50.000000
M109 S195.000000
;Sliced at: Sat 06-02-2016 18:27:59
;Basic settings: Layer height: 0.1 Walls: 1 Fill: 20
;Print time: 4 hours 51 minutes
;Filament used: 7.988m 23.0g
;Filament cost: None
;M190 S50 ;Uncomment to add your own bed temperature line
;M109 S195 ;Uncomment to add your own temperature line
G21        ;metric values
G90        ;absolute positioning
M82        ;set extruder to absolute mode
M107       ;start with the fan off
G28 X0 Y0  ;move X/Y to min endstops
G28 Z0     ;move Z to min endstops
G1 Z15.0 F4800 ;move the platform down 15mm
G92 E0                  ;zero the extruded length
G1 F200 E3              ;extrude 3mm of feed stock
G92 E0                  ;zero the extruded length again
G1 F4800
;Put printing message on LCD screen
M117 Printing...

foosel, 2016-02-10:
Can you verify that the first line is being sent to the printer? (Terminal tab)

nixomose, 2016-02-10:
Just happened now, here's the terminal...

Connecting to: /dev/ttyUSB2
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x70131730, open=True>(port='/dev/ttyUSB2', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: M110
Recv: start
Send: M110
Recv: Info: 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000
Recv: Free RAM:10732
Recv: ok
Changing monitoring state from 'Connecting' to 'Operational'
Send: M21
Recv: ok 0
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Send: M105
Recv: ok 0
Recv: T:23.49 /0 B:23.76 /0 B@:0 @:0
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Changing monitoring state from 'Operational' to 'Printing'
Recv: wait
Send: N0 M190 S50.000000*115
Recv: skip 0
Recv: ok
Send: N1 M109 S195.000000*74
Recv: ok 1
Send: N2 G21*24
Recv: TargetExtr0:195
Send: N0 M110 N0*125
Recv: T:23.49 /195 B:23.56 /0 B@:0 @:255
Recv: T:23.49 /195 B:23.56 /0 B@:0 @:255
Recv: T:23.76 /195 B:23.56 /0 B@:0 @:255
Recv: T:24.16 /195 B:23.56 /0 B@:0 @:255
Recv: T:24.83 /195 B:23.56 /0 B@:0 @:255
Recv: T:25.91 /195 B:23.56 /0 B@:0 @:255
Recv: T:26.98 /195 B:23.56 /0 B@:0 @:255

nixomose, 2016-02-10:
I just restarted and it worked....

Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Changing monitoring state from 'Operational' to 'Printing'
Send: N0 M110 N0*125
Recv: ok
Send: N1 M190 S50.000000*114
Recv: ok 1
Send: N2 M109 S195.000000*73
Recv: TargetBed:50
Recv: T:154.43 /0 B:24.03 /50 B@:255 @:0
Recv: T:153.74 /0 B:24.30 /50 B@:255 @:0
Recv: T:153.04 /0 B:24.50 /50 B@:255 @:0
Recv: T:152.35 /0 B:24.63 /50 B@:255 @:0
Recv: T:151.65 /0 B:24.83 /50 B@:255 @:0
Recv: T:151.04 /0 B:25.17 /50 B@:255 @:0
Recv: T:150.26 /0 B:25.44 /50 B@:255 @:0
Recv: T:149.64 /0 B:25.70 /50 B@:255 @:0
Recv: T:149.14 /0 B:25.97 /50 B@:255 @:0

So the one that doesn't work says this:
Send: N0 M190 S50.000000115
Recv: skip 0
Recv: ok

and the one that does work...

Send: N1 M190 S50.000000*114
Recv: ok 1

I don't know the protocol for the printer, can you decipher this?
It looks like the skip zero says the printer didn't ACK that command? whereas the "ok 1" means it did?
Is that the printer?
Is there some way I can force shove in some kind of no-op before the print starts to kind of kick it?

foosel, 2016-02-10:
I have never seen that "skip" stuff before, but that certainly looks like your printer simply decides to ignore the first command that is sent. Nothing OctoPrint can do about that. If I were you I'd talk to the manufacturer of my firmware and ask what that is supposed to accomplish, certainly doesn't conform to anything "officially defined", especially considering that the command is acknowledged ("ok"), just apparently not executed (wtf?!).

You could try adjusting your slicer to have your start GCODE prefixed with something like 5 or so M105s... there is no real no-op command, but M105 only queries the temperature so it shouldn't cause any issues, maybe that will cause one of the M105s to be ignored instead of a real command.

In any case, not a bug in OctoPrint but a firmware issue, so closing this.

closed 2016-02-10 by foosel

nixomose, 2016-02-10:
I appreciate the feedback, I just don't have experience with the printer protocol, thanks for explaining that.

nixomose, 2016-02-13:
Just another note on this in case somebody else comes across this problem.
I think I'm starting to see the pattern. It has something to do with the order that the bed and extruder are heated up in relative to the last time a temperature command was sent to it.
I don't get the details, but it has something to do with that.
The gcode cura generates always heats the bed first then the extruder, and that works fine. But once or twice I've poked the temperature buttons in octoprint and that changes the order or the last-thing-touched state in the printer. This is all a guess but I have a feeling it has something to do with it.
I think skip response comes out when the printer is expecting a temp command for one of the parts and gets the other one so it ignores the one it doesn't want.
Something like that anyway.

nixomose, 2018-05-30:
No root cause, but I have a crappy workaround:

I added this to my "After connection to printer is established" setting:

; turn on hotend
M104 S190.000000
; turn on bed
M140 S60

so it heats up when the printer starts and then everything stays happy.