OctoPrint/OctoPrint #1216: Print from printer's SD card

SFrav, 2016-02-10:
In the interests of print quality I don't want the gcode data transferred over USB, but rather from the SD card in the printer. 

I would expect when choosing to print a file that is on the printer's SD card that the print would continue even if the Octopi was unplugged. Testing this out, this is not the case.

Is there a way to ensure the print is being transferred from the printer's internal SD card?

closed 2016-02-10 by SFrav

nophead, 2016-02-10:
Whether it continues or not when you unplug the USB depends on whether your controller resets or not. Some controllers like Melzi have an auto reset jumper that needs to be removed for stand alone operation and replaced for firmware updates.

foosel, 2016-02-10:
What @nophead said - sadly nothing that OctoPrint can solve. Also no way to detect it usually, so not even a chance to warn the user to not connect to or disconnect from the printer while an SD print is ongoing.

SFrav, 2016-02-10:
Thanks for the replies. 
So when I do print a file on the 3d printer, the gcode is indeed transferred from the printer's sd card? That is rather than sent to the raspberry pi and then through the USB.

foosel, 2016-02-10:
If you choose a file on the printer's SD card, it is *already there*. Nothing to transfer. The printer will print it from the card. But the controller board might still reset if a USB connect or disconnect happens, so it depends on your printer electronics if you can disconnect after starting such a print or not. 

If on the other hand you print a local file, that is *not* sent to the printer's SD card first but streamed from your Pi to the printer via USB. If you unplug the cable or disconnect then, naturally the print will always abort since it is coming from OctoPrint as a series of commands.

foosel, 2016-02-10:
Argh... locking this since having that discussion ongoing at two places concurrently is not going to work ;) Discussion continues in the [G+ thread](https://plus.google.com/u/0/108503381392666720549/posts/ijBezZ2x8od)