OctoPrint/OctoPrint #1217: [request] Block print button on update process

BigAl42, 2016-02-10:
Today my octoprint informs me about the update to 1.2.9. I clicked ok and octoprint begins to update in the background. ( -> 1.2.9 master branch)
I know that octorpint has to restart after the update was successfully installed. Nevertheless, I started a print and after some minutes my printhead stops because octoprint was rebooting after finishing the update process. 

So I would recommend to block the print button while the update process is operating or until reboot.

ntoff, 2017-07-17:
@foosel Isn't this issue solved now via the popup update progress dialog which blocks all functionality during the update?

foosel, 2018-03-09:
If you are only a user (not an admin) and the admin has update checks disabled for users, that dialog won't necessarily show up I think. I've just made it a bit more likely to show up, but this still isn't solved I fear.

foosel, 2018-06-15:
The above commit makes it so that admins AND users will receive a blocking modal while an update is in progress. That not only blocks the print button but pretty much the whole UI. 

Commit is available on the `maintenance` branch, soon `devel` and to be released as part of 1.3.9.

foosel, 2018-07-25:
1.3.9 has just been released.

closed 2018-07-25 by foosel