OctoPrint/OctoPrint #1218: [Issue] - Disconnect issue after 1.2.9  update

kuntaykunt, 2016-02-11:
Hello ,

I got M3D printer and i was using 1.2.8 version yesterday.I have just updated to 1.2.9 today but i can not stay longer then 5 min. on the localhost which was not a case in 1.2.8 version.

I wanted to print a model from office , i tried several time to slice but site pop-ups a warning saying **server disconnected and reload the page**.

[octoprint.txt](https://github.com/foosel/OctoPrint/files/126019/octoprint.txt)

My printer website : [print.mypi.co](url)

Log file is attached. I hope , i did it right.

Thank you.

foosel, 2016-02-11:
Can you please first make absolutely certain that this is not caused by some network issue by connecting **locally** (not via the internet)? 

Judging from the log file you provided (thanks!) your client disconnects constantly on its own. It is impossible to diagnose if a shaky internet connection in the middle can cause it just as easily as some incompatibility introduced through the update.

Also, please test without any third party plugins being enabled (I see a couple of errors in the logs originating from plugins).

bengizmo, 2016-02-12:
I had the same issue today as well. I have successfully been printing on my printer remotely in 1.2.8, and I upgraded to 1.2.9 today. I can still access the Octoprint web page, but many of the features are laggy or do not work. My home router is setup to forward port 5050 to port 80 of the printer. 

My next step was - while still remote - to setup a SOCKS proxy through my home router and access the printer through port 80 using it's private IP address (192.168.x.x). This allows full functionality.

Perhaps it has something to do with the web server settings not being able to handle the NAT situation?

foosel, 2016-02-12:
Please, anyone encountering this, provide a log file, and also the contents
of the JavaScript console, and test in the local network as well. Without
that, there is no way to diagnose this.

There's a reason that kind of information is asked for in the "how to file
a bug report" section. Without it, it's like you are calling someone for
help when your car breaks down, say "I'm on route 66" and then hang up.

"Many of the features are laggy or don't work" - which features
specifically? Examples please.

I'm guessing it has something to do with the new connection throttling
(maybe when you are connecting through forwarded ports somehow you are not
on a proper web socket connection but a polling fallback and that causes
this) but I need more details to confirm and come up with a solution. A
couple of screenshots from the network tab in the browser developer tools
might also help.

kuntaykunt, 2016-02-12:
Hello ,

I've tested in local network. Just working fine. I uploaded the model and sliced it. I monitored from [octopi.local](url). There was not any disconnect message on the [octopi.local](url)

The problem is about outside of the localhost. I know ddns working fine. I have fixed ip adress as well.

The website connecting the webserver however it is not staying connected.Error message is attached. 

Logs are attached.

Note : I did some homemade display thing on PCD 8544 Nokia screen. 
I used [https://github.com/exilaus/simpleoctodisplay](url) source to do so. Basically i have **wiringPi** ,** jq** , **PCD8544** libraries to do some physical display. I think it is nothing to do with the network. I am monitoring some values from my desk which is a useful thing. I used **crontab** on reboot to run a script which run another script every 0.9 seconds to refresh screen. I attached the picture.

[image]
[image]

[octoprint_log.txt](https://github.com/foosel/OctoPrint/files/127619/octoprint_log.txt)
[plugin_printhistory_console_log.txt](https://github.com/foosel/OctoPrint/files/127621/plugin_printhistory_console_log.txt)
[octoprint_stats_db.txt](https://github.com/foosel/OctoPrint/files/127620/octoprint_stats_db.txt)

foosel, 2016-02-12:
@kuntaykunt 

> Please, anyone encountering this, provide a log file, **and also the contents of the JavaScript console**, and test in the local network as well. Without that, there is no way to diagnose this.
> [...]
> A couple of screenshots from the network tab in the browser developer tools might also help.

Any chances to get that kind of information? :)

MrFredH, 2016-02-12:
I can confirm that this issue exists on my setup also.

I’m running standard with all updates.

(1.2.9) I have a PS3 EyeToy camera, octoprint frequently runs at high CPU load as does mjpg_streamer.

Static IP direct local lan access

I’m using the chrome browser on Windows 7 and Android Tablet. Same behaviour both devices.

It is not internet exposed.

 

# top

  PID USER      PR  NI  VIRT  RES  SHR S  %CPU %MEM    TIME+  COMMAND

25350 pi        20   0  211m  31m 5560 S  48.7  7.2 277:51.79 octoprint

2801 pi        20   0 39756 4888 3824 S  47.8  1.1   6288:09 mjpg_streamer

 

# uptime

07:48:47 up 5 days,  4:50,  1 user,  load average: 3.82, 3.14, 2.58

 

# uname -a

Linux octopi 4.1.17+ #834 Mon Feb 1 15:13:47 GMT 2016 armv6l GNU/Linux

 

# ifconfig

eth0      Link encap:Ethernet  HWaddr b8:27:eb:27:51:fc

          inet addr:192.168.1.21  Bcast:192.168.1.255  Mask:255.255.255.0

          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1

          RX packets:1896932 errors:0 dropped:7717 overruns:0 frame:0

          TX packets:1951804 errors:0 dropped:0 overruns:0 carrier:0

          collisions:0 txqueuelen:1000

          RX bytes:185747767 (177.1 MiB)  TX bytes:1180910656 (1.0 GiB)

 

Please let me know if you need more information from my install.

Kind Regards

 

Fred Houweling

I used https://github.com/exilaus/simpleoctodisplay source to do so. Basically i have wiringPi ,** jq** , PCD8544 libraries to do some physical display. I think it is nothing to do with the network. I am monitoring some values from my desk which is a useful thing. I used crontab on reboot to run a script which run another script every 0.9 seconds to refresh screen. I attached the picture.

 <https://cloud.githubusercontent.com/assets/6482979/13001118/55ed391e-d16b-11e5-9acc-49f6fa2bded7.JPG> img_1640
 <https://cloud.githubusercontent.com/assets/6482979/13001120/58afdbac-d16b-11e5-8145-49ca2291785f.jpg> error_message

foosel, 2016-02-12:
Yes I do need more information than this, as repeatedly pointed out above :P Please please please could someone *finally* provide me the output on their Javascript console?

MrFredH, 2016-02-12:
Java script console output attached.

 

 

This is the Network tab “Copy all as HAR“ output.

{

  "log": {

    "version": "1.2",

    "creator": {

      "name": "WebInspector",

      "version": "537.36"

    },

    "pages": [

      {

        "startedDateTime": "2016-02-12T08:23:07.708Z",

        "id": "page_1",

        "title": "http://192.168.1.21/",

        "pageTimings": {

          "onContentLoad": 8369.594999938272,

          "onLoad": 8767.121999990195

        }

      }

    ],

    "entries": [

      {

        "startedDateTime": "2016-02-12T08:23:07.708Z",

        "time": 2300.8099999278784,

        "request": {

          "method": "GET",

          "url": "http://192.168.1.21/",

          "httpVersion": "HTTP/1.1",

          "headers": [

            {

              "name": "If-None-Match",

              "value": "\"7af6c7d99a49017de0af5992a617789d64e89de9\""

            },

[... 3684 lines cut ...]
        "timings": {

          "blocked": 0.752000021748245,

          "dns": -1,

          "connect": -1,

          "send": 0.15400000847876105,

          "wait": 2714.360999991183,

          "receive": 89.06300005037656,

          "ssl": -1

        },

        "connection": "216591",

        "pageref": "page_1"

      },

      {

foosel, 2016-02-12:
Correction on the above: Please start OctoPrint with ``--debug``, THEN reload the browser, reproduce the issue and then provide the output on the Javascript console. What I'm hoping for is output like this:

[image]

that will help figuring out via which transport method your browser is connected to the server and what the adaptive rate limiting is doing.

foosel, 2016-02-12:
@MrFredH thanks, but two things. 

First of all, please paste large output like that into pastebin.com or gist.github.com and only paste the LINK into your reply, never the full output! That dump of your network tab just made the issue page on Github at https://github.com/foosel/OctoPrint/issues/1218 (which you are posting to by replying by mail like you are doing) pretty much unusable ;) 

Secondly, that was the network tab. That is interesting too, but what I need is the javascript console (and in debug mode on top of that, which is what I forgot earlier, see above for what to do) and on top of that your Network tab dump is truncated and stops right when it comes to the interesting bits :(

So, in conclusion (tasks for you @MrFredH, @kuntaykunt and @bengizmo):

Please start OctoPrint in debug mode, on OctoPi this can be done by doing THIS on a console (ssh into the machine for that):

  * `sudo service octoprint stop`
  * `/home/pi/oprint/bin/octoprint --debug`

Leave the console open while the server is running, if you close it the server will stop! Try to reproduce the issue. Once you do provide the contents of the Javascript console (see above for an example and what we are looking for specifically here).

After you are done you can stop the server in the console by pressing "Ctrl+C" and restart it "properly" with `sudo service octoprint start`.

MrFredH, 2016-02-12:
@foosel Thanks for the advice, Apologies for the excess of information.
Please find attached a console log.
[192.168.1.21-1455270935468.log.TXT](https://github.com/foosel/OctoPrint/files/127781/192.168.1.21-1455270935468.log.TXT)

foosel, 2016-02-12:
@MrFredH Thank you very much, that's what I've been looking for! :) Was this from when the UI was misbehaving? Because the only thing I can so far spot is

```
WebSocket connection to 'ws://192.168.1.21/sockjs/466/5m8xiqzk/websocket' failed: WebSocket is closed before the connection is established.
Closed transport: websocket SimpleEvent(type=close, code=2007, reason=Transport timeouted, wasClean=false)
Opening transport: xhr-streaming  url:http://192.168.1.21/sockjs/466/cout1a35  RTO:11130
```

which at least tells us that it's falling back to a slower streaming method, but it should still work fine with that (and indeed does in my tests - I just set up some port forwards just to test that scenario, forced fallback situations to a slower transport, and I cannot reproduce anything "odd").

Can you and @kuntaykunt  and @bengizmo (from you two I'd still need the above additional information too) please describe the symptoms you are seeing in detail? "Offline" dialog popping up more than before? Terminal scrolling slower? Button clicks not immediately leading to a change in the UI? Also some details on your network setup please... WIFI or LAN, router hops in between, reverse proxies besides the one installed on OctoPi etc.

Since I cannot reproduce anything that feels wrong so far, I'm a bit at a loss as to what I'm actually looking for and how to potentially reproduce it, so *anything* might help here.

Considering that there's at least three of you that run into problems with the current release, there probably is a bug, and probably that one is even easy to fix, but I need to somehow narrow it down first, so, type ahead :)

MrFredH, 2016-02-12:
Do you have the same plugins?

foosel, 2016-02-12:
Most of them by chance, but let's reverse the question: Does the problem (which problem?) manifest if you disable all plugins that aren't bundled with OctoPrint?

foosel, 2016-02-12:
Also, just for the record, I also now tried what would happen if I accessed the internal OctoPrint instance, through the internet via a forwarded port, via GPRS from my mobile. No disconnects, no other issues that make it feel "laggy". So - whatever it is, it is not inherent to port forward setups or slow connections.

Checklist for anyone running into "laggyness" on 1.2.9, whether they are accessing OctoPrint from the internet or the local LAN:

  * Are the issues present on 1.2.8 as well?
  * What are the exact issues you are experiencing that weren't present with 1.2.8? Disconnect dialogs, button click response times, refresh rates etc. Be as precise as possible here every tiny detail matters - so far we only have "laggyness" and "me too" in this thread, this is not very helpful sadly, so please explain exactly what specifically is not behaving as expected ("when I do X I expect Y to happen, but Z happens instead"). Making a video of the symptoms and uploading the to YouTube would be great!
  * If you are accessing it from the internet, does the issue go away when you access OctoPrint through your LAN?
  * If you have any plugins installed that are not bundled with OctoPrint, does the issue still manifest if you disable them (don't forget a restart & cache-clearing reload in the browser when testing this)?
  * How are you connecting to OctoPrint and from which connection, device and browser? Through WiFi or wired LAN or maybe a mobile network? Are there port forwards or reverse proxies involved? If so, where, which and how are they setup? Where is OctoPrint installed?
  * Running OctoPrint in debug mode (`--debug` parameter on start, see above for how to do that on OctoPi), reproduce the issue and then provide the full output in the Javascript console *and* your `octoprint.log` on pastebin.com or gist.github.com and link it here. This is very very important since it would provide information whether the only likely candidate among the changes between 1.2.8 and 1.2.9 to cause this is indeed at fault or not.

I want to solve this issue if it's caused by something in OctoPrint, but I cannot reproduce it myself, so I need as much information as possible on how it manifests for you. 

So please, everyone, provide ALL the above information. As precisely as possible. Thanks!

MrFredH, 2016-02-12:
Are the issues present on 1.2.8 as well? Yes
Connect to server, initial interface loads, then the timeout dialog as others have provided is displayed. it eventually goes away and normal operation resumes.
Issue present on local lan access.
My browser is not configured to use a proxy, direct local subnet connection to raw ip address over a LAN interface same network switch (Auto Gigabit Switch).
I disabled a majority of my plugins and it continued to do this, I will do some more testing over the next day or so and report back.

.

foosel, 2016-02-12:
> Are the issues present on 1.2.8 as well? Yes

Well... then you probably have an entirely different problem than the other two people here which reported problems after they updated to 1.2.9. Different issue on your side then, wrong ticket :)

> Connect to server, initial interface loads, then the timeout dialog as others have provided is displayed. it eventually goes away and normal operation resumes.

Does it return or does it stay gone then? If the latter, it sounds like you are seeing the initial disconnect dialog shorty because of your browser not being able to use the websocket connection and falling back to the xhr-streaming alternative (the log line i pointed out a bit up). That would be something I'd consider normal given the situation - why your browser is not able to establish a websocket connection would be a different question then, but that is most certainly an issue with your environment then and not with OctoPrint itself and so rather a case for a support thread on the mailing list or in the G+ community and not the bug tracker.

foosel, 2016-02-12:
@kuntaykunt Just noticed the link to your instance in your first post (sorry, that hid itself well for some reason). Currently have it open in a browser tab. So far no disconnects, no issues at all (apart from the webcam not working, not sure if that's intentional or not). Stable connection, log output working, perfect operation...

kuntaykunt, 2016-02-13:
@foosel I noticed that the website is running well on my phone , too (Iphone 6). However at my office as i mentioned before i can not stay more then 5 minutes. 

I have faced some printing & plugins problems when i started to work on a model. I can not even login now so will do fresh set-up again.

I will let you know if the clean installation solved the problem or not.

(For information : I counted the disconnect array. I did nothing , not logged in and watched the issue.I clicked on my website then after approx 3min. disconnected waited like 3 min. again disconnected again.I dont know if this helps.)

foosel, 2016-02-13:
I had your page open on my regular desktop browser for a couple of hours
yesterday, with the js console open and forced into debug logging. No
disconnect at all, not even any suspicious entries in the log. Of course,
no print was running and I wasn't logged in, so take that with a grain of
salt.

kuntaykunt, 2016-02-15:
@foosel After clean installation i have tested witm my iphone and 2 other firend on pcs which have active user rights. The test have done all good. There was not any problem. I think it was the office network causing the issue that time. Otherwise we do not know what is the issue but everything is ok about network and porting now. Thank you for your support and efforts.

foosel, 2016-02-15:
@kuntaykunt thanks for reporting back and glad to here that in your case my initial hunch was correct and it was the network which was to blame.

Now what about you @bengizmo?

foosel, 2016-03-07:
@bengizmo still waiting for info here.

foosel, 2016-03-10:
Considering this not a bug then since everyone who reported back (incl. OP) either had a completely different issue or the problems were caused by network connectivity troubles.

closed 2016-03-10 by foosel