OctoPrint/OctoPrint #1221: Octoprint should function standalone without relying on external javascript references

jamesarm97, 2016-02-13:
If octoprint is not on the network (which should work for when octoprint is being used to control a printer which may be taken somewhere but not necessarily on a network. Octoprint can already have sliced files or STL's stored locally and should be able to run and print). What I am seeing is some of octoprint loads correctly but it seems to be trying to load javascript libraries from the internet which fails with an error of something like packed_app.js error 401 not authorized (when it tries to load external js resources). I think there is a way to load external js libraries but if it fails try and load them from local storage and let the user choose to download the libraries or have a script / setup button that will download and save them on disk for later use. That way there will be a way to always get the latest js versions referenced by octoprint and store them for offline use.

markwal, 2016-02-13:
Huh.  My browser console doesn't show any external references at all.  It does some requests from the client browser back to pi (same place the page came from) including for packed_app.js.  Are you sure you're seeing something that requires internet access?  If so, you'll probably want to include logs/traces from the browser console that show what you mean.

Also, while you need connectivity from your PC to the Pi to see the UI, it'll happily continue the print without that.

I think there may be an issue here and that is: why is the request for the js unauthorized?  Does authentication cookie expire at some point?  Or some other problem that causes that 401.

jamesarm97, 2016-02-13:
I'll see what I can get off the browser debug console (where I saw the initial errors). I understand it will print without a PC and Internet but since I operate octoprint from a local touchscreen and use that to control the printer (homing, preheat, etc), it makes it difficult if the page doesn't load all the way from lack of internet. Now that you mentioned it, there where no typical 404 or page could not be loaded, just the 401 access denied with that particular js script.

markwal, 2016-02-13:
Ah.  That's useful information.  You're requesting that the local loopback works even when there is no network interface.

Well, there's two ways to think about that.  One is redesign octoprint so it isn't a client server model and the other is make sure the OS always can do loopback even when there is no network.  The first seems unlikely.  The second, I think, works already.

BTW I don't think you'd get a 401 when the OS thinks it can't get the url, you get a timeout or a no-internet-connection error.  401 means it was actually able to talk to the server and the server responded by saying: hey I don't know who you are.

jamesarm97, 2016-02-13:
Right. The odd thing is most of the content loads but a few errors make everything unusable. One example is the control page loads but none of the arrows or move buttons show and all the extrude buttons and stuff has moved up and are out of place. It looks like the error is combing from the packed js cache file that is created but I will have to wait until tomorrow after a print finishes to duplicate it and get some line numbers and logs.

foosel, 2016-02-13:
OctoPrint's UI doesn't load anything off the internet.

And the JavaScript files aren't even behind an authorization screen in stock install. 

Please provide all the information asked of you above by the bot (version, log files, etc), plus how you are accessing OctoPrint (because it sounds like there is something in between that's producing those 401s for you). Without that information there's nothing we can do here and everything is just blind guessing.

ntoff, 2016-02-13:
http://i.imgur.com/uUSaJWx.png

octoprint manual install in debian on a raspi 2b accessed via chrome on windows 10 pro x64 through wifi with the router's WAN DHCP lease disconnected

>Distributor ID: Debian
>Description:    Debian GNU/Linux 7.8 (wheezy)
>Release:        7.8
>Codename:       wheezy

foosel, 2016-02-13:
That is not a 401 when *accessing* the libs.js, that is a 401 triggered by a request sent *through* the libs.js which also include the jQuery ajax routines. Big difference.

That in the screenshot looks like something in the user session handing went wrong somewhere and I'm fairly sure there is already a ticket for that somewhere where I'm still waiting for proper reproduction steps since I cannot reproduce that.

It does not mean that anything is trying to access resources from the internet here.

ntoff, 2016-02-13:
like I said I'm not the OP but that only happens when the internet isn't accessible.

markwal, 2016-02-13:
It's probably the client browser doing something silly when it can't talk to dns or something.

foosel, 2016-02-13:
I just tried that by starting OctoPrint on my laptop, setting it to flight mode, then accessing the web interface on http://localhost:5000

[image]

Nothing suspicious, works and loads perfectly. Of course, the server threw an error when trying to download the plugin repository data, but that was caught properly within the corresponding error handlers and caused no issues at all. It ran fully functional and fully autonomous as expected.

So whatever does it for you and/or the OP when the internet isn't accessible, it isn't easily reproducible :/

ntoff, 2016-02-14:
>So whatever does it for you and/or the OP when the internet isn't accessible, it isn't easily reproducible :/

For me I seemed to be able to coax it into showing up if 

* I access the octoprint interface with the internet enabled
* release the DHCP lease on my router's WAN port (didn't try physically unplugging it) via it's GUI (on windows that'd be ipconfig /release)
* reboot the raspberry pi (either through putty or via the GUI prior to closing the client browser)
* close the client browser before the PI is back up
* access the GUI again once the pi is back up

jamesarm97, 2016-02-14:
From what I found it had to be a reboot without ethernet / wifi. If you had it plugged in at any time then it works ok so it does seem to be related to some underlying os thing. I think I got around it by editing hosts and adding my machine name and using that instead of localhost. I don't know if 127.0.0.1 made any difference from localhost but I could test that. This is what the browser log showed but didn't save the details for some reason. It was all related to trying to load the /api/xxxx files and if I clicked the offending link like the /api/printerprofile page I also got an error not authorized in a new tab and said the api key was not authorized (and obviously was not passed). I guess this can be closed and I will just use something other than localhost to access the pages.

Synchronous XMLHttpRequest on the main thread is deprecated because of its detrimental effects to the end user's experience. For more help, check http://xhr.spec.whatwg.org/.
less: parsed http://localhost:5000/static/webassets/packed_app.less?d41d8cd9 successfully.
less: css for http://localhost:5000/static/webassets/packed_app.less?d41d8cd9 generated in 115ms
less: css generated in 119ms
less: parsed http://localhost:5000/plugin/touchui/static/less/touchui.custom.less successfully.
less: css for http://localhost:5000/plugin/touchui/static/less/touchui.custom.less generated in 2720ms
less: css generated in 2720ms
Starting dependency resolution...
... dependency resolution done
Failed to load resource: the server responded with a status of 404 (Not Found)
Failed to load resource: the server responded with a status of 401 (UNAUTHORIZED)
Failed to load resource: the server responded with a status of 401 (UNAUTHORIZED)
Failed to load resource: the server responded with a status of 401 (UNAUTHORIZED)
Failed to load resource: the server responded with a status of 401 (UNAUTHORIZED)
Failed to load resource: the server responded with a status of 401 (UNAUTHORIZED)
Failed to load resource: the server responded with a status of 401 (UNAUTHORIZED)
Failed to load resource: net::ERR_INTERNET_DISCONNECTED
Failed to load resource: net::ERR_INTERNET_DISCONNECTED

ntoff, 2016-02-14:
>related to some underlying os thing

or some weird browser caching thing since I'm not accessing it via localhost and my OS is completely different to yours (yours being presumably debian where mine is windows 10)

are you also using chromium browser?

foosel, 2016-02-14:
@jamesarm97 can you also provide a screenshot of the console please? Just copying the text leaves out stuff, especially what's up with the last two lines (which I didn't see in my flight mode experiment, so I guess they are caused by something in your OctoPrint installation - maybe a plugin - or in your browser - extension? - that differs).

foosel, 2016-02-14:
Changed the title since it was misleading (OctoPrint does not rely on external files, something else is at play here)

foosel, 2016-02-16:
@jamesarm97 still waiting for some information from you here ;)

jamesarm97, 2016-02-16:
Sorry, I got home late and had to get a few last components for my new printer going. I did have some screenshots of what the gui looked like when it fails. I'll post those now and get shots of the actual errors tonight.

https://flic.kr/p/Ec6crT
https://flic.kr/p/Ec6cTp
https://flic.kr/p/E3QZh4

This last one is what I see when I click the logged 401 unauthorized error link (don't know that it is valid since I clicked it instead of it coming from code):
https://flic.kr/p/DCAW5c

foosel, 2016-02-16:
I know what the UI looks like when the API doesn't respond ;) I need a screenshot from the Javascript error console when stuff stops working. The copy pasted text output that you already provided wasn't enough.

Also the final screenshot is the expected outcome since when you only click the link you don't have an APIKey in the request (so your assumption here is correct).

In a nutshell:

  * reproduce the problem
  * provide a screenshot of the javascript error console

We need to figure out what component it is that's producing that 404 and those two ERR_INET_DISCONNECTED errors, but your text dump lacks that critical information.

jamesarm97, 2016-02-18:
#### What were you doing?

Trying to use Octoprint standalone as a front end to a printer without wifi or network connections.

#### What did you expect to happen?

Octoprint pages to load even without internet. Can still use it to control and print preloaded files.

#### What happened instead?

Pages did not load all the way. Errors in the console about 401 unauthorized.

#### Branch & Commit or Version of OctoPrint

1.2.10-dev6 maintenance branch

#### Printer model & used firmware incl. version

Repetier Firmware 0.92.8

#### Browser and Version of Browser, Operating System running Browser

RaspberryPi with Chromium browser

#### Link to octoprint.log

#### Link to contents of Javascript console in the browser

See attached link to photo

#### Screenshot(s) showing the problem:

https://flic.kr/p/DgXjXC

I have read the FAQ.

foosel, 2016-03-07:
@jamesarm97 I fell sick and hence no reply from my side until now.

The screenshot you provided only contains the 401s, but neither the 404 nor the ERR_INET_DISCONNECTED are present. Do you *only* see the 401s now?

jamesarm97, 2016-03-07:
The 404's may have been because the camera was disconnected and maybe the cam viewing links where not found because of that since mjpeg_streamer closes if the camera is unplugged

foosel, 2016-06-08:
The 401 issues are now tracked (and actually already solved in my local copy, just testing some final things before pushing it) in #1364. Closing this since the observed issues in case of missing net connectivity could not be reproduced and the seen 404s probably were caused by the local webcam server not being reachable either.

closed 2016-06-08 by foosel