OctoPrint/OctoPrint #1224: Could not upload to SD Card

Helisinus, 2016-02-13:
Hi, after updating to V1.2.9 i couldn't uplad gcode to SD Card. I See filse on SD Card, can print from SD Card, delete files, but i can't uplade files.

#### What did you expect to happen?
After pressing the upload button and uploading the file, octoprint should coppy the file to the SD Card

#### What happened instead?
The file was uploaded, but not copied to SD Card

#### Branch & Commit or Version of OctoPrinting
Version: 1.2.9 (master branch)

#### Printer model & used firmware incl. version
 Wanhao Duplicator i3. 

#### Browser and Version of Browser, Operating System running Browser

Chrome and Firefox

#### Link to octoprint.log

https://github.com/foosel/OctoPrint/files/129136/octoprint.txt

#### Link to contents of terminal tab or serial.log

https://github.com/foosel/OctoPrint/files/129135/serial.txt

Regards

Joerg

foosel, 2016-02-13:
Is this the terminal output from when you tried to upload to sd? I only see manual head movements and position requests in the one you provided, which is why I'm asking...

Helisinus, 2016-02-13:
Yes, but i don't see any try to copy. Normaly the file was uploaded to Opctopi and then the status was streaming or somthing like this. But now i don#t see anything.
Is there a way do downgrade the Octopi? For testing only?

foosel, 2016-02-13:
Yeah, but that looked to be related to the finish/cancel of an sd print (it's a bug though). I might be mistaken though, I'm not able to look into this right now.

To downgrade on OctoPi try this:

```
source ~/oprint/bin/activate
cd ~/OctoPrint
git reset --hard 1.2.8
python setup.py install
sudo service octoprint restart
```

Helisinus, 2016-02-13:
Yes, after the downgrade, octoprint will copy the file to the SD Card.

Next i will upgrade again to 1.2.9 und do another try.

[image]

Helisinus, 2016-02-13:
So, now i have upgraded to 1.2.9 again and the same isuue. 
After the upgrade i have delete the logfiles, restart Octoprint and try to upload a file to SD Card. The File was uplaoded to Opctoprint nut not copied to the SD Card. Here are the new Logfile.
https://github.com/foosel/OctoPrint/files/129165/octoprint.txt
If you need more tests, logfiles or infos give me a short info.

Jorg
[octoprint.txt](https://github.com/foosel/OctoPrint/files/129165/octoprint.txt)

Helisinus, 2016-02-13:
Now i have installed Version: 1.3.0.dev888+geacd9f7 (devel branch)
This Version works also without problems. Its a issue in Version 1.2.9 only

Joerg

foosel, 2016-02-13:
Considering that devel contains all changes from master (aka 1.2.9) that is
very much surprising for me now.

I'll take a look ASAP (not before Monday though)

Helisinus, 2016-02-13:
No Problem, 1.3.0 is working for the Moment

foosel, 2016-02-15:
Fixed on the `maintenance` branch, will be part of the 1.2.10 release (which will be done very soon, just want to see if other issues exist first before pushing a release only for this fix).

commit 0a3dc4b3 (2016-02-15): Fixed a bug that prevented the "Upload to SD" button from working

Introduced with the bug fix for #1196

Closes #1224
closed 2016-03-16 by foosel