OctoPrint/OctoPrint #1225: Timelapse failing, invalid URL in 1.2.9

kmarcisz, 2016-02-14:
Hi,
I'm using Octoprint on Wanhao i3. In 1.2.8 timelapse was working. After updating to 1.2.9 timelapse stopped working for me. Based on a log I could type in an absolute URL, but I would prefer relative. Is there any way to achieve this?
Here's the log excerpt. http://octopi.local/webcam/shot.jpg is accessible and displays the JPG of the current print.

```
2016-02-14 21:32:41,664 - octoprint.timelapse - ERROR - Could not capture image /home/pi/.octoprint/timelapse/tmp/top_20160214181120-0.jpg from /webcam/shot.jpg
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.9-py2.7.egg/octoprint/timelapse.py", line 461, in _perform_capture
    r = requests.get(self._snapshot_url, stream=True)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg/requests/api.py", line 69, in get
    return request('get', url, params=params, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg/requests/api.py", line 50, in request
    response = session.request(method=method, url=url, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg/requests/sessions.py", line 451, in request
    prep = self.prepare_request(req)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg/requests/sessions.py", line 382, in prepare_request
    hooks=merge_hooks(request.hooks, self.hooks),
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg/requests/models.py", line 304, in prepare
    self.prepare_url(url, params)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg/requests/models.py", line 362, in prepare_url
    to_native_string(url, 'utf8')))
MissingSchema: Invalid URL '/webcam/shot.jpg': No schema supplied. Perhaps you meant http:///webcam/shot.jpg?
```

markwal, 2016-02-14:
Huh.  I didn't think this would work for 1.2.8 either.  The snapshots for the timelapse are taken from the pi directly, not from the webpage you're looking at, so relative url's don't make any sense.  Relative to what?  You see, it is a naked request the pi is making.  Since it is the pi making the request, you can tell it to always ask itself for the snapshot no matter what its domain name is by using 127.0.0.1.

The thing that worries me here though is that the error message seems to also be wanting a query parameter '?' which is often the case for people that are using mjpg-streamer so that might be a new bug.  Please try this url:

http://127.0.0.1/webcam/shot.jpg

Please report back whether you get the same result or not.

foosel, 2016-02-15:
Configuration error on your end, see [the FAQ](https://github.com/foosel/OctoPrint/wiki/FAQ#how-do-i-have-to-configure-the-webcam-urls-in-octoprint-so-that-everything-will-work). OctoPrint needs to know how to reach the webcam, so it needs a fully qualified url. It isn't a browser running in a request context, it doesn't know what to do with "/webcam/shot.jpg".

@markwal the query isn't a query, it's the question mark belonging to the "perhaps you meant ... ?" question :)

closed 2016-02-15 by foosel

kmarcisz, 2016-02-15:
OK, got the FAQ link and it makes sense now. It is little confusing one path is suggested to be relative while the one below is absolute. Some explanation would help, and validating the URL requiring it to be absolute would not hurt.
I'll try to set up some HTML5 validation for the link and I'll send a pull request if I figure your codebase out.
Thanks for the help @foosel and @markwal !

foosel, 2016-02-15:
Version 1.3.0 (currently in the devel branch) already has a test button in
the UI to validate that stuff (see
https://github.com/foosel/OctoPrint/blob/devel/src/octoprint/static/js/app/viewmodels/settings.js#L223)
as well as some further explanation in the initial setup wizard (
https://github.com/foosel/OctoPrint/blob/devel/src/octoprint/plugins/corewizard/templates/corewizard_webcam_wizard.jinja2),
so no need for a PR for that. Since that test button depends on some new
features that are only in the devel branch, it can't be easily backported
to the maintenance branch (at least not without risking bugs, which I try
to avoid for the maintenance branch where possible).

In the meantime, improvements of the documentation are welcome!

The relative path is so that port forwarding scenarios (where the
hostname:port combo through which the UI is accessed can change) can work
without having to constantly reconfigure stuff. Since a browser knows what
to do with relative URLs, that works. The snapshot URL is accessed by the
server though. It doesn't make sense to use something relative or even
something not fully qualified ("no http") here - relative to what?
Localhost? The last accessed webpage request? What protocol? What host,
what port?

kmarcisz, 2016-02-15:
Awesome, I'll wait for the 1.3.0 branch to become stable and update then.

foosel, 2016-02-16:
Since I'm not entirely sure how you meant that: 1.3.0 release can actually
take a while, I would strongly advise against not updating anything anymore
until 1.3.0 gets stable.

Also - relative URLs for the snapshot URL still won't be possible then, the
problem persists with regards to "relative to what" ;)