OctoPrint/OctoPrint #1226: Disconnect after printing from SD card fails

hcomet, 2016-02-15:
#### What were you doing?

I am developing a plugin that automatically powers off a printer after a print completes. 
It runs on a Raspberry Pi and controls the power through GPIOs connected to relays. I 
noticed that OctoPrint does not close the connection after the printer is powered down so
I added a call to self._printer.disconnect() to disconnect before powing off the printer.
This works fine in OctoPrint 1.2.8. After upgrading to 1.2.9 the first call to 
_print.disconnect() after printing an SD stored file never returns. 

I also tried this scenario without the plugin and only using the web UI to see if it was a 
problem with my plugin. The problem is also present without any plugin interaction. If you
follow these steps:
1 - Connect to a printer
2 - select a gcode file on the printer's SD clear_status_LED
3 - printe the selected filter
4 - wait until print complete
5 - Try to disconnect through the UI. Click Disconnect. Connection state not updated.
6 - After waiting click the Disconnect button a second time. Connection state not updated.

The disconnect does not happen properly and the connection status is not updated. An 
exception shows up in the octoprint_log at this point. (link below)

Also did a mp4 screen capture of the steps on the terminal tab. (link below)

#### Branch & Commit or Version of OctoPrint

Version 1.2.9 (master branch)
running on Raspbian wheezy. Pi 2.

#### Printer model & used firmware incl. version

Rostock Max V2
Repetier 0.91.5
RAMBO board

#### Browser and Version of Browser, Operating System running Browser

Chrome Version 48.0.2564.109 m

#### Link to octoprint.log
Log with debug enabled for the UI based issue:
https://gist.github.com/hcomet/553ce471884e00a1837e

I also created a debug log with my plugin calling the disconnect:
https://gist.github.com/hcomet/a85be57979f2ac23b234

There was no exception shown in the log but there was one in the console.
Here is the console session:
https://gist.github.com/hcomet/ad81bec683bb7aaa17fc

#### Screenshot(s) showing the problem:

Video of the issue in the UI: 
https://dc1.safesync.com/LMHSnzgf/Stuff%20to%20Share/octoprint-1.mp4?a=zMqHejfGVfQ

Video of the issue using my pliugin:
https://dc1.safesync.com/LMHSnzgf/Stuff%20to%20Share/octoprint-2.mp4?a=D67XNmo3BPI

foosel, 2016-02-15:
Thanks for remembering me of that... I saw that in a different bug report, but of course forgot about it again.

foosel, 2016-02-16:
Solved on the `maintenance` branch, will be released with 1.2.10 within the next days.

commit d267a752 (2016-02-16): Capture NoSuchStorage error on SD card fails

We currently do not have a storage configured for SD card since
that hasn't yet been ported to the storage interface. So on an
SD card fail we need to capture that error when attempting to log
the print recovery data, or there will be issues processing
the cancel properly. A bit more of error handling is a good
idea here in any case too.

Fixes #1226
closed 2016-03-16 by foosel