OctoPrint/OctoPrint #1227: M110 withoug line number gives error and closes connection on M3D

jbliesener, 2016-02-16:
Hello,

I have an M3D printer and use it with donovan600's "M3D FIO" plugin on a Raspi 2. The latest firmware version (201601250) produces problems on connection startup, as an "M110" command without any line number not only gives an "e1" error, but also resets the printer. 

This, in turn, lets the /dev/ttyACM0 interface disappear and reappear a short moment later as /dev/ttyACM1. Then the process repeats itself (this time, ttyACM1 disappears and ttyACM0 reappears).

The comm thread seems to take a while to recognize the situation and continues to work. Sometimes (less than 2% of all cases), I actually am able to maintain the connection and print. Guess that's a favourable racing condition.

Suggestion: Could you allow to customize the "M110" command that is sent to the printer on connection establishment and to detect the baud rate? "M110 N0" also would be fine. But, as far as I understand GCode, the line number is NOT optional and must  be specified either as prefix (Nx M110) or as parameter (M110 Nx). So, the software actually is correct in complaining about a single M110 command without parameter. On the other hand, it clearly shouldn't reboot afterwards. 

Here's the terminal log:

---
```
Changing monitoring state from 'Offline' to 'Detecting serial port'
Connecting to: /dev/ttyACM1
Changing monitoring state from 'Detecting serial port' to 'Opening serial port'
Connected to: Serial<id=0x283afb0, open=True>(port='/dev/ttyACM1', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Starting baud rate detection
Changing monitoring state from 'Opening serial port' to 'Detecting baudrate'
Recv: wait
Trying baudrate: 250000
Recv: e1
Send: M110
Changing monitoring state from 'Detecting baudrate' to 'Closed'
Connection closed, closing down monitor
Connecting to: /dev/ttyACM2
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x2080730, open=True>(port='/dev/ttyACM2', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: M110
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1419
Changing monitoring state from 'Connecting' to 'Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1419'
Connection closed, closing down monitor
Connecting to: /dev/ttyACM1
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x2950a10, open=True>(port='/dev/ttyACM1', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: M110
Recv: e1
Recv: ok REPETIER_PROTOCOL:2 FIRMWARE_NAME:Micro3D FIRMWARE_VERSION:2016012501 MACHINE_TYPE:The_Micro X-SERIAL_NUMBER:BL15103001025011
Changing monitoring state from 'Connecting' to 'Operational'
Send: M21
Send: M117
Send: M114
Recv: ok ZV:1 S:130
Send: M619 S12 T4
Recv: ok  Z:41.2730
Send: M619 S16 T4
Recv: ok  PT:12 DT:1048576000
Send: M619 S20 T4
Recv: ok  PT:16 DT:1065940419
Send: M619 S24 T4
Recv: ok  PT:20 DT:1057408102
Send: M619 S28 T4
Recv: ok  PT:24 DT:1043934001
Send: M619 S32 T4
Recv: ok  PT:28 DT:-1094940398
Send: M619 S40 T1
Recv: ok  PT:32 DT:-1080932543
Send: M619 S41 T1
Recv: ok  PT:40 DT:2
Send: M619 S70 T4
Recv: ok  PT:41 DT:90
Send: M619 S74 T4
Recv: ok  PT:70 DT:0
Send: M619 S78 T4
Recv: ok  PT:74 DT:0
Send: M619 S82 T4
Recv: ok  PT:78 DT:0
Send: M619 S86 T4
Recv: ok  PT:82 DT:0
Send: M619 S94 T4
Recv: ok  PT:86 DT:0
Send: M619 S102 T4
Recv: ok  PT:94 DT:1153138688
Send: M619 S106 T4
Recv: ok  PT:102 DT:1153138688
Send: M619 S110 T4
Recv: ok  PT:106 DT:1153138688
Send: M619 S114 T4
Recv: ok  PT:110 DT:1114636288
Send: M619 S118 T4
Recv: ok  PT:114 DT:1120665600
Send: M619 S98 T1
Recv: ok  PT:118 DT:1135869952
Send: M420 T100
Recv: ok  PT:98 DT:1
Recv: ok
Recv: wait ..
Unexpected error while reading serial port, please consult octoprint.log for details: TypeError: ''NoneType' object is not callable' @ comm.py:_readline:1419
```
---

And kern.log:

```
Feb 16 01:32:26 octopi kernel: [96132.872689] usb 1-1.4: new full-speed USB device number 115 using dwc_otg
Feb 16 01:32:26 octopi kernel: [96132.976049] usb 1-1.4: New USB device found, idVendor=03eb, idProduct=2404
Feb 16 01:32:26 octopi kernel: [96132.976076] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb 16 01:32:26 octopi kernel: [96132.976094] usb 1-1.4: Product: The Micro
Feb 16 01:32:26 octopi kernel: [96132.976110] usb 1-1.4: Manufacturer: M3D
Feb 16 01:32:26 octopi kernel: [96132.976126] usb 1-1.4: SerialNumber: BL15103001025011
Feb 16 01:32:26 octopi kernel: [96132.978737] cdc_acm 1-1.4:1.0: ttyACM1: USB ACM device
Feb 16 01:32:40 octopi kernel: [96146.969704] usb 1-1.4: USB disconnect, device number 115
Feb 16 01:32:40 octopi kernel: [96146.969987] cdc_acm 1-1.4:1.0: failed to set dtr/rts
Feb 16 01:32:40 octopi kernel: [96147.242780] usb 1-1.4: new full-speed USB device number 116 using dwc_otg
Feb 16 01:32:40 octopi kernel: [96147.346452] usb 1-1.4: string descriptor 0 read error: -32
Feb 16 01:32:40 octopi kernel: [96147.346481] usb 1-1.4: New USB device found, idVendor=03eb, idProduct=2404
Feb 16 01:32:40 octopi kernel: [96147.346501] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Feb 16 01:32:40 octopi kernel: [96147.347850] cdc_acm 1-1.4:1.0: ttyACM2: USB ACM device
Feb 16 01:32:42 octopi kernel: [96149.530048] usb 1-1.4: USB disconnect, device number 116
Feb 16 01:32:42 octopi kernel: [96149.530342] cdc_acm 1-1.4:1.0: failed to set dtr/rts
Feb 16 01:32:43 octopi kernel: [96149.762721] usb 1-1.4: new full-speed USB device number 117 using dwc_otg
Feb 16 01:32:43 octopi kernel: [96149.866297] usb 1-1.4: New USB device found, idVendor=03eb, idProduct=2404
Feb 16 01:32:43 octopi kernel: [96149.866325] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb 16 01:32:43 octopi kernel: [96149.866342] usb 1-1.4: Product: The Micro
Feb 16 01:32:43 octopi kernel: [96149.866358] usb 1-1.4: Manufacturer: M3D
Feb 16 01:32:43 octopi kernel: [96149.866375] usb 1-1.4: SerialNumber: BL15103001025011
Feb 16 01:32:43 octopi kernel: [96149.867691] cdc_acm 1-1.4:1.0: ttyACM1: USB ACM device
Feb 16 01:32:59 octopi kernel: [96166.684353] usb 1-1.4: USB disconnect, device number 117
Feb 16 01:32:59 octopi kernel: [96166.684643] cdc_acm 1-1.4:1.0: failed to set dtr/rts
Feb 16 01:33:00 octopi kernel: [96166.922831] usb 1-1.4: new full-speed USB device number 118 using dwc_otg
Feb 16 01:33:00 octopi kernel: [96167.026583] usb 1-1.4: string descriptor 0 read error: -32
Feb 16 01:33:00 octopi kernel: [96167.026616] usb 1-1.4: New USB device found, idVendor=03eb, idProduct=2404
Feb 16 01:33:00 octopi kernel: [96167.026636] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Feb 16 01:33:00 octopi kernel: [96167.027970] cdc_acm 1-1.4:1.0: ttyACM2: USB ACM device
Feb 16 01:33:02 octopi kernel: [96168.988650] usb 1-1.4: USB disconnect, device number 118
Feb 16 01:33:02 octopi kernel: [96168.988933] cdc_acm 1-1.4:1.0: failed to set dtr/rts
Feb 16 01:33:02 octopi kernel: [96169.222808] usb 1-1.4: new full-speed USB device number 119 using dwc_otg
Feb 16 01:33:02 octopi kernel: [96169.326277] usb 1-1.4: New USB device found, idVendor=03eb, idProduct=2404
Feb 16 01:33:02 octopi kernel: [96169.326308] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb 16 01:33:02 octopi kernel: [96169.326325] usb 1-1.4: Product: The Micro
Feb 16 01:33:02 octopi kernel: [96169.326341] usb 1-1.4: Manufacturer: M3D
Feb 16 01:33:02 octopi kernel: [96169.326358] usb 1-1.4: SerialNumber: BL15103001025011
Feb 16 01:33:02 octopi kernel: [96169.327799] cdc_acm 1-1.4:1.0: ttyACM1: USB ACM device
```

foosel, 2016-02-16:
> `Recv: ok REPETIER_PROTOCOL:2`

These are all set?

[image]

> But, as far as I understand GCode, the line number is NOT optional

That is a wrong assumption, sorry to say. Repetier ticks a bit different here and demands line numbers and checksums with all commands. Marlin doesn't. Firmwares differ heavily here, and not only there, a lot of stuff behaves completely differently between various implementation thanks to there not actually being a standard definition of the "GCODE" used for printing. And the actual GCODE standard for CNC usage (as defined by [NIST](https://www.nist.gov/customcf/get_pdf.cfm?pub_id=823374)), which has pretty much nothing to do with the "GCODE" dialect used in 3D printing, doesn't even define M110, line numbers or checksums. Unsurprisingly I might add, it's a file description format, not a communication protocol.

Furthermore:

> I have an M3D printer and use it with donovan600's "M3D FIO" plugin on a Raspi 2

If I remember correctly the plugin also injects itself into the communication flow, so please also make sure this is even actually caused by OctoPrint and not that plugin.

jbliesener, 2016-02-16:
Thank you, Gina, for your quick reply. 

Setting the Repetier features doesn't help much. Looking at the issue again, it seems possible that the root cause may not be the M110. As you mentioned, M3DFio intercepts the connection sequence and executes some commands that aren't echoed (like the "M115" that reports the software version)

I'll have to dig deeper, but that won't happen until the end of the month. Please leave the "awaiting information" and "needs testing" labels on the issue and keep it opened. If it proves to be related to M3D FIO, I'll close it myself.

Again, thank you for your help and for the outstanding project.

jbliesener, 2016-02-26:
Seems to be solved with new m3d-fio version.

closed 2016-02-26 by jbliesener

bruster999, 2016-05-06:
Hi,

I have the exact same problem as described above but my printer is a generic RepRap Prusa i3 model. I have two questions: 

1) Is there a way to have the job continue from where it left off? The job is a 7 hour job and is 90% finished so it would be preferable to continue rather than start over.

2) Do the setting noted above apply to all printers (particularly my model) or just the M3D? At the moment my settings are as noted below (by default).

[image]

Thanks,
Bruster

foosel, 2016-05-06:
@bruster999 

To quote the [contribution guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md):

> **Very important:** Please make absolutely sure that if you find a bug that looks like it is the same as your's, it actually behaves the same as your's. E.g. if someone gives steps to reproduce his bug that looks like your's, reproduce the bug like that if possible, and only add a "me too" if you actually can reproduce the same issue. Also **provide all information** as [described below] 
(https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report) and whatever was additionally requested over the course of the ticket even if you "only" add to an existing ticket. The more information available regarding a bug, the higher the chances of reproducing and solving it. But "me too" on an actually unrelated ticket makes it more difficult due to on top of having to figure out the original problem there's now also a [red herring](https://en.wikipedia.org/wiki/Red_herring) interfering - so please be very diligent here!

You say you have a completely different printer with probably an even more completely different printer firmware but are observing the same problem - that's somewhat unlikely, and in order to help you we'd need a full bug report please. There's a [template](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report) to fill out for that, and of special importance here is the *exact* output of what you are seeing in the printer communication when the problem arises. So ideally a `serial.log`, at the very least the contents of the Terminal tab.

> 1) Is there a way to have the job continue from where it left off? The job is a 7 hour job and is 90% finished so it would be preferable to continue rather than start over.

Not built in, and it's a bit fickly and not very certain that it will work. You can manually edit the gcode file to only contain the last 10% of your print, making sure to move the print head to the correct position before. Of course that only works if you have not moved the already printed model in any way on the bed.

> 2) Do the setting noted above apply to all printers (particularly my model) or just the M3D? At the moment my settings are as noted below (by default).

The stuff that has a little "Repetier" next to is **must** be checked if your printer runs Repetier firmware.