OctoPrint/OctoPrint #1231: pan/tilt control

boelle, 2016-02-18:
just loaded octoprint and got an idea

mount the raspberry pi cam on a pan/tilt thing and attach 2 servo's

any plans to support this?

if not how could i add 4 buttons on the controls page that can call 4 python scripts?

Salandora, 2016-02-19:
Hi
This would be a nice feature.
There is actually no way to run a python command through custom controls.
You could still do it by a plugin.
Attach the servos to the printers hardware and use the servo commands to control them. E.g. by a slider.
Or manually add some controls to the page and add some backend commands.

boelle, 2016-02-19:
I have no free pins left, hence the reason i asked :-)

foosel, 2016-02-19:
You can have a plugin add arbitrary functionality to the control tab.

Needs registered view model with getAdditionalControls (
http://docs.octoprint.org/en/master/plugins/viewmodels.html) + javascript
control type sending an ajax request to the backend (
http://docs.octoprint.org/en/master/features/custom_controls.html#controls)
+ SimpleApiPlugin/AssetPlugin in the backend to inject js and handle ajax
request (http://docs.octoprint.org/en/master/plugins/mixins.html)

boelle, 2016-02-19:
thats over my head, so new eletronics seems to be the trick for me

Salandora, 2016-02-19:
@boelle not really, I can build such a plugin for you if you want, it would also be a nice feature to add to my printer, atleast the tilt function would be great for me xD.
So I have a personal interest in this.

maybe you wanna join the octoprint channel, so we can talk about it.

boelle, 2016-02-19:
i can do a bit later, i have started update to windows 10

but channel... you mean IRC chat?

Salandora, 2016-02-19:
ohh yeah sorry forgot to add IRC in front of channel xD.

boelle, 2016-02-19:
will be there as soon win10 is done.... at update 120 out of way to many :-/

boelle, 2016-02-19:
yaa... finally it says upgrading to win10, so in chat in less than an hour

boelle, 2016-02-19:
trying to get in the chat.... very slow :-/

boelle, 2016-02-19:
i only get the 

IRC Smileys

text at top

i use this link: http://en.irc2go.com/webchat/?net=freenode&room=octoprint&nick=boelle

foosel, 2016-02-19:
Use a proper irc client, or whatever freenode recommend with regards to a web front-end. I can recommend Hexchat.

And please keep in mind that everyone subscribed to this bug tracker gets notified about every single one of your comments here - that's starting to get a bit spammy between "updating to win 10" and "almost there!" ;)

ntoff, 2016-02-19:
that'd be so cool, could the pan feature be triggered via an end stop? so when the bed homes the camera can automatically point itself at the home position in case where that is, is out of view from the normal print location

Salandora, 2016-02-19:
Well we decided to let the plugin call a script to control a servo/servos.

So sorry no endstop planned :-(
But you can make your own python script :-)

ntoff, 2016-02-20:
so for what I asked about I wouldn't need the plugin at all? Just a python script monitoring the GPIO pins in much the same way I currently have one monitoring for a button press to shut the rpi down?

well, guess I know what my next project will be.

boelle, 2016-03-06:
@Salandora hi, how is it going?

I'm in the process of gathering stuff so I can build assemble the hardware

Salandora, 2016-03-07:
I'm going to start today. Had less time last week... sorry

boelle, 2016-03-07:
np at all

It will take me some time too before i'm ready :-D

Salandora, 2016-03-19:
@boelle here is a picture Work in Progress:
http://screenshot.co/#!/dc8940e964

Still need todo the settings part, so you can set a maxValue for pan and tilt and the path to the script which should be called on each button press.

boelle, 2016-03-19:
Looks excatly what i have been thinking off.

I got the servos on the table and some other stuff. Some friends in UK are sending me a free set of Dibond sheets so will rebuild my printer from ground up while adding this. Also need to figure optimal spot to mount the camera

Salandora, 2016-03-26:
@boelle here is the link to my repo: https://github.com/Salandora/OctoPrint-PanTilt
It is still work in progress.
I still need to check if the script gets called correctly with the correct parameters but I think so.

boelle, 2016-03-26:
np

i'm still waiting for dibond so i can rebuild my printer

zootoy, 2016-04-23:
Hello
I am hoping this is the right place to ask.
I am looking to execute a script from a button with custom controls.
The idea is simply to turn a set of LED lights on and off with 2 buttons on the control page.
I have a working "lightson.sh" and "lightsoff.sh" that simply turn a gpio on and off respectively.
I have tried using the custom controls to do those scripts by the buttons but nothing happens.
In this thread Salandora mentions: "Well we decided to let the plugin call a script to control a servo/servos."

Further down another user, ntoff mentions: "Just a python script monitoring the GPIO pins in much the same way I currently have one monitoring for a button press to shut the rpi down?"

Is this where i have been going wrong? I should be using an executable python script rather than .sh?

Side note: i have been successfully able to use the custom system commands to have a functioning lights on/lights off control via the drop down at the top but i would like to have on/off buttons on the controls page.

Sorry if I haven't explained myself properly or if this is the wrong place to have posted this.
Thank you for any help.

Salandora, 2016-04-23:
@zootoy what you are requesting is not possible out of the box.
You would need to write a plugin to achieve what you want to do.
This plugin would need to inject 2 buttons for your LED scripts and call them on a button click.
So it is nearly the same as the pantilt plugin I have written for this thread.
You could easily use it as a base, fork it and modify it to your needs.

claustro, 2016-09-26:
Hi all any news regarding this project?
I'd love to drive my pan tilt camera inside octoprint web page unfortunately I am totally new here and I am unable to make it by myself :-(
Anyway I am keep trying , for now I installed   servoblaster and I am able to drive my servo through command line , just a small step i know :-)

Salandora, 2016-09-26:
Uhm I'm not sure anymore because it is quite a time since I made the last changes but you should be able to use my plugin and use your command line tool and pass the values to it.
Might be that you'll need a wrapper shell script but that wouldn't be to hard to work out.
If you show me some example for pan and tilt with your command line tool I might be able to help you get everything working.

claustro, 2016-09-26:
Wowww thank you very much for your fast answer! I tried to load you plugin but  then I had problem logging in and so I uninstalled  the plugin . Maybe I can try another time tomorrow.
I am a totally NOOB in linux stuff , so I followed a couple of guides around internet to make the actual setup. 
at this point I am able to command the camera through command line using a script front .cgi like this 

`#!/bin/bash
echo P1-11=40% >/dev/servoblaster
echo P1-16=42% >/dev/servoblaster`

and then I can move the servo simply typing  
`pi@EVO3D:/usr/local/bin $ ./front.cgi`
This is  my dead end :-)

I also installed "custom control editor " thinking it could be exec the script but obviously I was wrong and it doesn't meant to this purpose.

claustro, 2016-09-26:
I forgot to mention. I don't know if servoblaster and octoprint can work togheter , I had to disconnect the power of both servo 'cos when I started octoprint both servo started to move randomly by themselves.

foosel, 2017-01-27:
@Salandora is that plugin something we should maybe register in the repository (and then close this issue)?

Salandora, 2017-01-27:
WEll it's not finished yet so I wouldn't recomment to put it in the repo right now, but I think we can close this issue if @boelle is ok with it. I think it would be better to talk about this in IRC instead of an issue tracker.

camnop, 2017-11-01:
how can I set the "Path to Script"? 

My path would be it

/home/pi/scripts/servopwm.sh
(receiving 2 variables having values between 0 to 180.. like:
/home/pi/scripts/servopwm.sh 20 40

[image]
[image]

My script is working very well when I test using BASH but when I try on Octoprint GUI using your plugin nothing happens.
Is my Path to Script wrong?

*My script that I'm using (pigpio bash shell script) is uploaded here: https://github.com/c-devine/OctoPrint-PanTilt-Script

alex-gibson, 2017-11-20:
Hi, I just caught up with this, and I would love to be able to pan/tilt my camera using servos via Octopi.  Is it currently possible to install this as a plugin for OctoPi and use it?  What does 'getting started' look like right now?  I have Pi, webcam, Picam, servos, build plenty of electronics and can give good test feedback.
Thanks, Alex

foosel, 2017-12-04:
@Salandora what's the status of your plugin? Had someone just the other day wish for a pan-tilt-control on reddit ;)

alex-gibson, 2017-12-04:
+1!!

Thanks
Alex

Tapped on my mobile phone.

marcboivin, 2018-02-10:
I looked at the plugin and the key missing part, from what I get, is that you need a script to do the actual tilting and panning. 

I want to add the pan/tilt to my printer too. One thing I found while digging on the internet is this maker magazine article from 4 years ago : https://makezine.com/projects/raspberry-eye-remote-servo-cam/

Still, the code to control the device is using the same stack as octoprint does. So what I'm thinking is mashing the code from this repo https://github.com/MattStultz/PiCam into an actual plugin for Octoprint. This would allow people to follow the tutorial from maker magazine and a more straightforward approach to the whole thing.

marcboivin, 2018-02-14:
I have a first running version. Still not ready to be used but : 

* Tilt servo is working
* Everything is hooked up to the RPi headers (guessing it's the most frequent use case)
* Controls are based off of @Salandora's work

https://github.com/marcboivin/OctoPrint-PanTilt/tree/feature/picam-variant

alex-gibson, 2018-02-14:
That sounds brilliant!  Please do it if you can, and I will beta test!

Tapped on my mobile phone.

marcboivin, 2018-02-17:
As you would expect, things aren't as easy as they first seem. 

Good news is I made the plugin usable. The settings pane wasn't showing up so it was basically unusable.

I also added a script folder in which I provide a pi-blaster.sh script. So, basic idea is (if you're running octopi) 

Install pi-blaster https://github.com/sarfata/pi-blaster (I did it from source with a `sudo make install` at the end to have a porper install)

Checkout my develop branch https://github.com/marcboivin/OctoPrint-PanTilt/tree/develop
Install the plugin.

In the Octoprint's settings, under Pantilt set the script to :
`/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_pantilt/scripts/pi-blaster.sh`

I do make the assumption that you use GPIO17 (physical pin 11) for pan servo and GPIO18 (physical pin 12) for your tilt servo. I use ground and 5V pins on the Pi to power the servos. They work fine on my Pi3.

So once that less then ideal setup is done, you should have new buttons under the control tab to make the servos move around. 

I will try to make the plugin easier to install and more flexible, but right now it's all the time I had to work on this. 

Don't look to closely at my git history it's dirty and I tried lots of stuff before getting something half decent.

claustro, 2018-02-18:
> I will install it and try it, thanks!!

Thank you very much for your work I am going to test it asap.
thx
Andrea

closed 2019-03-19 by boelle