OctoPrint/OctoPrint #1235: [Bug/Optimization] Spiralized gcode causes OctoPrint to run away allocating memory until tab crashes.

thantik, 2016-02-19:
**What were you doing?**
printing spiralized, single-wall vase, of a rocketship.

**What did you expect to happen?**
Memory usage to stay stable.

**What happened instead?**
The gcode viewer seems to treat all the very tiny z movements from the spiralize as individual layers.  This seems to cause _extraordinarily high_ memory usage under Google Chrome.  Eventually crashing the tab.  Until the tab crashes, everything works fine.  I can start/stop print, etc.

**Branch & Commit or Version of OctoPrint**
1.2.8 Master Branch

**Printer model & used firmware incl. version**
DeltaMaker, Marlin

**Browser and Version of Browser, Operating System running Browser**
Mac, Windows, Google Chrome latest version.

**Link to octoprint.log**
If this is needed, I can obtain it.  Not sure if relevant - print still continues, octoprint still works.

**Link to contents of terminal tab or serial.log**
Tab crashes, no terminal tab output to copy.

**Link to contents of Javascript console in the browser**
Tab crashes, no applicable javascript console text.

**Screenshot(s) showing the problem:**
Tab crashes, I'm sure you know what that looks like. :)

I have read the FAQ.

A temporary workaround is to disable the gcode viewer.  This allows the interface to remain stable and not run away with memory allocation.

LazeMSS, 2021-03-31:
@thantik  do you have the file that causes this issue? I know it long time since you posted.

thantik, 2021-03-31:
@LazeMSS I do not - I haven't had this issue in some time due to switching away from Chrome and I am not sure that it still happens with newer versions.  You're welcome to close this issue out if you're cleaning house.  Back when I tested it, it was easily reproducible, but the gcode viewer has had multiple optimizations since then.

closed 2021-03-31 by jneilliii