OctoPrint/OctoPrint #1236: Temperature spikes when heating hotend

wildcardfox, 2016-02-19:
#### What were you doing?

I was starting a print.

#### What did you expect to happen?

A constant temperature curve that gradually shows the rise, and reporting or current temperature readings when the print finishes.

#### What happened instead?

Temperature spikes when heating the hotend. It shows a drastic fall for one reading like in the log that I have attached where the temperature reporting goes from 157.61 to 59.49 then to 161.77.

Recv: T:157.61 /220 B:44.89 /43 B@:0 @:255
Recv: T:59.49 /220 B:44.89 /43 B@:0 @:255
Recv: T:161.77 /220 B:44.89 /43 B@:0 @:255

When the print is finished current temperatures fail to report. 

The problem occurs every time on both of my machines that are running Repetier firmware 0.92.8. I have checked the thermistors and the readings on Repetier Host, and the temperature curve is constant. 

#### Branch & Commit or Version of OctoPrint

Version 1.2.9

#### Printer model & used firmware incl. version

Two Delta custom printers (Rostock and Large Kossel running Arduino Due with RADDS shields); Repetier 0.92.8

#### Browser and Version of Browser, Operating System running Browser

Octoprint, Browser Safari, Google chrome.

#### Link to octoprint.log

https://gist.github.com/wildcardfox/55cb3ec97d9bd63974bd

#### Link to contents of terminal tab or serial.log

terminal

http://pastebin.com/pp82KMJ1

serial.log

https://gist.github.com/wildcardfox/9696c09e86ddc1ec24d6

#### Link to contents of Javascript console in the browser

#### Screenshot(s) showing the problem:

Screen shot

[URL=http://s979.photobucket.com/user/wildcardfox/media/wildcardfox028/Screen%20Shot%202016-02-19%20at%208.34.59%20AM_zpsxov7z8kk.png.html][IMG]http://i979.photobucket.com/albums/ae280/wildcardfox/wildcardfox028/Screen%20Shot%202016-02-19%20at%208.34.59%20AM_zpsxov7z8kk.png[/IMG][/URL]

I have read the FAQ.

foosel, 2016-02-19:
OctoPrint can only show on the temperature graph what your printer is telling it. And as you pointed out, the data received from the printer contains these spikes and drops. It actually looks like it simply swallows a 1 between the "T:" and the "59.49..." (the whole line minus the "Recv:" comes from the printer like this).

You say you run repetier firmware and the issue doesn't exist with repetier host. Host and firmware can talk a different (binary) protocol with each other, whereas OctoPrint uses the "mainstream" textual protocol. My guess would be there's a bug in the repetier firmware here in the implementation of the textual protocol side, and something causes the values to be reported wrong occasionally.

This is nothing that OctoPrint can fix, it relies on your printer reporting accurate and meaningful values, which at this point it doesn't. You'll probably see the exact same behaviour with anything else that is not repetier host (printrun, cura, etc). If I were you I'd open a ticket with the firmware.

wildcardfox, 2016-02-19:
Thank you for crossing Octoprint off the possible suspects. I will open a file with Repetier. Thank you.

closed 2016-02-19 by wildcardfox