OctoPrint/OctoPrint #1237: [Request] Audio output by reached Temperature

DeathmannX, 2016-02-19:
Hi Guys, 

first of all THX for this nice Feature for so many 3D Printers. :+1: 

Maybe you can implement a simple Audio File that will be activate if the Hotend & Heatbed reatched the Temparture.

Something like this maybe: http://www.fromtexttospeech.com/output/0570230001455918674/10717246.mp3

Or creat a upload Button to choose own Sound Files. ;)

Greets

ntoff, 2016-02-20:
does your printer have a beeper / buzzer? I added M300 S1500 P100 after the heat up to notify me via the piezo that heating is finished.

markwal, 2016-02-21:
Do you want the sound to play out of the pi, the browser or the printer?

boelle, 2016-03-06:
i think he means out of the PI, its the only thing that would make sense

playing it from printer have nothing ro do with octoprint, and if doing it from browser would prevent sound being played if browser is closed

of course it would make sense since the pi could be at a place where you cant hear it, and it would require speakers and miniamp

@DeathmannX

markwal, 2016-03-06:
@DeathmannX Well?  For when you're in the house with your browser openand the printer is in the garage.  *or* Just out of speakers attached to the Pi?    Either way, I think a plugin could be made, but one would primarily be in javascript and the other would primarily be written in python.

ntoff, 2016-03-07:
>i think he means out of the PI, its the only thing that would make sense

I would say browser. You might not even be physically close enough to the pi to hear it but if you've got a print going then theoretically you'd leave a browser window open to watch it (at least I do).

markwal, 2016-03-07:
I suppose you could make a plugin do both, but why did @deathmannx drop a request and disappear?

foosel, 2016-03-07:
> why did @deathmannx drop a request and disappear?

It's a surprisingly popular sport. There are other variants too, involving bug reports and support requests. :football:

boelle, 2016-03-07:
I might sound rude or something

but could the bot be set to close those issues where the OP suddenly goes dead so that you can concentrate about stuff that matters most, ie where people are active

foosel, 2016-03-07:
It already does that for issues if it detects that info is missing and that doesn't get updated within 14 days. Not for feature requests though, and that would be hard to implement to not produce false positives all the time.

But I regularly go through anything that still sits on `status:awaiting information`, ping the people involved and if nothing happens, "Close and comment" it is.

boelle, 2016-03-07:
Did the same while i was involved in Marlin Firmware

When i started out some issues where more than 3 years old :-D

foosel, 2016-03-10:
Ok, that's it, I'm closing this one now due to lack of interest by the OP to further specify what it actually is they are requesting in the first place. "Drive-by" anything just doesn't fly here.

And audio-anything is something that ever would be added through a plugin anyhow.

closed 2016-03-10 by foosel

R0dri, 2017-06-13:
I have an inquiry about this subject. I was thinking instead of a beeping sound, which i believe could be more complex, a notification popup message is created. I think that an API like interface could be great, you could easily do a facebook api from that point or something to send you a message to your phone or pc if your printer is ready to work. I usually put the printer to heat up while i am slicing or doing other stuff. It could be really helpfull to know the exact time it is ready.

R0dri, 2017-06-13:
This could work in setting some 'fail-safe' alert. I have a low power hot end, when i tried to print at a high temperature with the fan on, the hot end could not reach even the melting point so i ended up with a messy extruder. It would be very helpful to have some kind of alerts maybe user set to watch out for those unhandled behaviors.

R0dri, 2017-06-13:
Should i create a new issue on this?