OctoPrint/OctoPrint #1238: Inconsistent connection to printer

ntoff, 2016-02-21:
#### What were you doing?

Trying to use octoprint as I have done many times before

#### What did you expect to happen?

Connect to the printer on start up

#### What happened instead?

Constant sending of M110 with no connection to the printer on first start, hitting the connect button then causes the connection to complete as usual

#### Branch & Commit or Version of OctoPrint

Version: 1.2.10.dev16+gc82ec3d (maintenance branch)

#### Printer model & used firmware incl. version

My own concoction built with a standard RAMPS + Arduino running Marlin v1.0.2

#### Browser and Version of Browser, Operating System running Browser

Google Chrome Version 48.0.2564.116 m  on Windows 10 x64 though I suspect that has nothing to do with octoprint's USB connections

#### Link to octoprint.log

octoprint log http://pastebin.com/K2wCzCDQ

#### Link to contents of terminal tab or serial.log

serial log http://pastebin.com/abkQ3nsm

#### Link to contents of Javascript console in the browser

Pretty sure this is not required as it's an issue between the octoprint pi and the printer.

#### Screenshot(s) showing the problem:

As above, issue is between pi and printer, not the UI

I have read the FAQ.

Additional notes: I changed to the "master" branch (which is apparently 1.2.4) and all was normal so I suspect it's something to do with the recent revamp of the connection routines in the maintenance branch. First time seeing this issue and I've been using the maintenance branch for ages now.

foosel, 2016-02-21:
Thanks for reporting this! Also good to know that you run maintenance usually, that is awesome since it helps finding issues in the next release before they hit the general public, please keep reporting stuff like this! :)

Looks like commit c82ec3d3 needs to be reverted then since it has side effects in some cases - tested fine against the printer I tested it, but that stuff can be tricky when it comes to general good behaviour. I'm currently out sick but will correct that as soon as possible, unless @Salandora or @markwal beat me to it that is ;)

PS: master should actually report as 1.2.9, might be that you haven't fetched all tags though in which case it will report as some older version plus a bunch of commits.

ntoff, 2016-02-21:
I just ran "git checkout master" in the octoprint directory and then did the reinstall. Only did it to test so wasn't really too fussy about which version it was other than it NOT being the maintenance branch just to see if perhaps it was something on my end since for some reason my pi also happened to not want to connect to my wifi at the time as well but that turned out to be completely unrelated.

I wonder if it's because my arduino is a "fake" using the CH340 USB-Serial chipset. I'm currently printing something for the next hour & a half but if I'm still awake when it's done (it'll be about 1am here) I'll hook it up to the Mendel90 which I believe uses an ATMega16U2 for USB/Serial and see if I get the issue there as well.

ntoff, 2016-02-22:
Update: I got myself a shiny new arduino / ramps setup tonight and it uses the Atmel chip for USB stuff and it also encounters this connection issue with repetier firmware version 0.92 on it.

I find that if I hit the reset button a couple of times (on the arduino) it will then connect ok.

foosel, 2016-02-22:
I've reverted commit c82ec3d with commit 35e20162ec8dc38cb, that should get things going again. Could you verify?

From the locks it looks like the persistent sending of M110 just "overwhelmed" your firmware. Would you be open to test things when we get around to attempt to find a better variant of a more solid connection sequence?

ntoff, 2016-02-23:
Yeah, I can test some stuff. If it gets put in the maintenance branch chances are I'll just automatically have at it anyway because as soon as I see a "Update available" notification I feel compelled to press it :)

Do you think it matters that I have the printer going through a powered USB hub rather than directly into the raspi?

foosel, 2016-03-10:
Just realized I never came back to this, sorry... been trying to get back on top of things after falling sick for two weeks.

The issue probably is rather caused by hardware than firmware config, but once we tackle the original `M110` tickling of the printer again, it still might be interesting. However, for now it seems to break things more than it fixes (especially after solving another `M110` related issue recently), so I'll leave my fingers off of another attempt for now ;)

Since this ticket was `maintenance` only anyhow and the issue is solved, I'm closing this now, thanks for the feedback!

closed 2016-03-10 by foosel