OctoPrint/OctoPrint #1239: [Request] login forgot password

TedBanana, 2016-02-22:
I login webpage of OctoPrint，but i forgot my password.
I only think that i can reinstall OctoPrint，but it's not a good way.
In login,how can I solve that  forgot my password or account ?

edzatool, 2016-02-22:
I screwed mine up also how do I reset password?

nophead, 2016-02-22:
There is a solution here:
https://printrbot.dozuki.com/Answers/View/7610/resetting+name+and+password

markwal, 2016-02-23:
Also here: https://github.com/foosel/OctoPrint/wiki/Configuring-Access-Control

Near the bottom, it says to rerun the wizard, do the following:
 1. edit ~/.octoprint/config.yaml and set  firstRun: true . It should be there already set to  false  in your case.
 2. remove ~/.octoprint/users.yaml

I would also:  sudo service octoprint restart

edzatool, 2016-02-23:
I don't get any of this.  Ed

markwal, 2016-02-23:
How about like this, from the command line, ssh into your pi then:

rm ~/.octoprint/users.yaml
nano ~/.octoprint

Search for firstRun in that file. Delete the "false" on that line and type "true". Save and exit. Then type this command:

sudo system octoprint restart

TedBanana, 2016-02-23:
Thanks.
I test method  that  solve my promble

foosel, 2016-02-24:
Suggestion for an easy way to make this information better observable:

- [ ] add "forgot password" link to login dialog, have that pop up a modal explaining the above steps + that admins can change user passwords

Better version:

- [ ] add new config setting to access control "allowPasswordReset", default value false
- [ ] if set to true display password change dialog on ui load (similar to first configuration wizard)
- [ ] allow user to change password of his admin account, set "allowPasswordReset" to false on confirm

tousdan, 2016-02-25:
I wouldn't mind trying my hand at this.

markwal, 2016-02-25:
Since people seem to have a hard time editing config.yaml successfully and they can't use the yamlpatcher for this case, perhaps an enhancement to the better version would be to trigger the password change when a particular file exists next to config.yaml (like ~/.octoprint/password_reset)?  Or perhaps when users.yaml doesn't exist, but accessControl is enabled?

foosel, 2016-02-25:
@markwal good idea with the file. I'd prefer not to have to nuke users.yaml because that would kill all accounts, just because the admin forgot their password. Sounds a bit bad ;) I'll adjust the check list above

devel branch also allows definition of additional command line commands, so something like `octoprint user reset_password someUsername` should also be considered here, but first id's go for one of the outlined solutions since they also are something that can be back ported to the maintenance branch and shipped soon that way.

@tousdan great! Will mark the ticket as grabbed shortly, please keep us posted here. No problem if you decide not to do it after all, but would be important to know about that :)

tousdan, 2016-02-26:
When you say "allow user to change password of his admin account" how should we figure out which account is the admin account? I can see in the user management that multiple users can have the admin role. Should we be able to pick which user to reset?

Also, once both conditions are met (accessControl.allowPasswordReset and the reset_password file exists) anyone accessing the app will get the opportunity to change the admin password. Is this what we want for a setup where multiple users might be accessing the app?

foosel, 2016-02-26:
The user should need to enter their account name. 

And yes, the access thing is a tiny issue, it would fall to the admin who forgot the password in that case to limit access. We have the same issue now with the "reset users.yaml" approach, so it wouldn't worsen the situation at least, since now everyone would see the setup dialog instead too, which is just as bad.

The CLI option certainly is nicer, but depends on functionality in 1.3.x (CLI in 1.2.x still is pretty horribly mashed together and a nightmare to extend), so while that should also be tackled, it would be nice to have a solution in the mean time that doesn't kill all user accounts in 1.2.x as well. 

Should only need to be relevant anyhow if there is only one admin who forgot their password since otherwise adminY can reset adminX's password without the hassle.

foosel, 2016-11-17:
@tousdan are you still looking into this? No problem if not, but I'd again mark it as up for grabs then :)

tousdan, 2016-11-17:
Feel free to! I've been on and off this issue and couldn't get my head around filling all the requirements to my own satisfaction.

gege2b, 2016-11-17:
hi
just a question here, why not mimic the method used almost everywhere ? I mean : 
force the admin to enter an email address, and send a "reset password" mail when a "forgot password" link is clicked on ?

It's a pretty standard way to reset password. Not perfect though, and I guess it would require more work than the suggestion above

my two cents :)

foosel, 2016-11-17:
Because"sending an email" is impossible without teaching thousands of users out there first how to enable their Pi's to send out e-mail. An application in the cloud can be securely preconfigured. An open source application running on a headless server under control of the user can't.

Iambecomeroot, 2017-10-13:
I don't have `~/.octoprint`. Octoprint is installed on my Arch Linux server using the octoprint package from the AUR.

markwal, 2017-10-13:
~ indicates *your* home folder.  . octoprint is in the home folder for the user that is running the daemon process.

Iambecomeroot, 2017-10-13:
`.octoprint` isn't in `/root` either. I am running octoprint with systemd.

markwal, 2017-10-13:
Yes. systemd is the engine that starts the daemons at boot.  But that doesn't tell you which user it is trying to use to start the daemon.

It'll be some folder like:

/home/something/.octoprint

But it may not be getting that far.  You may need to attempt to run it from the command line and see if it is failing before it can create the log folder (permissions issue perhaps)

Iambecomeroot, 2017-10-13:
The `.service` file has `User` and `Group` set to `octoprint` but there is no `octoprint` folder in `home`. Do I have to create that? If that doesn't exist, where is it saving everything?

jamespayne, 2017-12-27:
Since this issue seems a bit dead right now and I just encountered it today, I've recently added some documentation for deleting the users.yaml and editing/updating the config.yaml file for OS X [here](https://github.com/foosel/OctoPrint/wiki/FAQ/fcef7efa73700f37cff926385716d6865ececf9e).

@foosel Currently looking at option 2 and will let you know if I think I can implement it.

kantlivelong, 2017-12-27:
@Salandora with your upcoming changes for user permissions, is it worthwhile  to add a password reset feature now or would it be better to wait?

slimshizn, 2018-03-17:
Re-enabling via SSH brought my login back to life. Now before I uninstall astroprint, is it SAFE to or will that kill it again?

markwal, 2018-03-18:
>  Now before I uninstall astroprint, is it SAFE to or will that kill it again?

Who knows?  This is an octoprint bug list.  What does astroprint have to do with anything?  And SAFE to uninstall?  What do you mean?  Kill it again?  Kill what?  How did your login lose its life and it what way?

alainchiasson, 2018-05-06:
@foosel Has anyone made headway on a "password reset" console command ? Can you give a quick pointer to the password function ? That may help. 

I would like to have a look to see if I can make a go at it. I'm coming to Python via Ansible and OpenStack - so looking for little useful things to do.

foosel, 2018-05-08:
@alainchiasson nope, no one, and a PR for that against the `maintenance` branch would be really welcome! :)

aliaksei135, 2019-07-22:
@foosel Can this be closed?

foosel, 2019-07-23:
Once 1.3.12 is out, yes

foosel, 2019-10-22:
1.3.12 has been released.

closed 2019-10-22 by foosel