OctoPrint/OctoPrint #1240: Feature Request - Buttons that run scripts

heathicus, 2016-02-24:
I love using Repetier-Host when leveling my bed as I make use of the 5 manual control buttons that execute scripts.  With a tap of the button I can raise the nozzle, move it to a corner or the center of the bed and re-home Z.  Check level, then move to another location until the bed is leveled.  But I hate dragging my computer over to the printer, disconnecting the Pi, connecting the PC, etc.

I'd like to request a plugin or feature of Octoprint to add that functionality.  Add to the Control tab several buttons (at least 5) that the user can configure with gcode to send to the printer. Thanks!

markwal, 2016-02-24:
This already exists.  You can add a bunch of buttons to the control tab to send gcode.

* You can do it via config.yaml: http://docs.octoprint.org/en/master/features/custom_controls.html
or
* There's a plugin by @Salandora to allow you to edit this in settings: http://plugins.octoprint.org/plugins/customControl/

closed 2016-02-24 by markwal

heathicus, 2016-02-24:
Thanks!  I looked through the plugins, but never saw that one.  I was hoping for something a little less complicated, but will see what I can do with those options.