OctoPrint/OctoPrint #1241: Printer freezes or stutters when loading web interface, results in artefacts

thedoble, 2016-02-24:
#### What were you doing?

Loading the OctoPrint web interface

#### What did you expect to happen?

The print to continue normally

#### What happened instead?

My printer stops for a few seconds, stutters ... and then keeps printing normally.

#### Branch & Commit or Version of OctoPrint

Version: 1.2.9 (master branch) 

#### Printer model & used firmware incl. version

Rostock MAX V2, Repetier 0.91

#### Browser and Version of Browser, Operating System running Browser

Firefox 43.0.1, Windows 7 Pro x64
Problem also occurs when accessing from Safari on iPhone

#### Link to octoprint.log

https://dl.dropboxusercontent.com/u/908196/octoprint.log

#### Link to contents of terminal tab or serial.log

Excerpt from a stutter when loading the web interface:

Recv: ok 884
Send: N885 G1 X8.493 Y9.059 E152.8347*90
Recv: ok 885
Send: N886 G1 X-9.625 Y-9.059 E154.2554*95
Recv: wait
Send: N887 G1 X-10.190 Y-9.059 E154.2868*109
Recv: wait
Send: N888 M105*47
Recv: ok 886
Send: N889 G1 X7.928 Y9.059 E155.7075*94

#### Link to contents of Javascript console in the browser

http://pastebin.com/FmT6nzST

#### Screenshot(s) showing the problem:

This picture shows a print artifact (gap in print) which was caused due to a particularly long 'freeze' of the printer (around 5 seconds) - most of the time when the freeze occurs, a blob is left by the printer as the print head stops moving but plastic oozes out.

https://www.dropbox.com/s/o57mphukbpz9a98/Photo%2024-02-2016%2C%205%2043%2052%20PM.jpg?dl=0

I have read the FAQ.
I was directed here by the OctoPi team - https://github.com/guysoft/OctoPi/issues/197

Thanks

foosel, 2016-02-24:
I see it's trying (And weirdly enough failing) to delete time lapse data there in the log. Now, this is only a hunch but: did this also happen with 1.2.8? And if not (our you don't know) could you please try deleting/moving the existing time lapse files, so it doesn't have to fetch those from disk on a reload and see if it makes a difference in behaviour?

Easiest way on OctoPi would be to log in via ssh and do this:

    sudo service octoprint stop
    cd ~./octoprint
    mv timelapse timelapse.bck
    sudo service octoprint start

That will move the existing time lapse data top a new backup folder, so the data is safe but OctoPrint doesn't care about it. Then try to reproduce the issue and report back. 

You can get your timelapse data back to the right location with

    sudo service octoprint stop
    cd ~./octoprint
    mv timelapse timelapse.new
    mv timelapse.bck timelapse
    mv timelapse.new/*.mpg timelapse 
    rm -r timelapse.new
    sudo service octoprint start

Or, if you didn't do time lapses during the test or don't need to keep them

    sudo service octoprint stop
    cd ~./octoprint
    rm -r timelapse
    mv timelapse.bck timelapse
    sudo service octoprint start

thedoble, 2016-02-24:
Thanks for your reply. 

Yes this did happen with 1.2.8 as well.

I'll try moving the existing timelapse data as you say, I'm halfway through a print now so will do it tomorrow.

thedoble, 2016-02-25:
I found that after moving the timelapse data, the printer did not stutter in my testing during the first 30 minutes of the print.

I have just tested again (8 hours into print) and the printer is back to stuttering when loading the interface :(

ntoff, 2016-02-27:
Maybe your SD card is too slow so while it's trying to read the UI data and write the timelapse footage at the same time the card is getting bogged down? Try making an image of your SD card and duplicating it to another known to be good quality card.

thedoble, 2016-02-28:
Hi

The SD card is a Sandisk Class 10 Ultra 16gb. The timelapse files are around 200kb and are only written every 30 seconds, I doubt this would cause the SD card to bog down?

I have loaded up iotop on the Raspi and will do some testing to confirm this.

Loup911, 2016-03-23:
it happened to me the same thing. 
During printing if I reload the web page of my print server I notice the processuce octoprint take 100% of the assigned CPU 

I'am wits a raspberry pi 2B with 4 core 

I think the solution might lie in separating the serial and web in 2 difent process. so they used a diferent process and could be assigned a diferent CPU.

I do not know if linux can assign priority level of the process. If so would require serial communication is in high priority over the web interface.

markwal, 2016-03-23:
Since I don't have this difficulty with either RPi 2 nor 3 and the CPU is not heavily loaded in either case, I doubt it is CPU.  It's either different add-on hardware (SD card, wi-fi dongle, etc.) or some other software installed or busy doing something that mine is not (like deleting old timelapses, searching through old gcode files updating analytics and stats).

Loup911, 2016-03-23:
[image]

Loup911, 2016-03-23:
printing  job runing and updating the web page of octoprint server

gddeen, 2016-03-23:
Hmmmm...

I stopped doing time lapses, and changed the port speed to 250K to the printer and keep my laptop connected and try to never watch a print.... (I waited for RPI3, but gave it away to daughter. Less stressful than watching prints.)

markwal, 2016-03-23:
Huh.  It is the CPU.

Loup911, 2016-03-23:
I think to use a different application to manage the serial communication and the website could be an advantage. 

Using one application that handles serial interface only, it would be easier to manage more than one printer with the same server.

What do you think?

Loup911, 2016-03-23:
Currently I have three computer watching the video stream in real time and mjpg_streamer consumes 7 to 20% of the cpu, and it never Is the same cpu which is used by octoprint.

During assembly of the video file I have one of my cpu saturated at 100% and I print in the same time and have my video feed open on 1 computer. And the printing did not freezes

gddeen, 2016-03-23:
I don't think that would fix it. (Random guess) If it's a bottleneck problem because of some locked resource, spreading it to two processes won't unlock the resource. It may complicate the discovery of a solution.  I traced mine down to a timer and file io, then got rid of time-lapse. I replaced my SD card. I moved to a RPI2B (from dual core 64bit). I upgraded to a better video camera. I got rid of video camera. I shielded all cables and put ground planes beneath the boards. I added ferrite cores around wiring onto the board. I had a lot of fun... (I didn't really spend any time LOOKING for the problem...)

ntoff, 2016-03-24:
I just tried on my pi2b and I got a spike of 27% for CPU usage though I don't have timelapse turned on. Once open though it settles down to about 1.6 - 5%

theturtle32, 2016-04-10:
I have the same problem. I'm running OctoPrint 1.2.10 on a dedicated RasPi 2 Model B, connected with ethernet. Nothing connected except the 3d printer and the network cable. Nothing else installed and running. I have to be very careful not to load/reload the OctoPrint web interface after a print has started or it will ruin the print.

Every time the Web UI is loaded, it pegs one core at 100% for about 10-15 seconds, during which time the printer's buffer starves out. The flow of G-Code from the pi slows to a crawl, and the print head will be mostly stopped but will stutter a little bit here and there until the web UI finishes loading, at which point the print resumes full speed.

I don't have (and never did have) a webcam connected, and my timelapse directory is empty.
I do have 369MiB of data in the uploads folder, however. Does it scan through everything in that folder when serving the web UI?

markwal, 2016-04-10:
@theturtle32 It will do a complete tree on the uploads folder in order to deliver the file list to the web UI.  The operation should be on the order of the number of items in the folder not so much the size.  It will also read a history yaml file containing information about the duration of the last print or whether it was successful or not.

chaseacton, 2016-04-20:
+1 

I am not using the time lapse feature, but I do have a camera attached. This is on a Pi 2.

nallar, 2016-05-23:
I'm having the same problem on a Pi 2. Time lapse is usually enabled with per-layer, but I tried turning it off and it still happens.

Are there any diagnostics or logs I can provide to help diagnose this?

rincewind0803, 2016-08-05:
Im getting the same effect lately since the last update to 1.2.15. 
But i never experienced this behavior before.
Steps to Reproduce:
-Start print Job ( with timelapse)
-Close Browser (Firefox/Archlinux)
-Open Octoprint in new Browser
Printer completely stops, Print interface unresponsive.
[https://www.dropbox.com/s/9li39b11esgcf76/octoprint.log?dl=0](url)
It does not happen if i leave the Browser open or do not reopen a new Window. I can watch printing progress with the octodroid app that does not make the printer stop.

thedoble, 2016-08-05:
I thought I would post here again, I had stopped using Octoprint because of this issue causing blobs in my prints. I've recently started using it again, and have updated to v1.2.15

I have disconnected my USB cameras completely, so there is no longer any additional CPU use, and there is no timelapse being done.

As mentioned by @rincewind0803 the printer will stutter or freeze completely when loading the web interface.

Thanks

rincewind0803, 2016-08-05:
After further investigation it may rahter be a raspbian problem. 
I switched to RepetierHost Sever for a print Job today and this one also stopped printing after reopening in Browser. SSH Connektion was alive and Pi was responsive but Repetier Server was killed.
I remember updating debian for the first time in months the same day I updated to 1.2.15. 
I`ll post al list of packages updated if someone is interested.

rincewind0803, 2016-08-06:
Things are getting misterious: I downloaded a new image no plugins and upgraded the hardware to a brand new Pi 3, only made the 1.2.15 update and getting trouble again.

luke321, 2016-08-28:
I got the same problem with 100% CPU usage on 3 differernt Pi 2. I think it started with version 1.2.7.
The printer does not get enough Gcode lines an stutters or even stops in place for a few seconds.
Currently I use the telegram plugin to check the status of the printer.

foosel, 2016-08-28:
Since I haven't seen this behaviour myself yet but it was already pointed out that OctoPrint will scan your upload folder when loading the file list via the API (which is part of what happens during a reload): how many files are people experiencing this have in there? And the same question for the timekeeper folder.

luke321, 2016-08-30:
Actually my problem was caused by the history plugin. Probably because of the long list of prints loading each time. 
I never had more than 4 files in the upload folder, no camera and no timelapse videos.
I will reinstall the history plugin and double check if the stuttering comes back and than post an issue in the plugin repo.

nallar, 2016-08-30:
Probably encountering the same issue here. I keep the uploaded files clean (currently 2 .gcodes there), but have hundreds of past jobs recorded by the history plugin.

foosel, 2016-08-30:
Yikes. That definitely looks like a likely culprit then. The login request probably takes so long because the server is blocked by the history request from just before it, that's just a wild guess right now.

Some proper caching headers on the history plugin's resource might already help tremendously here. 

Meanwhile I'm also looking into some better cache handling on the API's of OctoPrint itself, to reduce overhead here.

nallar, 2016-08-30:
Is it possible to have a separate process for communicating with the printer? Doing it in its own thread should be fine, but python still has the damn global interpreter lock so that doesn't actually work.

I understand it's likely to involve a lot of work, but it's probably the only way to avoid occasional reports from users of serial pauses in the long run.

In terms of just fixing the history plugin: It should set caching headers, as you've pointed out. It could also `time.sleep(0)` in the large loops it's doing, which would yield and allow serial communication to happen in the middle of it. maybe?

foosel, 2016-08-30:
@nallar I've spent a lot of time thinking about splitting it into multiple processes, but that will definitely be something that not only is a lot of work but also means a complete loss of backwards compatibility for some existing plugin extension points. The idea would be to move the whole serial communication into its own process (with high process priority) and have some fancy inter process communication between that and the server component. Problem is that plugins that depend on ANY modification or just tracking of what goes over the line or even replace the serial transport completely would stop functioning if they depend on shared memory - there's no way that I've been able to think of so far to have a plugin exist in both processes and transparently share state without side effects regarding compatibility.

And that's why it's currently pushed far far down the TODO list, since while it indeed is something I need to find a solution for in the long term, it's such a breaking change that it's out of the question to look closer at this currently with everything else that's already in the pipeline.

But yes, I curse the GIL daily ;)

sfsdfd, 2016-09-13:
Contributing some additional data about resource caching.

I reloaded OctoPrint three times, at about 30-second intervals (since each load took about 20 seconds), and kept an eye on Chrome's network tab. OctoPrint was printing at the time, so some things obviously changed - but not a ton in the span of 70 seconds.

Results:

http://imgur.com/a/vJzm2

Objects that are cached during successive loads:

* stylesheets, images, and a font
* some packed libraries
* less.min.js and Worker.js

Objects that aren't cached during scucessive loads:
* info, files, printerprofiles, slicing, timelapse, connection, custom, settings, history, profiles, login, automaticshutdown, logs, check, pluginmanager, channels
* the currently printing G-Code

That's kind of a lot of data to exchange, isn't it?

The files that caused the most delay, and some thoughts about each:

* gcode (13.5 seconds)
This item has ?ctime=___ appended to it, which of course means it reloads every moment. And it's the same file every single time - and exactly the same file that I uploaded. Seems that the parameterization is causing this item not to get cached. This is easily the biggest source of delay in loading OctoPrint, and maybe the most easily avoided by removing the URL parameterization.

* automaticshutdown (3.96 seconds)
* login (3.82 seconds)
* timelapse?unrendered=true (1.30 seconds)
These are just static scripts, right? No identifiable reason why they're not caching.

* settings (1.31 seconds)
This contains all of the settings. Again, they're not changing at all from one instance to the next, so there's no reason it shouldn't be cached.

* history (3.19 seconds)
* history (2.96 seconds)
This history gets updated whenever a new print gets queued, so it's not changing while the same print is in progress. No reason why it's not cached. But worse than that - the exact same resource is being fully loaded twice, and imposing a significant hit each time.

That's the bulk of it. Everything else loads in less than 1 second, and much of it concurrently. Many of those could be resolved through caching as well - and some of them would probably be easy, if techniques developed to cache the big objects can be equivalently applied to smaller, static ones as well. 

Resolving these cache misses could likely reduce the reload time of OctoPrint from 18+ seconds to 3-4 seconds, and maybe even less. Not only would it make OctoPrint *much* snappier - it would greatly reduce the load on the RPi during page refreshes, maybe reducing printer artifacts introduced by the load spike.

foosel, 2016-09-16:
I've already raised its priority and spent several days a couple weeks back adding caching to most of the resources you mentioned above where it made sense and was feasible, that's what I meant with

> Meanwhile I'm also looking into some better cache handling on the API's of OctoPrint itself, to reduce overhead here.

a couple of comments upwards ;) 

That stuff will be released with 1.3.0, not the current 1.2.x maintenance releases however, since just adding caching headers is only half the story, the tricky part is getting the cache invalidation to work properly, and if stuff goes wrong there, things WILL break - too risky for stability to put into a regular maintenance release.

Note that it only makes sense to cache GET requests though, everything else IS expected to change. And for some resources (e.g. the file list) reliably figuring out IF something changed or the cache is still valid is already something that depending on the underlying file system and available system resources can take some time. In case of e.g. the list of available printer ports it can become actually outright impossible to figure this out without doing the same work I'd need to do in order to fulfill the uncached request - in which case caching kinda stops making sense.

As a side note, I also learnt that there's either a jquery or a browser bug when it comes to handling consecutive requests to the same resource if ETag headers come into play - cost me a day.

Hope that makes it clear why - while it appears to be a low hanging fruit - it is not being tackled on the `maintenance` branch but instead rather pushed into the next big release. A better caching policy for the API endpoints in OctoPrint's core (I don't have control about plugins) IS being worked on though, so please, patience :)

nallar, 2016-09-19:
I think @imrahil may have stopped working on the print history plugin. No development on it since January, no updating of issues. It's a shame since lots of people are using it and it'll be causing print freezing.

imrahil, 2016-09-19:
Hi folks! Hi @nallar! 
"The reports of my death are greatly exaggerated" :) 
I'll try to sort this out in next few days!

fiveangle, 2016-12-16:
I see this is closed, but still very much an issue with 1.3.0; exacerbated by 2.5x jump in print speed now available with LINEAR_ADV in Marlin RC8.  Oh well, I see it's on a lot of people's minds beside mine.

thedoble, 2016-12-16:
Hi @fiveangle 

I have only done a couple of tests but so far it seems that disabling the 'history' plugin has resolved my issue. I suggest you give it a try if you are having the same problem.

Thanks everyone for their discussion and help with this, and thanks @foosel for her great work as always.

foosel, 2016-12-16:
@fiveangle also, THIS issue here hasn't been closed (even though so far all reports that actually did came back appear to indicate the history plugin). The ticket in the history plugin's tracker however has been closed, so anyone still experiencing issues like this here when this plugin is installed and active in the current version (of both OctoPrint and the plugin) and which vanish when the plugin is disabled or uninstalled should please report this to @imrahil. I can't solve issues with a plugin I do not maintain myself.

chaseacton, 2016-12-16:
I can confirm disabling the print history plugin on both of my printers fixed this issue for me. I can also confirm that this issue is no longer present with the latest version of the plugin.

fiveangle, 2016-12-19:
Disabling print history plugin did not resolve for me, but I have too many variables to state if it is OP yet.  Should have some time to test/eliminate in the next few days.

fiveangle, 2017-01-16:
In my case, it looks like it is related to octopi installation where octoprint goes to "sleep" or is otherwise not available and when refreshing page, haproxy instances are launched which temporarily starve the octoprint process for resources enough that it can't keep up with serial coms to Marlin, causing printing to stutter.

I'm still investigating but I moved back to 1.2.0-dev-649-gaa0e7ae6 (beta) by loading an old backup image of SD card.  With no other changes, I don't get the same behavior, but then I realize that's too far back to probably be of any use, so I'll keep plugging away.

Once thing I noticed is that tweaking Octoprint process to startup with priority "--realtime" (daemonstartstop options) helped reduced the stuttering since it helps prioritize OP resources over the haproxy ones launching (I tried setting haproxy daemon options to "--idle" but haproxy would not adjust priority even though it would launch).

I even upgraded to Pi3 to try and resolve (from Pi2) but didn't fully on the fastest prints with lots of small curves and LIN_ADVANCE option enabled on Marlin.

I do like the plugin features of OP (which are not available in 1.2.0 beta branch I have working) so I will try and figure this out more when I have time.

nophead, 2017-01-16:
When haproxy uses a lot of CPU time I think it is actually the wifi driver
that is using all the CPU streaming the video. I have to limit the frame
rate to 4 fps when using wifi.

fiveangle, 2017-01-17:
No, that is not it.

markwal, 2017-01-17:
Ah. That makes sense for some other reports I've heard.  I wonder if there's a bug such that the gcode viewer is downloading the file on page load (It's supposed to wait until you click on the tab). I've been noticing that it is already ready when I click over.  Over wifi that download would probably show up in haproxy's process if the Wi-Fi driver is eating it.

foosel, 2017-03-08:
@markwal actually, it never was coded to not load until the viewer is opened, only to stop life rendering when it's not visible. But I changed that in the above commit now since it definitely should help mitigate an initial bandwidth spike, especially if all you want is the current progress or something. So that might also help with bandwidth induced CPU load on page load.

markwal, 2017-03-08:
Ah. Oops. I guess I overgeneralized from some other tab switching code.  But nice feature, I think it should help some of these load spikes.

krisrok, 2017-07-30:
just wanted to chime in, it's still happening with the newest builds.

ctrl + f5 (or loading the website with a new browser) results in octoprint process going up for 5-10 secs. 
meanwhile, the print gets paused. after the huge startup delay there is some minorish stuttering while building/loading the website.

RPi 2B, connected via ethernet, pi camera connected

foosel, 2017-08-21:
@krisrok have you tried safe mode? 

I'm not sure what browser you are talking about here and I might be mistaken, but I think that Ctrl+F5 actually does enforce a cache refresh, which would explain performance issues. You are basically telling OctoPrint to throw out the (expensive to generate) cached copy of the webpage and generate it anew, which on an RPi2b depending on the printed model indeed will cause issues. That doesn't explain why it's also causing issues when you simply load the page in a new browser though. Please provide `octoprint.log`, that might contain some valuable insights here.

forkineye, 2017-08-24:
I just noticed an issue on my machine where the buffers were getting low (Wanhao I3) and it would start stuttering.  Logging into my Pi showed haproxy (and octoprint) at over 40% CPU each and a load average of over 5!  Close the tab in my browser left the haproxy usage high.  I had completely close the browser for haproxy to release the CPU.  Using Chrome  60.0.3112.101

klcjr89, 2017-09-11:
Through trial and error I've found out what can cause stutters and lag. The ethernet port is on the same bus as the USB, so it's actually better to use WiFi. If you have a USB webcam, well that's also on the same bus as the printer, so reduce your resolution and/or framerate and that should help quite a bit as well. This was tested on the Raspi 3 model B.

I have since upgraded to a 32 bit smoothieboard and unfortunately don't trust the hardware to keep the buffer full over usb, so I will have the Raspi dedicated to streaming 1080p videos and I will use the smoothie's crappy web interface for printing.

fiveangle, 2017-09-13:
Gina- Have you given thought to breaking out just the gcode->serial (usb) process so it could be run as a separate daemon and be given a high (highest) priority level ?  I know you said that would be too much work.  Do you still feel the same ? Printers aren't getting any slower.  I think everyone can accept dealing with the slowness of the UI if we have confidence that the code is being sent as real-time as possible (given the limitations of the platform, in this case Raspberry Pi).  Or is this just too much of an undertaking ?

foosel, 2017-10-09:
@fiveangle [See this](https://www.youtube.com/watch?v=JU_gZS1vJUs&t=2345s). It's not "too much work" that's the issue here, it's full blown backwards incompatibility. Making the comm layer modular is the first step towards this since it would allow coexistence of some backwards compatible layer and something separate/native - and something I keep getting distracted from by the regular "every day", regular maintenance and high prio bugs and such, which is why that is still an ongoing task.

dragonnn, 2017-10-30:
Sometimes you need to break backwards compatibility. My octoprint is running on Atom 1.6GHz and I am having this problem too. And the CPU don't hit 100% usage when loading the website, the load is between 50-70% max.

Diaoul, 2017-11-03:
I got this issue too running on my RPi 1. No issue when using Repetier Host (so smooth), no issue (or almost?) when I close the web UI. However as soon as it is open, the print stutters in the small holes and curves.

Personally I don't mind backward incompatibility, people that don't want the update can just keep the current version. Besides this issue, OctoPrint always gave me satisfaction for the past years.

Maybe something less radical could be done in the meantime like decrease the load of the webui, maybe in the terminal tab something can be done?

universam1, 2018-08-20:
Closing existing issues in favor of an apparently dead thread does not really solve the problem. 
Is there any idea what should help?

marcboivin, 2018-08-20:
It's not a simple matter. The comm stuff is really core to Octoprint and it's not a trivial thing to modify that heavily. Plus if we get into threading there are a lot more things that can go wrong. 

Only @foosel can tackle this with any hope of success. And if it's breaking compatibility, we would then talk about a V2 feature. 

We could maybe try to make new page load a bit more gentle on the CPU by deferring loading of some components and cache more aggressively. A more achievable endeavor, but not much guarantee.

foosel, 2018-08-21:
OctoPrint is already caching pretty aggressively (unless you keep forcing a full blown refresh, see above). The HTML code itself should only ever be generated once per targeted URL and locale (and that will even happen on server startup unless people have the preliminary cache disabled). The individual REST requests used to get file lists and whatnot from the backend also make heavy use of caching headers and validation checks. There are no really any low hanging fruits left here I fear, but I'd be happy to be proven wrong. 

I can't vouch for the caching behaviour and resource handling of any installed third party plugins however (which in the past have been the source of stuttering during page load, see above), so the first step towards debugging any kind of stuttering issues should also always be running in safe mode and checking if the issues vanish in that case - if so, a plugin is putting too much load on things.

> Closing existing issues in favor of an apparently dead thread does not really solve the problem.

Opening one ticket after another about the same phenomenon causes tremendous overhead for me, taking away a ton of time for ticket management that could otherwise be used for fixing issues or general improvement for anyone. All I'm asking for is to please keep the discussion and reports of one thing to one ticket so I don't have to spend hours every week trying to navigate the tracker and clean up duplicates.

universam1, 2018-08-21:
Thanks for the effort put into this project, OP is for sure a one-of-its-kind that has proven faithful for a long time!!
Hence I am really sad that the current situation forces me to think about alternatives. It's too bad that this i damaging basically every print if I am impatient to wait until the end of the print before going to its web page. But unfortunately this is a serious issue.

I disabled all plugins already because it smelled to me, but I will do a clean safe run as well according to your request.

I've also traced in detail the load on the system and it is clear that this is not the reason at all. So digging deeper to optimize the load will not bring the benefit. 
Nevertheless, something is blocking the comm handling routine and probably only @foosel will know what changed compared to older versions.

Threading that of from other workload is definitely key for the future success even if it might break something for a certain time. Timecritical stuff just cannot happen in a cooperative single threaded concept in a safe way - granted that I am not the expert here.

JimJamUrCode, 2018-09-03:
I spent countless hours trying to debug my printer to find out that loading the web interface to check on a print was causing defects in my prints. I really do love OctoPrint. I have 3 printers that are all being controlled by OctoPrint via Raspberry Pi 3. Each and every one of them has this issue. Fresh OctoPrint installs don’t change anything. Is the history plugin installed by default? I am going to try to boot into safe mode and see if that helps. Thanks for all of your hard work.

burner-, 2018-09-16:
I can confirm that I have same problem. I use latest marlin firmware with printer. When I open marlin web page print will not only lag or freeze but it can skip G-code rows. If I close and open web page at first layer and I have z-lift enabled it can cause skipping for lift row and then it mash head to table.

burner-, 2018-09-16:
Actually I just notice that if this is some kind communication problem with printer & octoprint this needs also more information what board people use. I use ultratronic and I notice that there is some odd code at fimware 
#if ENABLED(ARDUINO_ARCH_SAM)
 #undef TX_BUFFER_SIZE
  #undef RX_BUFFER_SIZE

So if you have this problem pleas tell what board and firmware you use with it. Lets try find some kind common factor why some of users have this problem and why some users does not have this problem.

[23 more replies]