OctoPrint/OctoPrint #1242: OctoPrint failing to start with custom gcode scripts

jminardi, 2016-02-25:
#### What were you doing?

Adding some custom gcode scripts to my config

#### What did you expect to happen?

OctoPrint to start without a problem

#### What happened instead?

OctoPrint failed to start

#### Branch & Commit or Version of OctoPrint

1.2.9

#### Printer model & used firmware incl. version

N/A

#### Browser and Version of Browser, Operating System running Browser

N/A

#### Link to octoprint.log

empty because octprint fails to start

#### Link to contents of terminal tab or serial.log

empty

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s) showing the problem:

I have read the FAQ.

Here is the error when I try to manually start octoprint:

```
Traceback (most recent call last):
  File "run", line 22, in <module>
    octoprint.main()
  File "/home/pi/OctoPrint/src/octoprint/__init__.py", line 81, in main
    octoprint.run()
  File "/home/pi/OctoPrint/src/octoprint/server/__init__.py", line 163, in run
    s = settings(init=True, basedir=self._basedir, configfile=self._configfile)
  File "/home/pi/OctoPrint/src/octoprint/settings.py", line 68, in settings
    _instance = Settings(configfile=configfile, basedir=basedir)
  File "/home/pi/OctoPrint/src/octoprint/settings.py", line 394, in __init__
    self.load(migrate=True)
  File "/home/pi/OctoPrint/src/octoprint/settings.py", line 587, in load
    self._migrate_config()
  File "/home/pi/OctoPrint/src/octoprint/settings.py", line 600, in _migrate_config
    dirty = migrate() or dirty
  File "/home/pi/OctoPrint/src/octoprint/settings.py", line 623, in _migrate_gcode_scripts
    self.saveScript("gcode", name, script.format(**replacements))
KeyError: "% snippet 'disable_hotends' %"
```

The problem is that it is trying to call `format()` on a string that includes jinja templates.

If I remove the two gcode scripts that have template placeholders in them then OctoPrint starts just fine.

Here is my full config:

```yaml
accessControl:
  enabled: false
server:
  firstRun: false
api:
  allowCrossOrigin: true
appearance: {}
feature:
  sdsupport: false
  waitForStartOnConnect: false
gcodeViewer:
  enabled: false
serial:
  port: /dev/ttyACM0
  baudrate: 250000
  autoconnect: true
  timeout:
    connection: 2
appearance:
  disabled:
    settings:
    - serial
    - printerprofiles
    - terminalfilters
    - section_features
    - features
    - webcam
    - api
    - folders
    - appearance
temperature:
  profiles:
  - bed: '50'
    extruder: '200'
    name: PLA
terminalFilters:
- name: Suppress M105 requests/responses
  regex: '(Send: M105)|(Recv: ok (B|T\d*):)'
scripts:
  gcode:
    afterPrintCancelled: "M410\nG4 S2\nG91\nG1 Z10 F500\n\nM42 P2 S0\nM42 P75 S0\n\n; disable motors\nM84\n\n;disable all heaters\n{% snippet 'disable_hotends' %}\nM140 S0\n\n;disable fan\nM106 S0"
    afterPrintDone: ";disable all heaters\n{% snippet 'disable_hotends' %}\n\nM140 S0\nM106 S0\nM42 P2 S0\nM42 P75 S0"
    afterPrintPaused: "G91\nG1 Z10\nG90"
    afterPrintResumed: "G91\nG1 Z-10\nG90"
webcam:
  stream: /webcam/?action=stream
  snapshot: http://127.0.0.1:8080/?action=snapshot
  ffmpeg: /usr/bin/avconv
plugins:
  _disabled:
  - cura
  cura:
    cura_engine: /usr/local/bin/cura_engine
  discovery:
    publicPort: 80
  softwareupdate:
    checks:
      octoprint:
        update_folder: /home/pi/OctoPrint
    octoprint_restart_command: sudo service octoprint restart
    environment_restart_command: sudo shutdown -r now
system:
  actions:
  - name: Shutdown
    command: sudo shutdown -h now
    action: shutdown
    confirm: You are about to shutdown the system.
    async: true
    ignore: true
  - name: Reboot
    command: sudo shutdown -r now
    action: reboot
    confirm: You are about to reboot the system
    async: true
    ignore: true
  - name: Restart Control Interface
    command: sudo service octoprint restart
    action: restart
    confirm: You are about to restart the Control Interface
    async: true
    ignore: true
  - action: update-automation-scripts
    command: cd /home/pi/.mecodescripts && git fetch && git reset --hard origin/master
    name: Update Automation Scripts
  - action: update-printer-ui
    command: cd /home/pi/printer_ui && git fetch && git reset --hard origin/master
    name: Update Printer UI
```

foosel, 2016-02-25:
You need to escape the jinja brackets. `{{% ... %}}` should work. Or even better, just put the file into the gcode script folder which id's were that stuff is now living, then it doesn't have to go through the migration step (which expects the old, not jinja backed format, and moves the old scripts as individual files to the new place).

jminardi, 2016-02-25:
Thanks, your suggestion of escaping worked. I am also going to just put them into the gcode folder.

closed 2016-02-25 by jminardi