OctoPrint/OctoPrint #1243: Octoprint disconnects from printer upon print completion

eaterofnames, 2016-02-26:
Once my Kossel Mini (folgertech 2020 Kossel) completes any print, the print server reports that something when wrong, and disconnects from the printer. I hunted through `octoprint.log` and found the following section:
`2016-02-25 13:54:32,921 - octoprint.util.comm - ERROR - Something crashed inside the serial connection loop, please report this in OctoPrint's bug tracker:
Traceback (most recent call last):
  File "/home/keg/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.0.dev892_gcb687f0-py2.7.egg/octoprint/util/comm.py", line 1219, in _monitor
    self._handle_ok()
  File "/home/keg/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.0.dev892_gcb687f0-py2.7.egg/octoprint/util/comm.py", line 1242, in _handle_ok
    self._continue_sending()
  File "/home/keg/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.0.dev892_gcb687f0-py2.7.egg/octoprint/util/comm.py", line 1247, in _continue_sending
    self._sendNext()
  File "/home/keg/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.0.dev892_gcb687f0-py2.7.egg/octoprint/util/comm.py", line 1554, in _sendNext
    line = self._getNext()
  File "/home/keg/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.0.dev892_gcb687f0-py2.7.egg/octoprint/util/comm.py", line 1521, in _getNext
    line = self._currentFile.getNext()
  File "/home/keg/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.0.dev892_gcb687f0-py2.7.egg/octoprint/util/comm.py", line 2196, in getNext
    raise ValueError("File %s is not open for reading" % self._filename)
`

Other errors reported in the log are here:
`2016-02-25 05:32:35,825 - octoprint.util.comm - ERROR - Unexpected error while reading from serial port
Traceback (most recent call last):
  File "/home/keg/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.0.dev892_gcb687f0-py2.7.egg/octoprint/util/comm.py", line 1491, in _readline
    ret = self._serial.readline()
  File "/home/keg/OctoPrint/venv/local/lib/python2.7/site-packages/pyserial-2.7-py2.7.egg/serial/serialposix.py", line 475, in read
    raise SerialException('device reports readiness to read but returned no data (device disconnected or multiple access on port?)')`

ntoff, 2016-02-28:
Are you using Repetier firmware? I just encountered this too, never seen it while using Marlin but I recently swapped to Repetier firmware and I just had this happen to me.

eaterofnames, 2016-02-28:
I'm using Marlin.

ntoff, 2016-02-28:
yeah ok I just discovered mjpeg_streamer service had stopped too so I think something went wrong with my actual OS. I doubt our issues are linked. Sorry.

markwal, 2016-02-28:
@eaterofnames could we get a complete octoprint.log and serial.log from a short print that reproduces the problem?  Please post on a file sharing service and post links here.

MertMG, 2016-03-02:
its happening on my side sometimes. if timelapse is on it may fail. i dont understand the issue yet.

eaterofnames, 2016-03-02:
I'll link two sets of logs here. One is from an older laptop running Crunchbang. I initially encountered the disconnect on complete error on this machine. The other set of logs are labeled Octoprint-pi and serial-pi, they came off my Raspberry Pi 2 running Raspbian. 

Once I get home from work I'll run another test on both machines to make sure it's repeatable. 

[logs.zip](https://github.com/foosel/OctoPrint/files/155421/logs.zip)

markwal, 2016-03-02:
Your serial.log's are empty because serial logging is turned off by default (they're a load on the server when generating them and they get big fast).  So to get a serial.log, you have to turn it on in settings.

Your octoprint-pi.log has two errors in it that repeat:
1- it looks like octoprint was started when it was already running (or some other web server process) since it couldn't attach to the port to listen
2- Several times a download of a file (vase_3.gcode several times, but also OctoPrint's css) were cancelled before the download was complete.  Are you having wi-fi connectivity issues also?

The Chrunchbang error is the one you posted originally and that looks like a classic serial connection reliability problem.  That could be the position of your USB cable, the USB cable's construction itself, internal wiring to the printer producing too much noise, etc.  Some things to try there are to swap out the USB cable and change the position of the host computer.

eaterofnames, 2016-03-02:
Thanks! I'll turn serial logging on when I get home, and run some more test cases on both machines.

closed 2016-03-12 by GitIssueBot