OctoPrint/OctoPrint #1245: Dremel 3D Printer

ante1550r, 2016-02-28:
How can i connect my dremel printer with my octopi ?

thanks for Helping 
Have a nice day

ante1550r, 2016-02-28:
So if got the Dremel 3d Printer Model 3D20 .
An Raspberry PI B or 2 an wifi doongle edimax
Printer Version V1.3 20150720

i can connect myself on the octopi.local server .
but how must i configurate my Printer settings to conect printer to pi ?
serial config ?

foosel, 2016-02-28:
AFAIK the Dremel is a rebranding of a Flashforge Dreamer, which to my knowledge uses a proprietary protocol. So unless someone writes a plugin (if that is even possible, certainly not me), you cannot use OctoPrint with this printer.

closed 2016-02-28 by foosel

ante1550r, 2016-02-28:
ok thank you very much !

hoe someone can/will do that :)

jchristman75, 2017-01-11:
I am interested in this, as well..  Here is a thread, that started, and even has some code from Dremel (or really FlashForge): https://github.com/foosel/OctoPrint/issues/1263