OctoPrint/OctoPrint #1246: Octoprint requires Raspbian Jessie based distro to work with Raspberry Pi 3

alex-gibson, 2016-02-29:
I have been testing ahead of the launch today, and noticed that the standard OctoPi distro gives rainbow screen of death on the Pi 3, and I could not follow the instructions (due to errors) to install for Jessie, using latest NOOBS distro as at last night.
For hassle free OctoPrint on the Pi 3 I recommend starting with the latest Jessie nightly build, which worked, and I demo'd at the launch party :)

alex-gibson, 2016-02-29:
GitIssueBot has correctly identified misplaced ticket report, newbie mistake.  Will close and repost at OctoPi

closed 2016-02-29 by alex-gibson

foosel, 2016-02-29:
Just as a heads-up, [it needs a complete new build based on the february raspbian image, not only any jessie based one, and this is a known limitation](https://plus.google.com/+OctoprintOrg/posts/69gNyynp6wK). Got the news of the Pi3 at the same time as everyone else, can't drop everything anytime that happens to be ready as soon to launch as possible, so that will take some time especially considering that usually a base image change means stumbling across newly introduced issues due to kernel or library updates from past experiences.

alex-gibson, 2016-02-29:
Hi, cross posted in the proper place, but just to offer you my test resource, have Pi 3s and printers and would be happy to carry out regression testing.  It's a positive story really, I got it working based on the comments in known issues, to the point I was happy to demo it.  Just want to highlight what worked for me.  The Pi 3 makes a huge difference to Octoprint usefulness and I will build it into my printers.  It's a great project!!

foosel, 2016-02-29:
I hopefully should have one soon too, but it's definitely good to know to have more "eyes" on things, so big thanks for that offer!