OctoPrint/OctoPrint #1247: POST /api/printer/command should return  printer response

tiagojanz, 2016-03-01:
#### What were you doing?

I'm developing  a QT frontend for raspberry 7"  panel using jessie-lite octopi.

#### What did you expect to happen?

The single command should have return something, like for example when M114  is issued the software should had receive

Recv: ok C: X:0.000 Y:0.000 Z:0.000 A:0.000 B:0.000 C:0.000  E:0.000

#### What happened instead?

HTTP/1.1 204 No Content

#### Branch & Commit or Version of OctoPrint

Version: 1.2.9 (master branch)

#### Printer model & used firmware incl. version

any

#### Browser and Version of Browser, Operating System running Browser

any

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

reporting communication issues. Never truncate.]

#### Link to contents of Javascript console in the browser

#### Screenshot(s) showing the problem:

I have read the FAQ.

foosel, 2016-03-01:
Technical issue, can't be solved, see #911

foosel, 2016-03-01:
Btw, if all that you want is display the communication log somewhere (your described scenario somewhat sounds like THAT is actually what you want to do), take a look at the [push interface](http://docs.octoprint.org/en/master/api/push.html) (keep in mind it uses the sockjs protocol, so messages are wrapped in that). There's a prototype of a client for consuming that interface [on the devel branch](https://github.com/foosel/OctoPrint/blob/devel/src/octoprint_client/__init__.py#L18), including a simple [usage example](https://github.com/foosel/OctoPrint/blob/devel/src/octoprint/cli/client.py#L150) - that should make stuff easier in the future, but taking a look at that could already help you now.

foosel, 2016-03-01:
I just realized you'd marked that as brainstorming ticket. Somehow I was under the impression it was marked as request, hence the closing because it would have been a duplicate as such. Argh, sorry for that.

Reopening and interpreting as "how to best access the terminal output from an api client" :)

tiagojanz, 2016-03-01:
Thats OK, I'm struggling to follow your lead, I find it smarter, but the
python syntax is way too different from c++ (QT)
I'll report when I got something to show.

thank you

foosel, 2017-03-08:
Don't see anything left from my side for this ticket, closing it for now. Just request a reopen if this should still be current.

closed 2017-03-08 by foosel