OctoPrint/OctoPrint #1249: M110 gcode causes issues on OctoPi (Geeetech M201 printer)

neilcoggins, 2016-03-01:
M110 gcode results in line number issues with a number of printers (e.g. Geeetech M201). Salient log entries:

```
Send: N6 M105*33
Recv: ok B:11.8 /0.0 T0:12.0 /0.0 T1:12.0 /0.0 T2:0.0 /0.0 @:0 B@:0
Changing monitoring state from 'Operational' to 'Printing'
Send: N0 M110 N0*125
Recv: Error: Line Number is not Last Line Number+1, Last Line: 6
Recv: Resend: 7
Changing monitoring state from 'Printing' to 'Error: Printer requested line 7 but no sufficient history is available, can't resend'
```

It appears that this is an issue with the STM32 firmware, but since M110 is not supported on a variety of firmware flavours, the line number reset being performed should be a configurable option.

nophead, 2016-03-01:
How are line numbers reset on such a machine or do they increase forever?

What happens if the board gets reset?

neilcoggins, 2016-03-01:
Apologies - to conform with posting guidelines for new bug submission:

- I was attempting to print from OctoPi v1.2.9 stable on RPi 2 Model B, connected via USB2 to Geeetech M201 printer
- I expected the print to commence, but because the firmware doesn't understand the M110 command (or at least, not the additional parameters supplied by OctoPrint), it rejects the command and causes the "line number + 1" issue that is well-known with a variety of firmwares
- Browser is Chrome Version 48.0.2564.116 (64-bit) on Debian Jessie 8.3 (tried IE10 / Windows 7 as well - same issue)

Here's the complete log: [https://gist.github.com/neilcoggins/4d45dc6a1d8ac97105c7](url)

@nophead I've seen them go up to 1M (curiously, the print crashed at **exactly** 1M+1 lines)

I have no doubt that my particular issue is buggy firmware (the manufacturers readily admit as much), but since M110 isn't a fully-supported gcode (according to [http://reprap.org/wiki/G-code#M110:_Set_Current_Line_Number](url) anyway) I thought it would be prudent to flag this up, and suggest that non-universal gcodes should be optional...

neilcoggins, 2016-03-01:
@nophead apols - if the board gets reset, the line numbers start from 0 again.

TBH, I'm raising this more because M110 is non-universal (not supported on Sprinter and MakerBot for example), as opposed to complaining because my cheap-as-kits-but-I-got-what-I-paid-for printer isn't working (a great deal about the Geetech M201 doesn't work currently, but that's not related to OctoPrint).

nophead, 2016-03-01:
Yes the board would start counting from zero but how does Octoprint know to
start from zero again?

I have this issue currently with SD writes so I am proposing adding another
M110 to fix it.

foosel, 2016-03-01:
M110 is crucial enough for printer communication that it has to be considered universal at this point. There is no way otherwise to get communications going with a printer that a) requires line numbers to be sent and b) doesn't reset on serial connect. No matter what that wiki page says (which btw appears to be edited by a lot of people who have never looked into existing firmware implementations and instead just claim things to be true that in fact aren't, with the firmware developers likewise never looking on the wiki page or into other implementations either), it is also widely supported enough that you are the first person in over three years to complain about it being unsupported.

Firmware which doesn't support M110 could probably be gotten to cooperate by rewriting the M110 to something that is supported (our stripping off unsupported parameters). This is also how Makerbot compatibility is achieved.

Sorry, but the whole communication situation is already horrible enough considering that basically every minor variant of various firmware solutions does things just different enough to cause issues across the board, I really don't want to go down the rabbit hole of having to support ancient firmware derivatives within the core code on top of that. Things are pretty much unsustainable as it stands.

@nophead can certify that this is a hornet's nest that you've just poked there.

neilcoggins, 2016-03-01:
@foosel @nophead I am sure that it is a thorny issue, and can appreciate the complexity of the task in addressing it (I'm lead developer on a closed-source system that just passed the 1M lines of code mark after 9 years, so I can empathise).

It is clear that OctoPrint would be as much use as a chocolate teapot if it was constrained to the intersection of functionality provided by every printer available. It is also clear that OctoPrint would be an unmaintainable mess if it tried to **directly** support each firmware variant.

The problem is that the market is expanding rapidly because of cheap I3 clone printers from the Far East, a number of which appear to be using customised firmware that doesn't **quite** work the same as one of the open firmware variants.

The solution has to be twofold, really:

1. get a **published open standard** adopted by the printer manufacturers - this will cut down on the provision of incompatible custom firmwares over time;

2. build a more advanced profile mechanism for modelling printer functionality (by that, I mean a combination of a configuration file that defines the command set for a given printer, and any "workarounds" needed to hack particular printers to make them work - this would act as the glue for the printers that were released before the standard was adopted)

Would it be possible to hook into the serial code (presumably the "sendCommand()" function itself) and perform a check against an in-memory lookup table to see whether the command about to be sent is even supported by the target printer? Presumably then you can return a tri-state from "sendCommand()" ("success", "failure", "skipped/not supported")... (I haven't yet delved deeply into the OctoPrint code, so I don't know if that is even viable, so feel free to shoot the idea down.)

foosel, 2016-03-01:
I'm on my way into bed, so I'll make this a fast reply, sorry in advance.

First of all, this project isn't about catering to lazy companies in for the quick buck while the hype lasts. I try to support users who run into problems with their alibaba deals by allowing OctoPrint to be extended through plugins and keeping stuff as configurable as possible at the moment. But everything has limits.

To your points:

  1. Yes, I've been preaching along those lines for years and am still working up my courage and patience again (next to maintaining and developing OctoPrint) to give it another try after my first attempt at trying to get the community into boat with the idea of standardizing about a year ago was shot down because "nobody should own that" (agreed, nobody wanted to) and "but that restricts my experimental freedom" (which got us into this mess to begin with). So, preaching to a very frustrated choir there.
  2. See the `dev/commRefactoringNG` branch. Complete rewrite of the communication layer, not with config files but strategies. Epic ton of work, and second attempt after I had to pretty much scrap the first one since the deeper I dove into firmware differences, the more bugs and other issues I discovered lurking therein.

And about your question regarding extensibility: http://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-comm-protocol-gcode-phase

neilcoggins, 2016-03-01:
@foosel I think we're on the same hymn sheet here. Some points:

1. the "alibaba deals" are going to spread as the results from these printers become more respectable and the costs come down more - this is unavoidable, I suspect

2. the hooks you have implemented - wow! They encapsulate exactly the functionality I was trying to espouse - I'm just playing catchup here (having installed OctoPi a few days ago, and checking out the source before tea tonight)

3. I'll take a look at the branch you mentioned tomorrow - it's my bed time too!

If you are in agreement, I'll close this ticket (if I can figure out how - I'm new to github as well) - but think that the discussion is worth retaining in the ticket-tracking system (if for no other reason than I am probably the crest of the "alibaba deals" wave that will end up looking for a better solution than Repetier Host...

markwal, 2016-03-01:
As a data point, "unsupported" on MakerBot means that line numbers don't make it into the over the wire protocol (x3g) so are stripped off by most gcode converters.  So M110 is generally ignored harmlessly.  If your firmware "unsupported" it this way it'd be fine.

Out of curiousity, I wonder if @neilcoggins firmware accepts M110 in a different form, say:

M110 N0

Of course, demanding

M7 M110 N0

would be pretty useless

markwal, 2016-03-01:
Closing since @neilcoggins expressed that intent and there's nothing to fix here. Github allows the conversation to continue in closed bugs of course.

closed 2016-03-01 by markwal

neilcoggins, 2016-03-01:
@markwal I just checked...

N0 M110 N0 generates an error:
`Send: N0 M110 N0*125
Recv: Error: Line Number is not Last Line Number+1, Last Line: 5
Recv: Resend: 6
Recv: ok`

And yep, you guessed it:

Send: N6 M110 N0 *91
Recv: ok

Now -technically- that's correct according to the G-code wiki... :-)

nophead, 2016-03-02:
Here's an idea: perhaps Gina could simply document the protocol she wants
OctoPrint to accept and then let all the firmware maintainers mod their
firmware to comply with that or else explain to their customers why not.

I think it's rapidly becoming impossible to maintain something that caters
for an infinite number of broken dialects.

neilcoggins, 2016-03-02:
@nophead I think you're espousing the "published open standard" that I referred to above, and I agree completely. In this particular instance, though, I'm coming around to thinking that this actually is a bug.

I've read the NIST doc RS274/NGC (which appears to be the origin of GCODE), and it doesn't say _anything_ about the Line Number value being reset (except that the numbers don't even have to be in order - that's clearly a throwaway remark).

The RepRap Wiki page says:

> RepRap firmware expects line numbers to increase by 1 each line, and if that doesn't happen it is flagged as an error. But you can reset the count using M110

and:

> M110 N123
This example sets the current line number to 123. Thus the expected next line after this command will be 124.

The implication of this is that the Line Number should **not** be reset until the line _after_ the M110, and therefore this cheap-as-chips printer I have obtained is actually (in this single instance and ignoring the dozens of problems that **are** the fault of Geeetech) working as it should.

I'm happy for someone to point me to a more definitive reference for GCODE that contradicts this interpretation, but I have a sneaking suspicion some Far East engineer read the RepRap G-Code Wiki and obeyed it to the letter when writing the firmware I have.

As to what to do about it... I have a very limited understanding of GCODE, a reasonable understanding of serial communications, and 20+ years coding experience with published standards (ISO/IEC 11172/13818 primarily). I can see how having every manufacturer putting their own "spin" on standards is a pain in the rear (in my world, you simply wouldn't get passed on conformance testing, and therefore couldn't advertise that you comply with the standard - so game over for your business).

What you can't (or rather, **shouldn't**) do is decide to follow these manufacturers' poor decisions by adopting your own "spin", rather than strict adherence to the agreed-upon "standard" - otherwise you're directly contributing to the problem by adding your own broken dialect.

In other words, you can't complain about manufacturers using broken dialects when you don't adhere to a known dialect yourself. Instead, as @nophead says, a standard needs to be defined and agreed upon (even if only by @foosel), and a line in the sand drawn.

The extensible functionality of OctoPrint is (from what @foosel pointed me to last night) **fantastic** and should be capable of meeting just about any requirements that somebody with a non-compliant firmware might have (caveat: this is coming from someone - me - with a limited understanding, remember). The key is to get OctoPrint core to work to a known standard in the first place, and that requires such a standard to exist and be accepted by those developing OctoPrint...

markwal, 2016-03-02:
Does that wiki page bear any resemblance to any of those other standards you mention? I think not. That wiki page just isn't up to the task. And 3d printing is sufficiently different and Usage has drifted so far to make references to the NIST standard for CNC fairly useless too.  There should be a standard, there is not currently.  And when there is, I propose that your firmware has a bug.

nophead, 2016-03-02:
Having M110 needing to be sent with the correct line number in order to
reset it is useless. The whole reason for resetting it is to start from a
known line number when the two ends have got out of sync due to one being
reset and not the other.

The NIST gcode standard is a file format. RepRap hijacked it and added OK
replies and line numbers and checksums in order to turn it into a
communication protocol. It doesn't really work though because the OK is
sent after a random delay that can be minutes, so it is impossible to set a
sensible time out.

neilcoggins, 2016-03-02:
@markwal "My" firmware has bugs aplenty, thanks (e.g. M92 not supported being a particular thorn in my side for calibrating at the moment). 

OK, I'm going to disregard the Wiki - I had assumed that it would be kept in a reasonably accurate state, but obviously the Internet being what it is today that's unrealistic.  That just reinforces the requirement for **something** to be published as a standard - otherwise, this sort of situation will continue forever. In the absence of a published standard, let's refer to the actual firmware code...

**Marlin 1.0.2-1** requires that the line number be reset as of the line issuing the M110, and ignores any specified N parameter for the M110 _if a line number is provided in front of the M110 command_ (use of strrchr() instead of strchr() would fix this, by the look of things):
```
if(strchr(cmdbuffer[bufindw], 'N') != NULL)
{
  strchr_pointer = strchr(cmdbuffer[bufindw], 'N');
  gcode_N = (strtol(&cmdbuffer[bufindw][strchr_pointer - cmdbuffer[bufindw] + 1], NULL, 10));
  if(gcode_N != gcode_LastN+1 && (strstr_P(cmdbuffer[bufindw], PSTR("M110")) == NULL) ) {
```
**Repetier 0.9.28** ignores any provided line number (line number value is overwritten by N in parseAscii() function) and uses the N parameter for M110:
```
if(M == 110)   // Reset line number
{
  lastLineNumber = actLineNumber;
  Com::printFLN(Com::tOk);
  waitingForResend = -1;
  return;
}
```
**Smoothie (edge)** requires a line number and then ignores the N parameter for M110:
```
//Get linenumber
if ( first_char == 'N' ) {
  Gcode full_line = Gcode(possible_command, new_message.stream, false);
  ln = (int) full_line.get_value('N');
  int chksum = (int) full_line.get_value('*');

  //Catch message if it is M110: Set Current Line Number
  if ( full_line.has_m ) {
    if ( full_line.m == 110 ) {
      currentline = ln;
      new_message.stream->printf("ok\r\n");
      return;
    }
  }
```

So it would seem that the best solution OctoPrint can provide (without per-firmware customisation) is send a line number (that being N0 in all cases), but don't send an N parameter (which from my cursory glance at the firmware source works for all 3). Which (unfortunately for me and others in the same boat as me) is completely contrary to the Wiki (and thus to what the firmware in our "Alibaba cheap deal" printers is written around).

I never wanted to get into a bunfight on this. The purpose of the exercise (for me as a newcomer to 3d printing) was to understand **why** OctoPrint doesn't work with a printer that works happily enough with Repetier Host, and see if there is a (simple) remedy (because I love the idea of OctoPrint / OctoPi, and don't want my laptop stuck at the printer all the time).

OctoPrint is a very well designed and implemented offering, and I applaud all those who have contributed to it. I am merely disappointed to learn that the 3d printing community is not pulling together to standardise the way these things work.  Until that happens, 3d printing will struggle to transition into a mass-market product. None of which is the fault of anyone here, BTW.

neilcoggins, 2016-03-02:
@nophead I am not disagreeing with you, I'm merely pointing out that the Wiki (yes, I realise now that it's garbage, but it's the **only** "reference" out there - what else are cheap Chinese printer makers going to use in that situation?) states that this behaviour is wrong. My review of the firmware code that I just posted supports your position 100%, of course. What is needed is a **better** reference to be published, so that the sorts of issues I have in my printer's firmware become a thing of the past.

***

So, how do _I_ get OctoPrint to either:
  1. not send line numbers on M110 commands, or
  2. make it send the "incorrect" line number at the start of the command (i.e. last+1 instead of 0)

Any ideas? I'm guessing I need to reimplement/hook the `octoprint.comm.protocol.gcode.sending` or `octoprint.comm.protocol.gcode.sent` calls - correct?

nophead, 2016-03-02:
I don't know anything about hooks but surely option 2 makes no sense. If you know the correct line number why would you want to reset it?

neilcoggins, 2016-03-02:
I have no idea, but the **only** way to get OctoPrint to print is to reset the printer **after** loading the file into OctoPrint, but **before** pressing "Print".  It **always** fails to print with an unknown firmware error (caused by the M110) if you follow what I would consider the logical workflow:

  1. connect to the printer
  2. load the file
  3. print the file

If I reset the STM32 board between [2] and [3] it prints. Otherwise, the M105 messages (which are sent with increasing line numbers) lead to the **N0 M110 N0*125** command being rejected (because the firmware expects a different line number).

I would concur, though - why is it being sent at all, if OctoPrint has sent a bunch of M105s with increasing line numbers prior?  Why reset with an M110 at all?

nophead, 2016-03-02:
My M105s (or any other commands) don't have line numbers unless they are during a print. It needs to be that way so I can switch my printer on and off without restarting OctoPrint. Then when I start a print or an SD upload M110 ensures the line numbers start from 0.

nophead, 2016-03-02:
If there is some option to always send line numbers (which I can't find) then it would probably make sense to not reset them when starting a print. However you have the problem of them getting very large and wasting comms bandwidth.

I am thinking of getting rid of them as I have no comms errors so don't need them or checksums. They just slow things down.

neilcoggins, 2016-03-02:
I would have thought that comms errors were a thing of the past with USB-connected printers - or are the clone RAMPS boards et al generating so much noise that it causes interference?

Regardless, I'd be in favour of dropping both line numbers and checksums (or at least options to turn each/both off). Certainly not sending line numbers at all would appear to resolve the problem I am seeing.

nophead, 2016-03-02:
Yes Arduino Mega / RAMPS based boards still have comms errors I believe. Also when PCs are involved the USB drivers can cause errors.

I have found RPi + Melzi with a genuine FTDI chip to be rock solid with USB, or direct connection, when the RPI is powered from the 5V standby of the same ATX PSU that powers the machine. That way there are no ground loop issues.

neilcoggins, 2016-03-02:
That's excellent advice, actually - I hadn't even thought about how to power the RPi on the (soon-to-be-assembled) MPCNC, but powering from the same ATX supply would seem to be a very sensible idea (especially since there is vast spare capacity on the PSU @ 5V)... As would picking a better firmware to run on it than I have on my current printer! ;-)

nophead, 2016-03-02:
This is how I did it: http://hydraraptor.blogspot.com/2014/06/lights-camera-action.html. I use a couple of MOSFETs to allow the RPi to turn the PSU on and off and some lights.

I also did [this](http://hydraraptor.blogspot.com/2016/01/ultra-low-dropout-regulator-for-flicker.html) to stop the lights flickering.

markwal, 2016-03-02:
@neilcoggins that setting is in "features", but it is called "always send checksums". Checksums and line numbers go together. Most printers only want checksums while printing.

markwal, 2016-03-02:
@neilcoggins Also, there is a setting in Serial->Advanced that is "Always send checksums with these gcodes" and I think M110 is there by default.  Although I haven't checked whether that is a new feature in the devel branch.  Anyway, you might want to try:

* turn off always send checksums so that you are only doing checksums during printing
* turn off always send checksums on M110.

So that, for your printer it doesn't checksum M110.