OctoPrint/OctoPrint #1250: Upload Fails but Files are listed

135erHeli, 2016-03-02:
I got Problem with a clrealy new Octoprint Image.
Updated to 1.2.9
Raspberry PI B+
Repetier Firmware 0.92

If i will Upload i got the Message Upload fails, but in Refresh the file ist Uploaded and Printable.But if ill Print them at 99% the Print Stops due a missing Line Number.

[image]

[serial.pdf](https://github.com/foosel/OctoPrint/files/154547/serial.pdf)

markwal, 2016-03-02:
As @nophead pointed out on the other bug, this happens because your end gcode is improperly formed:
```
 TargetExtr0:0
2016-03-02 08:40:50,283 - SERIAL - DEBUG - Recv: Printed filament:
172.15m Printing time:1 days 11 hours 8 min
2016-03-02 08:40:50,286 - SERIAL - DEBUG - Recv: ok 10141
2016-03-02 08:40:50,291 - SERIAL - DEBUG - Send: N10143 G1 E-5
F300*7
2016-03-02 08:40:50,295 - SERIAL - DEBUG - Recv: TargetBed:0
2016-03-02 08:40:50,300 - SERIAL - DEBUG - Recv: ok 10142
2016-03-02 08:40:50,304 - SERIAL - DEBUG - Send: to release some of 
the pressure
2016-03-02 08:40:50,308 - SERIAL - DEBUG - Send: N10144 G91*17
2016-03-02 08:40:50,310 - SERIAL - DEBUG - Recv: ok 10143
2016-03-02 08:40:50,315 - SERIAL - DEBUG - Send: N10145 G1 Z10
F{travel_speed}*18
2016-03-02 08:40:50,325 - SERIAL - DEBUG - Recv: Error:Missing
linenumber
2016-03-02 08:40:50,327 - SERIAL - DEBUG - Changing monitoring st
```
Two problems: You've got comments that are not marked as comments (should start with a semicolon) and you have a token {travel_speed} that should have been expanded to a value so it is the wrong token for your slicer (they vary on this between cura and slic3r for example).

markwal, 2016-03-02:
The reason this works when you copy to SD card manually is that the firmware only complains about it in the context of line numbers and checksums and it only uses those to improve serial communication reliability.  They aren't in the SD card file, so reading it out, it just skips those lines and/or interprets them funny (maybe parses the {travel_speed} as 0 and then assumes that means no change or some such thing).

The reality is that there isn't much OctoPrint can do for you here.  It shouldn't, I think, complain about your gcode because if your printer likes it, it shouldn't get in the way, so it's up to the firmware to complain, which it did.

I think if you fix your end gcode in your slicer, not only will you not hit this problem, but also it'll be more likely to have the intended effect when printing.

135erHeli, 2016-03-02:
But the same Value is at the Start in the Gcode,why Octoprint stops working not in the beginning?

`; -- START GCODE --
G21        ;metric values
G90        ;absolute positioning
M107       ;start with the fan off
G28 X0 Y0  ;move X/Y to min endstops
G28 Z0     ;move Z to min endstops
G1 Z15.0 F1200 ;move the platform down 15mm
G92 E0                  ;zero the extruded length
G1 F200 E15              ;extrude 5mm of feed stock
G92 E0                  ;zero the extruded length again
G28 Z0     ;move Z to min endstops
G1 F{travel_speed}
;M301 P17.44 I1.07 D71.17 ;set the pid for 200 PLA
;M301 P15.07 I0.92 D61.50
;M97 A0.5 W1.0 P180
G92 E0
G1 E-7.0000 F5400
G1 Z0.320 F1002
; layer 1, Z = 0.45
T0
; tool H0.450 W0.480
; skirt
G1 X18.362 Y18.360 F6000
G1 E0.0000 F1620
G92 E0
G1 X18.629 Y18.093 E0.0339 F3000
G1 X18.690 Y18.038 E0.0413
G1 X27.695 Y10.648 E1.0874
G1 X27.761 Y10.598 E1.0948
G1 X28.075 Y10.389 E1.1287`

nophead, 2016-03-02:
Because your firmware doesn't give an error for ```G1 F{travel_speed}``` but it won't do what it should do, which is set the travel speed.

The error at the end ```to release some of the pressure``` is what causes the problem that makes it stop.

135erHeli, 2016-03-02:
ok maybe i can fix this in Simplify and remove all descriptions from the gcodes, but whats with the Upload Problem

nophead, 2016-03-02:
The problem is ```to release some of the pressure``` should have a ; in front of it to make it a comment.

markwal, 2016-03-02:
Just make sure that the comments are preceded by a semicolon.

And just put in a number for that {travel_speed}.  Something reasonable for your Z-movement, like maybe 1200

And that *is* the upload problem.  For upload, it needs to be properly formed so that it doesn't fail the line number checksum test.  For print, it doesn't do any such thing.

135erHeli, 2016-03-02:
No it isnt, i tryed a Cura and Slic3r file also and they also wont Properly uploaded
`; generated by Slic3r 1.2.9 on 2016-03-02 at 21:21:43

; external perimeters extrusion width = 0.20mm
; perimeters extrusion width = 0.34mm
; infill extrusion width = 0.34mm
; solid infill extrusion width = 0.34mm
; top infill extrusion width = 0.34mm

M107
M190 S80 ; set bed temperature
M104 S250 ; set temperature
G28 ; home all axes
G1 Z10 F1200 ; lift nozzle

M109 S250 ; wait for temperature to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
G1 E-7.00000 F1800.00000
G92 E0
G1 Z-0.800 F7200.000
.
.
.

G1 E5.70299 F1800.00000
G92 E0
M107
M104 S0 ; turn off temperature
M190 S0 ;  turn off bed temperature
G28 X0 Y0  ; home X axis
G1 Y190 ;
M84     ; disable motors

; filament used = 15805.8mm (40.2cm3)

; avoid_crossing_perimeters = 0
; bed_shape = 0x0,195x0,195x190,0x190
; bed_temperature = 80
; before_layer_gcode = 
; bridge_acceleration = 0
; bridge_fan_speed = 100
; brim_width = 0
; complete_objects = 0
; cooling = 1
; default_acceleration = 0
; disable_fan_first_layers = 6
; duplicate_distance = 6
; end_gcode = M104 S0 ; turn off temperature\nM190 S0 ;  turn off bed temperature\nG28 X0 Y0  ; home X axis\nG1 Y190 ;\nM84     ; disable motors\n
; extruder_clearance_height = 20
; extruder_clearance_radius = 20
; extruder_offset = 0x0
; extrusion_axis = E
; extrusion_multiplier = 1
; fan_always_on = 0
; fan_below_layer_time = 60
; filament_colour = #FF1400
; filament_diameter = 1.8
; first_layer_acceleration = 0
; first_layer_bed_temperature = 80
; first_layer_extrusion_width = 200%
; first_layer_speed = 30%
; first_layer_temperature = 250
; gcode_arcs = 0
; gcode_comments = 0
; gcode_flavor = reprap
; infill_acceleration = 0
; infill_first = 0
; layer_gcode = 
; max_fan_speed = 100
; max_print_speed = 80
; max_volumetric_speed = 0
; min_fan_speed = 35
; min_print_speed = 10
; min_skirt_length = 0
; notes = 
; nozzle_diameter = 0.2
; only_retract_when_crossing_perimeters = 1
; ooze_prevention = 0
; output_filename_format = [input_filename_base].gcode
; perimeter_acceleration = 0
; post_process = 
; pressure_advance = 0
; resolution = 0
; retract_before_travel = 2
; retract_layer_change = 1
; retract_length = 7
; retract_length_toolchange = 10
; retract_lift = 0
; retract_restart_extra = 2
; retract_restart_extra_toolchange = 0
; retract_speed = 30
; skirt_distance = 6
; skirt_height = 1
; skirts = 3
; slowdown_below_layer_time = 30
; spiral_vase = 0
; standby_temperature_delta = -5
; start_gcode = G28 ; home all axes\nG1 Z10 F1200 ; lift nozzle\n
; temperature = 250
; threads = 2
; toolchange_gcode = 
; travel_speed = 120
; use_firmware_retraction = 0
; use_relative_e_distances = 0
; use_volumetric_e = 0
; vibration_limit = 0
; wipe = 0
; z_offset = -1
; dont_support_bridges = 1
; extrusion_width = 0
; first_layer_height = 0.2
; infill_only_where_needed = 0
; interface_shells = 0
; layer_height = 0.1
; raft_layers = 0
; seam_position = random
; support_material = 0
; support_material_angle = 0
; support_material_contact_distance = 0.2
; support_material_enforce_layers = 0
; support_material_extruder = 1
; support_material_extrusion_width = 0
; support_material_interface_extruder = 1
; support_material_interface_layers = 0
; support_material_interface_spacing = 0
; support_material_interface_speed = 100%
; support_material_pattern = rectilinear
; support_material_spacing = 2.5
; support_material_speed = 60
; support_material_threshold = 0
; xy_size_compensation = 0
; bottom_solid_layers = 5
; bridge_flow_ratio = 1
; bridge_speed = 80
; external_fill_pattern = rectilinear
; external_perimeter_extrusion_width = 0
; external_perimeter_speed = 50%
; external_perimeters_first = 0
; extra_perimeters = 1
; fill_angle = 45
; fill_density = 20%
; fill_pattern = honeycomb
; gap_fill_speed = 40
; infill_every_layers = 1
; infill_extruder = 1
; infill_extrusion_width = 0
; infill_overlap = 15%
; infill_speed = 80
; overhangs = 1
; perimeter_extruder = 1
; perimeter_extrusion_width = 0
; perimeter_speed = 80
; perimeters = 3
; small_perimeter_speed = 40
; solid_infill_below_area = 200
; solid_infill_every_layers = 0
; solid_infill_extruder = 1
; solid_infill_extrusion_width = 0
; solid_infill_speed = 80
; thin_walls = 1
; top_infill_extrusion_width = 0
; top_solid_infill_speed = 60
; top_solid_layers = 5
 `

What should the Problem Now in this Gcode?There is Nothing in what should be calling the Problem before...

markwal, 2016-03-02:
You need to upload another serial.log for us to diagnose the same way, but likely there is some other problem with your gcode.

135erHeli, 2016-03-02:
Is mir zu blöd dann geh ich halt wieder auf Repetier Server ....

closed 2016-03-02 by 135erHeli

markwal, 2016-03-02:
There are some malformed lines in what you pasted, but that may have been just a copy and paste problem.  You could upload the gcode directly to dropbox/onedrive/etc and paste a link here, that would be easier to see.  But you've got funny line endings and extraneous dots in what you pasted.

foosel, 2016-03-02:
@135erHeli Tu was du nicht lassen kannst. Anscheinend willst du ja nicht geholfen bekommen, sonst würdest du ja die Informationen zur Verfügung stellen, die dazu benötigt werden.

foosel, 2016-03-02:
@markwal don't bother. He apparently wants his problem magically solved without helping to figure it out in the first place, because apparently that is too much effort and he'd rather go back to Repetier Server now because "this is stupid". Whatever flies his boat I guess.

135erHeli, 2016-03-02:
es ist alles da, in den logs wird nix über den fehlerhaften upload aufgezeichnet da er ja nach dem Datei reload die Datei anzeigt, das kann kein gcode problem sein.Wenn ich Octoprint unter Windows Installiere geht der upload ja problemlos

foosel, 2016-03-02:
OctoPrint cannot write the metadata of your upload because you moved the upload folder to some external filesystem and that apparently doesn't support file ownership, which hints at some misconfigured Windows vs Linux situation:

```
2016-03-02 11:38:54,766 - octoprint.filemanager.storage - ERROR - Error while writing .metadata.yaml to /media/Extern/.octoprint/ uploads
Traceback (most recent call last):
File "/home/pi/oprint/local/lib/python2.7/site-packages/
OctoPrint-1.2.9-py2.7.egg/octoprint/filemanager/storage.py", line
1013, in _save_metadata
shutil.move(file_obj.name, metadata_path)
File "/usr/lib/python2.7/shutil.py", line 301, in move
copy2(src, real_dst)
File "/usr/lib/python2.7/shutil.py", line 131, in copy2
copystat(src, dst)
File "/usr/lib/python2.7/shutil.py", line 100, in copystat
os.chmod(dst, mode)
OSError: [Errno 1] Operation not permitted: '/media/ Extern/.octoprint/uploads/.metadata.yaml'
```

Either fix the permission situation on the external storage or keep the uploads folder where it is located by default. Also make sure the upload folder and *all* files therein are owned by the user `pi` or you'll directly run into the next issue when trying to delete stuff.

Finally, keep the attitude in check. You don't pay a dime for the software or the support, so don't get cheeky with the people trying to help. Be glad there is support at all, there wouldn't be if people like @markwal and @nophead weren't willing to sacrifice their own free time to help others.

135erHeli, 2016-03-03:
Excuse my Reaction Yesterday,but i was totaly nerved after trying the whole Day to get this running.My Fault was the External HDD was formated in fat, User pi could read and write.Now ill formated them to ext4 and now it works,also the Print will 100% finished.Thx for help. Have a nice Day.