OctoPrint/OctoPrint #1251: [Request] No [...] for filtered text in the terminal window

ntoff, 2016-03-02:
Makes filtering rather redundant when the filtered lines are just replaced with dots under certain filtering conditions / terminal output. It'd be nice to maybe just have a sticky warning label (like the one in the timelapse configuration) warning us the terminal is being filtered, with no feedback in the terminal window itself.

http://i.imgur.com/7Ke6ypK.png

nophead, 2016-03-02:
+1 but it seems like they all go into the buffer anyway and are filtered for the display, so eventually all the information is flushed away by M105s.

ntoff, 2016-03-24:
I take it the filtering stuff is all done client side (i.e. in the web browser)? I was just wondering, would it overload octoprint to have that filtering done server side so the data is simply never sent to the client? 

Of course that would mean no client would ever get the full output unless it was enabled server side but I'm not sure that's necessarily a bad thing?

foosel, 2016-03-24:
It might not be a bad thing for you, but others would see it differently (myself included - I don't want stuff to be filtered in my browser because some other user currently looking at my server decides he doesn't like to see `M105`).

The stuff is decoupled like it is now for a reason - to leave it up to the client what to display and what not, individually, and allow the server to concentrate on what matters. That means that on connect to the server an initial log segment of whatever were the last 300 lines the server saw gets pushed, and after that things stay synchronized by getting pushed the deltas. That has the advantage that I don't have to transfer all data to all clients on every update or calculate client-individual deltas, but only the couple of updated log lines that were received and can have the client worry about whatever post processing it wants to do there, or not. Any server side post processing (such as filtering, even if it happened centralized) would also put additional load on the server - a central goal in OctoPrint's design is to minimize any impact of stuff that is not necessary for actual printing operations on the server and off-load that to the client instead (hence no gcode analysis while printing, no timelapse rendering while printing etc, client-side gcode rendering etc).

The dots btw have a reason too. If it only filters out every second line it doesn't really shrink the displayed bits, but it does add the information that something was filtered out there (which is very helpful for debugging purposes, both for the user and also when I have people copying the terminal contents into tickets and complaining that commands where not sent), and especially when multiple consecutive lines are filtered out (e.g. when the printer sits idle and you have M105 plus reply filtered you'll get one `[...]` for hundreds of request/response pairs) it minimizes the log but still has it keep the information that things happened but were removed from display.

The whole filtering functionality was added in order to allow people to remove request/response pairs from the output they aren't that interested in... removing clutter so to speak. It was never meant for filtering basically one half of the whole conversation, like you are pretty much doing by adding `ok` to the filters. 

The other complaint, that stuff starts scrolling out anyhow due to the filtered lines becoming invisible... well, that's something we can discuss, but the problem here is that you'll most certainly find two parties here, those that prefer being able to "unfilter" on the fly (like now, because the data is only made invisible, not actually thrown away) and those that would rather always be able to see 300 lines and throw away altogether what doesn't interest them at the moment, even if it means not being able to retroactively unfilter already received lines of the log in case horrible things happen.

For debugging purposes, the first variant is the better one, because it means that if stuff stall and breaks, I can disable the filters and still see what happened. I guess it depends on how you use the terminal (personally, I only use it for checking things in case of errors and of course for debugging adjustments to the serial communication layer).

I'll happily accept PRs that add alternatives, without forcing those alternatives on people, don't impact server load and also keep the client from exploding however.

nophead, 2016-03-24:
It would be nice to be able to switch off the M105 and M27 requests rather
than suppressing them. It makes manual interaction with the firmware for
debugging much easier. Obviously it would prevent the temperature graph and
the time left updating. Before OctoPrint I used Pronterface and that had
that option. IIRC it also had a button to press to trigger an update so you
could see the current temperature if you needed to.

foosel, 2016-03-24:
That sounds rather like a completely different feature request to my ears ;)

But just for the record: When I say "debugging", I don't mean as in "during development" but in the field. If the past couple of years of maintaining this project has taught me anything it's that users lie. Not out of mean spirit, but simply because they don't know better. So if I tell them "Have you filtered that log" they'll say "no" even if they have every single command and firmware response filtered. Maybe because they didn't understand the question, maybe because they forgot, or maybe because they didn't pay attention to the question.

Most people don't understand that they need to enable the serial.log in order for it to not be empty. So in most cases the contents of the terminal tab is all that's left for a bug report. Hence I'm trying to not have that throw more stuff away than necessary, and making it obvious if stuff is missing from that due to filters etc.

ntoff, 2016-03-24:
>It might not be a bad thing for you, but others would see it differently (myself included - I don't want stuff to be filtered in my browser because some other user currently looking at my server decides he doesn't like to see M105).

I meant It would be a server side setting so non user settable so if you're the admin you'd see whatever you wanted to see and users get no say in it.

closed 2016-03-24 by ntoff