OctoPrint/OctoPrint #1252: [Request] Hotplugging (Automatic USB Detection/Connection)

mirage335, 2016-03-02:
Please add an option to automatically connect whenever the USB device is detected, rather than just at startup. This will make it easier for people to bypass the print server and plug in directly through USB when desired.

Please note, I am asking on behalf of a Makersmiths operator currently using Repetier Server, as well as for myself.

markwal, 2016-03-02:
The PortLister plugin could be easily modified to provide this functionality.  That is in a plug-in because the connection detection is not platform independent (ie currently won't work on Windows).

markwal, 2016-03-02:
Oops.  Forgot I actually already added auto connect to PortLister.  So if you have a pi (or I suspect any linux host), installing and configuring PortLister should do this for you.  It currently has a 20 second delay from when it notices the port until it tries to autoconnect.  That delay is only configurable via config.yaml at the moment.

http://plugins.octoprint.org/plugins/portlister/

mirage335, 2016-03-02:
Indeed, that provides autoconnect, but does not offer the requested auto*re*connect functionality. Have successfully tested PortLister's autoconnect functionality though, thanks for the quick reply.

markwal, 2016-03-02:
It should auto*re*connect if the autoconnect feature is turned on, does it not?

mirage335, 2016-03-02:
No, it did not, and autoconnect was enabled.

markwal, 2016-03-03:
Hmmm... I can't reproduce.  If I turn off the printer, OctoPrint eventually disconnects.  If I turn the printer back on, OctoPrint reconnects.  To the plugin, there's no difference between "connect" and "reconnect".  It just tries to "connect" 20 seconds after it sees the port show up (if it is the port selected on the last manual connect).

mirage335, 2016-03-03:
Since my Raspberry Pi is wired to the RAMBO 5V supply, I cannot shutdown the printer independently of the server. However, physically disconnecting and reconnecting the USB port does not result in automatic reconnection. Is there any way to gather more debugging information?

markwal, 2016-03-03:
I'll see what I can come up with.

markwal, 2016-03-04:
I updated PortLister with more logging.  They should show up as INFOs and WARNINGs in the octoprint.log every time a new port is detected.  In testing unplugging and plugging in the usb cable here, it only works if the cable is unplugged long enough for OctoPirnt to notice and fail the connection before reinserting the cable.  This is because PortLister is trying to avoid changing ports when you've already got a good connection, so if OctoPrint thinks the connection is still good, no autoconnect will occur.  Also, if the OS decides to change the port name, it will also not reconnect (and I couldn't figure out when the OS decides to do that).

By the way, is your port set to AUTO?  That might be why I'm seeing something different.  AUTO doesn't work with my printer.

mirage335, 2016-03-04:
Thank you for updating the plugin to include debugging information. Upon (re)connection of USB cable to the printer electronics, this is what the log files show.

`2016-03-04 21:23:06,697 - octoprint.plugins.portlister - INFO - on_port_created connection_options {'portPreference': '/dev/serial/by-id/usb-UltiMachine__ultimachine.com__RAMBo_74031303437351C05161-if00', 'autoconnect': True, 'baudrates': [250000, 230400, 115200, 57600, 38400, 19200, 9600], 'ports': ['/dev/serial/by-id/usb-UltiMachine__ultimachine.com__RAMBo_74031303437351C05161-if00', '/dev/ttyACM0'], 'baudratePreference': 250000}
2016-03-04 21:23:06,706 - octoprint.plugins.portlister - INFO - Checking if /dev/ttyACM0 is in ['/dev/serial/by-id/usb-UltiMachine__ultimachine.com__RAMBo_74031303437351C05161-if00', '/dev/ttyACM0']
2016-03-04 21:23:06,714 - octoprint.plugins.portlister - INFO - autoconnect_delay 20`

Have tried setting the port to AUTO, /dev/ttyACM0, and /dev/serial/by-id/<id>. In all cases, autoconnect does not occur.

markwal, 2016-03-04:
That's interesting.  That last line said, ok, it's my intention to autoconnect, I'm gonna do it, but it's gonna take at least 20 seconds before I start and probably more like 30 seconds before you notice in the browser.

20 seconds later, it should say "Attempting to connect to..."
In the case of the log that you pasted, the reason it doesn't is because your 'portPreference' doesn't match the discovered port.  My intention there was to avoid autoconnecting to things that are not printers, but I forgot there are synonyms.  I might be able to do something about that by asking the OS for a canonical name.  In the meantime, here's what you do:

* Disconnect using the UI.
* Set the port to /dev/ttyACM0
* Check the boxes that say "Save connection settings" and "Auto-connect on server startup" (I think this is the key bit that will update your 'portPreference', if you just connect to a different port it doesn't change the preference)
* Click connect

Now, when you open Settings, the selected port should be /dev/ttyACM0 and the log should put the portPreference to match the discovered port.  Hopefully.

And then later on I'll attempt to make the thing work with aliases.

foosel, 2016-03-10:
Does anyone have any objections to close this issue? I mean, the core problem is solved through the existing plugin, so there's not much "request" left?

mirage335, 2016-03-10:
Pardon, have been extremely busy the last few days. Please give me a chance to verify this now works, now that markwal has determined what was not working.

mirage335, 2016-03-14:
Works now, after removing all other ports. Much thanks for helping with this. Still, adding the ability to auto-reconnect with /dev/serial/by-id device files would be nice.

closed 2016-03-14 by mirage335