OctoPrint/OctoPrint #1259: Octoprint not receiving temp updates while heating (bed or extruders)

leofaoro, 2016-03-06:
#### What were you doing?

Sent a gcode file to print. It goes thorugh the GPX plugin though. Happens with any file I submit.

#### What did you expect to happen?

Upon telling the printer to heat up, expected to have every 5s update of the temperatures. 

#### What happened instead?

It starts heating up the bed, and then no communication from the printer until the bed reaches target temp. Same with when extruder is told to heat up. The print time also stops running.

#### Branch & Commit or Version of OctoPrint

1.2.9 master

#### Printer model & used firmware incl. version

Qidi Tech I running Sailfish 7.8

#### Browser and Version of Browser, Operating System running Browser

Windows 7, Chrome 48. 

#### Link to octoprint.log

[On gist.github.com or pastebin.com. Always include and never truncate.]

#### Link to contents of terminal tab or serial.log

https://www.dropbox.com/s/o411tp7hzm2g5nr/octoprint%20logs.zip?dl=0

#### Link to contents of Javascript console in the browser

#### Screenshot(s) showing the problem:

https://www.dropbox.com/s/ok8ery6xudig8ec/Capture.JPG?dl=0

I have read the FAQ.

markwal, 2016-03-06:
Your serial.log doesn't cover the time in question.  It is from the end of the print and your problem occurs at the beginning of the print.  If there is a bug here, it is likely in the GPX plugin since it is the plugins responsibility to keep returning temperatures while waiting.  If you agree, you should close this issue and open a new one on the GPX issue list: https://github.com/markwal/GPX

I'll need a serial log that sends the M140 and M104's that set the temperature and the M116 or whatever you're using that waits for the temperature and the response in order to properly diagnose.  No need to wait until the print is finished, you can download the serial.log at approximately the same time as you see the result.

It'd also be helpful if you also included the gcode file you were trying to print as well as tell me two of the settings from the GPX settings panel: machine type dropdown and gcode flavor.

leofaoro, 2016-03-06:
Thanks Mark. I will close it here and open in the GPX project.

closed 2016-03-06 by leofaoro