OctoPrint/OctoPrint #1260: [Request] Stream to youtube

boelle, 2016-03-06:
Hi

I want to record some of my prints with youtube,

I know its possible here: https://support.google.com/youtube/answer/2474026?hl=en

basicly i should just stream to rtmp://a.rtmp.youtube.com/live2 

and it will show up automatic, of course a stream key is needed also but that one i will not show here

Naesstrom, 2016-03-08:
I second this, was just checking the issues to see if there was a suggestion about it already! :+1:

foosel, 2016-03-08:
Since OctoPrint is only a consumer of the webcam itself, it doesn't make sense building this into OctoPrint (even as a plugin). Why add on to the load of tasks the tool responsible for keeping your printer fed for something that could be more easily (and better) done with an external tool? I'd suggest looking into vlc or streamer, feed in the mjpg url of the webcam server and have it dump that to YouTube, that's what tools like that are for. OctoPrint is a print server, not a OS ;)

closed 2016-03-08 by foosel