OctoPrint/OctoPrint #1264: GCode Viewer is not centered

clovett, 2016-03-07:
#### What were you doing?

I used the "Center" option in my printer profile (for Mendel printer) and it prints correctly, but GCode viewer still shows the model on the bottom left.  I noticed issue 682 is supposed to be fixed, so how do I configure the GCode viewer so it honors the printer profile setting?

#### What did you expect to happen?

The GCode Viewer should show the model in the center of the plate.

#### What happened instead?

The GCode Viewer showed the model in the bottom left corner of the plate.

#### Branch & Commit or Version of OctoPrint

[Can be found in the lower left corner of the web interface.]

#### Printer model & used firmware incl. version

Mendel 90

#### Browser and Version of Browser, Operating System running Browser

Chrome Version 49.0.2623.75 m, and Microsoft Edge 25.10586.00.

#### Link to octoprint.log

N/A

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s) showing the problem:

N/A

I have read the FAQ.

foosel, 2016-03-07:
Version & branch of OctoPrint, example gcode file and ideally more info on your configured printer profile please.

foosel, 2016-03-10:
@clovett ping?

clovett, 2016-03-11:
I think you can close the issue, somehow after rebooting the pi and setting the printer as default I can no longer repro this problem.  It seems to be working fine now.  I'll reopen it if I can get a clear repo.

closed 2016-03-11 by clovett