OctoPrint/OctoPrint #1265: Octoprint Z offset

SYC-O, 2016-03-08:
Hi, When I print directly from Simplify3D my Z offset is used and the first layer is at the correct height, however when I save the GCode and upload it to Octoprint, it seems to ignore the Z offset and start's the print too high. 
I checked the gcode in the terminal view and it shows a G1 command of Z0.4 right after doing bed auto leveling, where my offset should be Z-0.5.

This has only just started happening, I've had many successful prints with it previously, can anyone suggest whats wrong?

Many thanks Simon

WheresWaldo, 2016-03-08:
This really isn't an Octoprint issue as octoprint shouldn't mess with existing gcode.

Z_Offset can be added to all prints in two different ways. In S3D simply go and edit the process, choose **Scripts** > **Startup Script** and see if your _Z_Offset_ command is there (typically **M565** or **M851** depending on specific implementation of Marlin). Alternatively if you have EEPROM settings enabled in your firmware, you can issue the appropriate command for setting your _Z_Offset_ followed by an **M500** command to store that setting in EEPROM.

foosel, 2016-03-08:
Like @WheresWaldo. Not an issue, and for support requests/questions there are the mailing list and the G+ group. Closing.

closed 2016-03-08 by foosel