OctoPrint/OctoPrint #1266: Feature Request: G-Code for take timelapse picture

mrvn, 2016-03-08:
I love the timelaps feature in octoprint. Unfortunately using lift for moves totaly screws with taking pictures on Z change. It would be nice to have a special G-code to take a picture. One could insert that easily as "custom G-code between layers" in many slicer. When encountering such a G-code octoprint would take a picture and omit the G-code from the stream so as not to confuse the printer.

foosel, 2016-03-08:
PR #1148 will solve the issue with z based time lapses and z-hops.

mrvn, 2016-03-08:
concering PR #1148: The patch assumes that you have a constant z hop height and have configured the same value in the slicer and octoprint. That probably often works. But then someone will tune his slicer config and suddenly the timelaps is all screwed up again. Then there are tool changes and wiping that will introduce further Z movement that would have to be compensated. Or the different layer height for supports or for combined infills will throw it off. Overall it seems fragile.

On the other hand with a picture G-code the slicer can insert it at the right moment. You can take a picture for every real layer but skip the support and infill layers. Or only take a picture every 3 layers. Or only between 10mm and 20mm height where the interesting bits are.

Would you accept a patch for this when I find the time to write one?

mrvn, 2016-03-10:
Note: The G-code already exists (available in Marlin firmware):

     M240: Trigger camera 
    Triggers a camera to take a photograph. (Add to your per-layer GCode.)

blddk, 2016-05-01:
Would work, I am mostly doing timelapses of vases with continuous Z movement, and used to use M240 which is inserted by the slicer.

idh, 2016-07-08:
I need to be able to take unobstructed pictures for a plugin used by a commercial project. I'm still new to OctoPrint and Python so I will need some help getting it to work. If someone is willing to help me, I can see if my employers would be willing to help fund you and/or the project. My thought would be to take the code from the working plugin and then fold it back into a mixin along with a corresponding example plugin. The resulting code should be automatically triggered using Timelapse or by each M240 in the G-code file.

gruvin, 2017-05-30:
For my two cents ... IMO, the existence of `M240` is errant. Octoprint is not the machine under control and it should therefore not be intercepting and interpreting gcodes. Instead, I would suggest using special tags in gcode comments, as it done elsewhere.