OctoPrint/OctoPrint #1268: Octoprint Cura sets Bed Temp

135erHeli, 2016-03-10:
What were you doing?

Slicing via Cura in Octoprint with Profiles

What did you expect to happen?

No Bed Temp is Set in the Slicing Profiles, but Octoprint Sets Bed Tempratures for PLA to 70°C

Branch & Commit or Version of OctoPrint

Master 1.2.9
[Can be found in the lower left corner of the web interface.]
Printer model & used firmware incl. version

P3Steel Repetier Firmware 0.9.2
[If applicable, always include if unsure.]
Browser and Version of Browser, Operating System running Browser

Firefox and Safari MAC OSX 10.10.5
[If applicable, always include if unsure.]
Link to octoprint.log
[octoprint.pdf](https://github.com/foosel/OctoPrint/files/166806/octoprint.pdf)

[On gist.github.com or pastebin.com. Always include and never truncate.]
Link to contents of terminal tab or serial.log

[serial.pdf](https://github.com/foosel/OctoPrint/files/166805/serial.pdf)

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.]

Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.]
Screenshot(s) showing the problem:

[plugin_cura_engine.pdf](https://github.com/foosel/OctoPrint/files/166807/plugin_cura_engine.pdf)
[Cura High Quality PLA Profile.pdf](https://github.com/foosel/OctoPrint/files/166808/Cura.High.Quality.PLA.Profile.pdf)

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.
I Did

Should not be a Bug but maybe a Profile Editor would be Nice
Where i can find the Point where Octoprint sets Bed Tempratures himself

0

foosel, 2016-03-10:
Fixed on `maintenance` branch, will go into 1.2.10 release.

Uncheck "Heated bed" in your printer profile until then.

135erHeli, 2016-03-10:
Ahh ok thx.

closed 2016-03-10 by 135erHeli

DrCord, 2016-06-14:
I am using octoprint 1.2.12 and having this issue still.

foosel, 2016-06-14:
@DrCord please provide a filled out [issue template](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report) and a profile with which to reproduce.

DrCord, 2016-06-14:
Turns out the upgrade from 1.2.7 to 1.2.12 somehow changed my machine profile to have the "heated bed" option checked which was causing the issue.

commit 23d2cdec (2016-03-10): Use 0 as default bed temperature in slicer

Should prevent slicing result to contain bed temperature settings even if
printer has a heated bed configured unless slicing profile also contains a
temperature.

Fixes #1268